rstrouse/ESPSomfy-RTS #70: Possibility to connect Buttons to GPIO

d3bu9, 2023-06-13:
Is it possible to attache some Buttons on free GPIO Ports to use the ESP directly as a Remote Control?

I learned all the Motors and it works great. Thanks for your Work. Now, next Level would be, if I directly could attach some Buttons.

[image]

rstrouse, 2023-06-13:
Are you thinking about up/my/down buttons or something else like a reboot button?

d3bu9, 2023-06-17:
Yes, it's more of a feature Request. My Idea ist to connect Buttons to controll up, my, down for 3 blinds in the living room to repleqce the existing somfy smoove wall Button, which can only handle one.

rstrouse, 2023-06-17:
That is doable in the firmware but setting it up would be a challenge because it would need to manage up to 32 signals to select the shade.  There simply aren't enough gpios to cover it.  You would need 5 pins for the selector and another 3 for the basic functions.  But to make that happen you would also need some encoding circuitry to pull the pins in the proper direction.

They do make a Smoove that has 4 channels.  I assume you have this paired up so this smoove switch controls all 3 at the same time.  With 3 blinds this would allow you to control each blind individually, leaving the 4th channel to control all 3 at once.

ic3ek, 2024-01-08:
I would have suggestions, using the same settings as in IO-Remote , connecting the button "monostable" under the appropriate GPIO will trigger the remote control signal to be sent,

Changing IO-Remote to IO-Button

I know that the number of GPIOs is limited, but most will use 1 or 2 buttons, not more.

And I think converting the program from OUTPUT to INPUT will not be a challenge, as far as GPIO trigger state is concerned, I think you can specify one low/high, no choice, you specify one and everyone will adapt to it ! They will do as you program.

If you change the software from OUTPUT to INPUT , I would be happy to test the device to see if it works properly this way.

padwyatt, 2024-02-02:
I'd also be interested in this enhancement. I am controlling espsomfy by issuing rest commands from a Shelly that is attached to a built in wall switch. It works, but it would be even better if I could wire the espsomfy directly to the wall switch! Only up and down inputs are needed.

PS: great project, with excellent documentation!

rstrouse, 2024-02-02:
@padwyatt you can already do what you want and connect to a relay on the GPIO pins of the ESP32.  I have never used a shelly but if it has input signals then it should be able to trigger from ESPSomfy RTS.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Controlling-Motors-with-GPIO

padwyatt, 2024-02-09:
@rstrouse Thankyou. I did read that page quite carefully; but I think I want to do the reverse. Rather than triggering an external relay from the output of the ESP32, I want instead to trigger the ESP32 using an external (manual) switch, via the GPIO.

Does that make sense?

rstrouse, 2024-02-09:
So the Shelly isn't controlling a bare motor?  Which Shelly are you using?

padwyatt, 2024-02-23:
The setup is that the wall switch is directly connected to a shelly i4ac. This catches the up and down button presses, and then sends requests to a esp32 in another cupboard running your lovely software... which then in turn controls the blinds via rf.

Ideally, I'd like to chop out the shelly, and just put the esp32 directly behind the wall switch.

rstrouse, 2024-02-24:
I'll have to think this through.

This is the potential here using pins on the ESP32 and the corresponding channel possibilities.

| Channels |  # of Pins | Comments |
| ------ | ------ | ------ |
| 1 | 3 | No channel select pins are used but it can only control a single channel |
| 2 | 4 | Up to 2 channels by enabling a single channel select pin |
| 4 | 5 | Up to 4 channels by enabling combinations of 2 channel select pins |
| 9 | 6 | Using different pin mappings over 3 channel select pins |
| 16 | 7 | This uses 4 channel select pins |

fredcazaux, 2025-08-30:
> I would have suggestions, using the same settings as in IO-Remote , connecting the button "monostable" under the appropriate GPIO will trigger the remote control signal to be sent,
>
> Changing [...]

+1

I've imagined exactly the same. Quickly tried to read the code to implement this but didn't find where to start with for now... Any tip please ?

Thx