rstrouse/ESPSomfy-RTS #72: Can't pair shade

Calimerorulez, 2023-06-15:
Hi,

First of all, thank you for your great project.

I'm trying to pair my first roller shade, but I can't seem to make it work. I've configured the radio, and when I press my Somfy RTS Situo 5 Pure II remote, I can see the commands appearing in the logs. When I press the Prog button on the remote, the motor jogs, but when I click Pair on the Pair shade screen, the shade does not jog.

Is there a way to check whether the radio transmits, when I click Pair?

Calimerorulez, 2023-06-15:
Small update, I've checked if the radio transmits when I click Pair, and it does transmit. I've used an SDR-stick, and I see the transmission on 433,42 MHz.

Calimerorulez, 2023-06-15:
OK, I swapped the RX and the TX in de GUI of the radio settings, like the German guy mentioned in the recent Youtube video, and now it seems to work :)

https://youtu.be/dtvTQM6dqcQ?t=852

rstrouse, 2023-06-16:
Ok I am going to close this one out and make some edits to the docs to add a troubleshooting section.  The tell-tale sign of having the RX and TX swapped it being able to hear but not speak.  This is because the CC1101 can only speak on GDO0 and can listen on GDO0 or GDO2.

closed 2023-06-16 by rstrouse