rstrouse/ESPSomfy-RTS #73: Request: Add GPIO33

Calimerorulez, 2023-06-16:
Hi, it's me again :)

I'm trying to get an Atom Lite to work with the CC1011, and so far so good. But the config page for the radio is missing any GPIO higher than GPIO31 (only the RX pin has them).

The Atom Lite has a GPIO33 pin. Is it possible to add the higher pins to the dropdown boxes in the GUI?
Thanks!

rstrouse, 2023-06-16:
Yep I will be releasing v1.7.1 this weekend but have posted the code and binaries and code to the repo already.

rstrouse, 2023-06-17:
v1.7.1 has been released and contains the additional selections

closed 2023-06-17 by rstrouse

pafnow, 2025-03-28:
> Hi, it's me again :)
>
> I'm trying to get an Atom Lite to work with the CC1011, and so far so good. But the config page for the radio is missing any GPIO higher than GPIO31 (only the RX pin has [...]

Hi, I'm trying to achieve the same, could you share your pin mapping that you used ? TIA

Calimerorulez, 2025-03-28:
> Hi, I'm trying to achieve the same, could you share your pin mapping that you used ? TIA

[image]

I managed it to do without GPIO33 :)

pafnow, 2025-03-30:
Thanks @Calimerorulez