rstrouse/ESPSomfy-RTS #74: 315 MHz gas fireplace control

mhmatttu, 2023-06-18:
I wished I had found this before ordering parts on the slow boat to build an rpi RF transmitter for my somfy shade.  I may still move towards this solution.  The work and support you've provided on njsPC is amazing so I'm certain this will be a much better option.

For a friend, I have also been researching a solution to control a 315 MHz RF gas fireplace (real fyre) through home automation (hubitat, homebridge, mqtt, alexa), but have found very few options and those seem more technical than I can manage.   Of course there is Bond that would do the trick but at $100, I was looking for more of a home brew option.

From what I can tell, this transmitter (CC1101) is capable of 315 MHz.  I'm curious if ESPSomfy could be used/modified to control (on/off) this gas fireplace.

Thanks!

rstrouse, 2023-06-18:
Yes the CC1101 can receive and transmit on 315Mhz but the devil is in the details.  The protocol, baud rate, and encoding are really unknowable without analysis of the signal.  I think what the Bond folks are doing is playing mockingbird with the remote.  If there is any sort of additional encoding then they code around that but only for devices they support directly.  In the end, what that produces is largely one-way communications.  Which is fine when you really don't care about the current state.

When I went looking for a Somfy solution that is pretty much what I found.  I also realized when I started listening to the patient, many of the original assumptions were a bit deceptive as to how the RTS frames are encoded.  I really wanted to know what position these shades were in as reliably as I possibly could.

You will find lots of stuff out there to send a command and the methods used in ESPSomfy-RTS will likely be very similar for that device as it is most likely using ASK/OOK modulation, the encoding will be much different and probably much simpler. 
 That being said I am not in the position to decode that protocol for you.

Much of the ESPSomfy-RTS code is there to allow a user friendly approach to configuration and management of the peripheral radio and motor setup.  But it also has code in there to make it easy to integrate with other systems that you could easily make use of.  It's goal wasn't to be the control platform but act as a contained device to interact with other platforms.

mhmatttu, 2023-06-18:
Understood!  Thank you for the detailed information!

rstrouse, 2023-06-21:
I am going to close this one as I do not have that equipment.

closed 2023-06-21 by rstrouse