rstrouse/ESPSomfy-RTS #75: Automatic sun not updating

Henk72, 2023-06-20:
When automatic sun is enabled either via the physical Somfy RTS remote or via this software, the awning responds with a jog and it works automatically as expected.

The issue is that if the awning is extended/retracted due to the sun sensor, the awning reacts, but HA does not get updated. So the result is that the awning status is not correct. 

I noticed that the sun sensor is also not updated in HA. So if the sun is up (RTS sensor), it is not reflected in HA.

I will later try to get a log for sun on and sun off and attach it.

rstrouse, 2023-06-20:
Yeah I need that log so I can see if I am decoding the frame correctly.

wutz-ESP, 2023-06-21:
Hi, is is possible to activate automatic sun to Universal Slim Receiver RTS Plug or Soliris Mod / Var Slim Receiver RTS Plug ?
I am using this to external blinds with remote Somfy radio remote control Situo 5 Var A/M RTS Pure II.

Awnings controled by Situo 5 RTS Pure II get symbol to activate automatic sun.

rstrouse, 2023-06-21:
> Hi, is is possible to activate automatic sun to Universal Slim Receiver RTS Plug or Soliris Mod / Var Slim Receiver RTS Plug ?
I am using this to external blinds with remote Somfy radio remote control Situo 5 Var A/M RTS Pure II.

I believe both of those support the Soliris sun/wind sensor.  ESPSomfy RTS supports that 80-bit remote including the trim wheel.

Henk72, 2023-06-21:
The receiver seems to get burst of the sun sensor and after that it quiets down. I have added a screen shot for your convenience. I pressed the My button to check if the lack of sensor data is caused by the receiver, which it is not. The receiver is also very close to the sensor for this test.

I checked the RTS Solaris manual and it mentions that the awning is actuated with a sun signal after at lease 2 minutes. It closes down after 15 or 30 minutes no sun. The 15/30 depends on the time the sensor was on or off. This is working as a hysteresis.

The weird thing is that the Sun sensor is only sending once in a while for about 17 seconds. See next 2 instances:

image
g width="490" alt="image" src="[https://github.com/rstrouse/ESPSomfy-RTS/assets/14115589/f730a038-b405-40e4-a9b2-c9f135c75eb5">](https://github.com/rstrouse/ESPSomfy-RTS/assets/14115589/f730a038-b405-40e4-a9b2-c9f135c75eb5%22%3E)

[{"encKey": 160,
"address": 9277338,
"rcode": 2,
"command": "Sensor",
"rssi": -75,
"bits": 56,
"valid": true,
"sync": 4,
"pulses": [2434,2568,2433,2562,4752,1319,1234,1303,1275,658,618,666,626,634,639,664,623,644,630,1311,1247,676,610,665,617,1320,615,666,595,677,1264,1300,1256,663,626,649,620,1309,615,669,602,683,1258,1295,1259,665,619,674,611,1291,1272,1307,613,664,618,675,1262,646,623,1298,1265,645,624,658,637,680,595,1298,1260,670,623,679,609,1290,619,677,611,662,612,672,599,678,1262,1298,629,659,606,677,1247,1308],
"time": "2023-06-21T16:20:46.930+0200"
},
{"encKey": 160,
"address": 9277338,
"rcode": 2,
"command": "Sensor",
"rssi": -74,
"bits": 56,
"valid": true,
"sync": 14,
"pulses": [598,26903,2421,2578,2420,2559,2448,2552,2449,2552,2446,2553,2448,2559,2444,2550,4753,1316,1249,1334,1222,677,611,659,625,672,603,658,638,652,620,1291,1269,669,631,634,638,1291,619,678,609,665,1276,1304,1235,671,622,657,633,1291,619,675,609,658,1267,1326,1225,669,629,659,614,1315,1247,1315,620,670,605,659,1264,645,626,1310,1250,697,584,686,601,671,621,1306,1253,669,610,657,630,1299,629,662,612,664,623,666,607,660,1255,1318,609,664,624,668,1252,1309],
"time": "2023-06-21T16:20:47.106+0200"
},
{"encKey": 160,
"address": 9277338,
"rcode": 2,
"command": "Sensor",
"rssi": -75,
"bits": 56,
"valid": true,
"sync": 14,
"pulses": [595,26779,2458,2552,2442,2569,2440,2571,2410,2566,2435,2568,2453,2538,2449,2562,4751,1297,1256,1316,1245,683,607,652,645,646,633,659,616,642,626,1310,1264,671,607,659,616,1321,608,662,613,666,1274,1314,1247,656,608,683,621,1298,636,630,624,668,1256,1312,1243,685,605,666,619,1303,1257,1293,635,658,617,669,1255,684,598,1317,1240,678,618,649,636,659,613,1300,1255,663,624,670,628,1286,622,666,628,660,606,678,619,672,1235,1316,622,653,634,641,1274,1298],
"time": "2023-06-21T16:20:47.314+0200"
},
{"encKey": 160,
"address": 9277338,
"rcode": 2,
"command": "Sensor",
"rssi": -75,
"bits": 56,
"valid": true,
"sync": 14,
"pulses": [604,26802,2458,2564,2430,2564,2424,2562,2448,2559,2449,2561,2423,2555,2468,2534,4775,1301,1250,1316,1246,680,608,663,619,671,607,660,631,651,619,1303,1265,669,602,680,613,1314,595,678,613,659,1276,1298,1257,670,600,683,608,1289,646,656,611,663,1250,1321,1256,670,601,682,613,1291,1272,1286,623,669,628,660,1257,670,601,1308,1267,652,633,660,627,645,623,1310,1263,669,607,659,615,1322,606,664,611,671,622,655,613,683,1242,1323,612,665,619,648,1278,1289],
"time": "2023-06-21T16:20:47.342+0200"
},
{"encKey": 160,
"address": 9277338,
"rcode": 2,
"command": "Sensor",
"rssi": -75,
"bits": 56,
"valid": true,
"sync": 14,
"pulses": [595,26816,2449,2581,2427,2558,2449,2551,2445,2549,2444,2556,2449,2557,2444,2556,4746,1303,1277,1290,1248,678,609,681,594,670,628,659,617,661,621,1297,1261,671,623,656,609,1314,620,651,636,663,1253,1297,1265,674,597,675,608,1307,644,657,609,664,1251,1325,1238,671,622,655,609,1315,1272,1285,624,668,628,658,1261,645,624,1310,1263,670,606,662,613,664,619,1298,1259,25,667,625,655,633,1305,621,656,611,684,597,669,629,658,1266,1297,608,684,612,663,1246,1310],
"time": "2023-06-21T16:20:47.480+0200"
},
{"encKey": 160,
"address": 9277338,
"rcode": 2,
"command": "Sensor",
"rssi": -77,
"bits": 56,
[... 694 lines cut ...]
"pulses": [630,26775,2477,2534,2447,2536,2470,2535,2449,2553,2449,2557,2449,2551,2445,2532,4776,1293,1270,1314,1244,650,636,665,628,645,621,656,639,641,648,1272,1289,616,672,617,632,1313,623,643,631,637,1293,1271,1282,654,643,626,635,1288,648,644,630,637,1285,1272,1309,626,642,651,630,1292,1269,1281,654,646,622,655,1262,649,645,1277,1266,668,632,635,639,632,646,1319,1243,648,647,623,638,1312,618,647,640,637,652,620,646,655,1264,1295,630,659,634,639,1274,1281],
"time": "2023-06-21T16:36:14.904+0200"
},
{"encKey": 160,
"address": 9277338,
"rcode": 2,
"command": "Sensor",
"rssi": -74,
"bits": 56,
"valid": true,
"sync": 14,
"pulses": [632,26775,2450,2539,2477,2543,2447,2535,2476,2525,2471,2535,2450,2559,2452,2529,4782,1267,1295,1284,1272,647,635,636,640,645,627,659,630,639,641,1277,1292,645,625,655,634,1289,622,657,635,632,1324,1259,1267,646,648,635,640,1288,619,651,634,639,1281,1319,1240,668,628,636,635,1282,1300,1288,622,642,631,661,1258,642,652,1283,1266,676,612,635,640,648,644,1284,1278,645,631,661,615,1293,644,653,611,660,622,644,653,635,1287,1271,633,661,641,652,1270,1263],
"time": "2023-06-21T16:36:15.048+0200"
},
{"encKey": 160,
"address": 9277338,
"rcode": 2,
"command": "Sensor",
"rssi": -75,
"bits": 56,
"valid": true,
"sync": 14,
"pulses": [625,26818,2438,2544,2453,2540,2472,2539,2471,2515,2475,2536,2466,2531,2473,2509,4776,1292,1274,1288,1273,649,634,635,650,643,647,635,637,664,622,1318,1235,647,642,629,663,1265,645,648,635,660,1277,1273,1283,640,621,657,636,1284,646,650,613,667,1263,1298,1265,646,641,653,634,1282,1269,1285,623,643,653,633,1287,640,645,1281,1282,643,632,661,613,652,647,1277,1269,643,650,634,632,1292,644,653,610,665,616,647,655,639,1285,1268,662,631,640,642,1279,1269],
"time": "2023-06-21T16:36:15.180+0200"
}]

Henk72, 2023-06-21:
The screen shot went wrong, I will post a new one, but if can also be found in the above data

Henk72, 2023-06-21:
I paired the remote in the past, but in the app overview I see a different number (see attached screenshot). I do know that the listed 9277338 is my remote as it responds to the button press in the logs immediately.

Screenshot:
[image]

Henk72, 2023-06-21:
> I am seeing the sensor frame about every 3 to 4 seconds and it sends it 5 times (1 + 4 repeats). I assume this is the sunny condition. 

Yes it is the sunny condition

> Does it just continue to send out at this interval when there is no sun?

I think it just happened and the log is giving a value 0 instead of 2. See next log:

```
[{"encKey": 163,
  "address": 10147277,
  "rcode": 4426,
  "command": "My",
  "rssi": -72,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [10416,7339,2474,2556,2464,2574,4799,1309,1253,1339,1247,681,605,688,626,1302,623,668,607,684,1255,1344,589,704,596,672,609,691,619,668,612,686,615,671,1256,1336,1249,1311,620,694,611,664,1249,1323,629,676,596,700,1256,665,606,1333,1266,679,607,666,619,1347,601,664,621,697,582,691,620,668,611,686,1251,1321,1255,1321,1267,1341,1234,1342,596,699,1235,1325,611,688,1249,695,608,1317,618,671,1266],
  "time": "2023-06-21T17:00:52.097+0200"
},
{"encKey": 160,
  "address": 9277338,
  "rcode": 2,
  "command": "Sensor",
  "rssi": -75,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [2441,2548,2454,2528,4784,1287,1266,1300,1281,643,630,633,659,617,668,620,659,637,631,1301,1257,656,640,644,629,1284,641,645,630,633,1290,1298,1266,663,618,650,633,1287,643,645,634,638,1290,1271,1285,645,622,657,637,1291,1275,1284,641,646,636,636,1303,622,628,1290,1274,650,636,640,652,670,602,1280,1280,642,656,638,641,1268,645,655,635,634,648,641,627,660,1264,1291,628,637,659,628,1295,1261],
  "time": "2023-06-21T17:06:32.991+0200"
},
{"encKey": 160,
  "address": 9277338,
  "rcode": 2,
  "command": "Sensor",
  "rssi": -75,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [627,26867,2464,2558,2450,2524,2483,2523,2467,2541,2465,2526,2484,2525,2456,2540,4778,1273,1287,1266,1273,658,631,650,641,654,614,636,667,622,645,1309,1266,644,628,635,639,1291,629,638,665,628,1296,1259,1281,647,631,659,641,1271,634,636,663,640,1273,1285,1269,659,634,640,652,1271,1289,1265,645,628,664,615,1302,646,634,1262,1290,656,640,641,650,616,649,1287,1277,644,635,635,655,1275,632,637,662,649,619,646,638,639,1267,1297,660,633,625,646,1272,1293],
  "time": "2023-06-21T17:06:33.123+0200"
},
{"encKey": 160,
  "address": 9277338,
  "rcode": 2,
  "command": "Sensor",
  "rssi": -74,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [619,26744,2455,2536,2452,2545,2452,2565,2447,2535,2473,2532,2446,2556,2447,2556,4754,1293,1269,1285,1291,652,611,659,625,647,649,634,636,641,645,1273,1286,647,645,652,611,1292,644,648,634,636,1281,1273,1294,644,619,649,637,1310,643,628,636,639,1291,1270,1290,651,620,647,635,1290,1265,1283,639,640,645,656,1266,647,644,1279,1291,619,658,638,631,649,644,1278,1267,645,650,635,635,1291,646,628,661,640,640,620,647,655,1264,1300,628,635,657,642,1269,1284],
  "time": "2023-06-21T17:06:33.255+0200"
},
{"encKey": 160,
  "address": 9277338,
  "rcode": 2,
  "command": "Sensor",
  "rssi": -75,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [623,26822,2441,2543,2461,2543,2453,2543,2471,2539,2449,2536,2475,2531,2468,2536,4777,1266,1297,1288,1271,626,638,665,626,646,624,658,642,631,648,1271,1289,649,618,650,637,1289,644,647,635,636,1305,1247,1288,648,643,628,635,1313,619,648,635,637,1280,1298,1260,648,644,631,637,1310,1269,1287,626,644,650,634,1294,621,644,1286,1280,646,633,635,659,627,643,1301,1259,645,649,633,635,1289,648,637,637,634,643,645,634,633,1286,1294,633,660,616,644,1276,1286],
  "time": "2023-06-21T17:06:33.394+0200"
},
{"encKey": 160,
  "address": 9277338,
  "rcode": 2,
  "command": "Sensor",
  "rssi": -74,
  "bits": 56,
[... 366 lines cut ...]
  "time": "2023-06-21T19:50:35.735+0200"
},
{"encKey": 160,
  "address": 9277338,
  "rcode": 0,
  "command": "Sensor",
  "rssi": -75,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [597,26728,2449,2541,2448,2539,2470,2530,2472,2531,2468,2532,2447,2554,2450,2525,4770,1294,1273,1288,1273,647,636,639,640,643,649,634,632,661,619,1300,1258,669,627,635,640,1285,1271,1288,1272,1301,1267,642,630,660,633,1277,1276,1291,1270,1279,1290,645,634,639,629,1299,1279,1269,645,646,634,635,1280,644,632,1290,1273,674,612,635,653,648,622,1308,1250,669,609,659,642,1268,631,662,631,641,643,647,636,635,1280,1293,639,640,625,644,1278,1291],
  "time": "2023-06-21T19:50:35.735+0200"
},
{"encKey": 160,
  "address": 9277338,
  "rcode": 0,
  "command": "Sensor",
  "rssi": -74,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [596,26797,2469,2546,2439,2544,2462,2541,2475,2532,2448,2536,2448,2551,2447,2554,4757,1283,1272,1286,1275,648,636,659,616,643,651,633,641,650,621,1300,1268,645,624,658,637,1280,1269,1290,1267,1303,1265,645,648,633,636,1306,1246,1315,1245,1301,1260,667,629,635,634,1305,1273,1266,645,643,630,660,1257,643,650,1290,1277,646,634,636,629,647,646,1280,1282,644,629,661,634,1280,622,656,636,634,646,645,628,659,1265,1296,635,632,641,643,1280,1283],
  "time": "2023-06-21T19:50:36.038+0200"
}]
```

Henk72, 2023-06-21:
> I got the screenshot. Is 9277338 listed as a linked remote for the awning?

It seems this 9277338 is the address of the Solaris RTS sensor, not the RTS remote or awning motor.

In the last log the sun sensor went from value 2 to value 0 at 19:50. Exactly 10 minutes later the awning retracted automatically (RTS driven). I did not receive further frames in the log.

rstrouse, 2023-06-21:
> It seems this 9277338 is the address of the Solaris RTS sensor, not the RTS remote or awning motor.

You are correct 9277338 is the address of the sensor and you must link it in ESPSomfy RTS to the awning. 
https://github.com/rstrouse/ESPSomfy-RTS/wiki/Configuring-the-Software#linking-remotes-and-sensors

> In the last log the sun sensor went from value 2 to value 0 at 19:50. Exactly 10 minutes later the awning retracted automatically (RTS driven). I did not receive further frames in the log.

As it turns out the sensor does not actually send commands to the motor it only broadcasts sensor data.  It is up to the motor to determine how it responds to these.  We currently have this set to 20min per the docs which says 15-30min.

rstrouse, 2023-06-21:
That is the linked remote.  You must also pair the sensor.  So you will have two entries in there.

Henk72, 2023-06-21:
OMG I missed this part since I could see the data in the log (and assumed it was therefore paired). Assumption is the mother of all f@ck ups.

I have paired the sensor by adding a remote and pressing the prog button on the sensor. It is now paired. Tomorrow I can confirm if it works.

Henk72, 2023-06-22:
> ESPSomfy RTS will hear any RTS signal in range and decode it. Just to be sure you linked the sensor using the Link Remote button, is that correct?

Yes, see screen shot for the result:

[image]

rstrouse, 2023-06-22:
Perfect let me know what you see when the sun come out or goes in.

rstrouse, 2023-06-22:
Who said they were going to make it easy.

Henk72, 2023-06-23:
Haha, today it was easy an everything worked well. The sun was visible in HA and the awning got updated.

Henk72, 2023-06-23:
I didn’t check how long the HA update took after the automatic extension. I’m thinking that maybe no hysterese is needed. If the awning is not retracted and the sun sensor gives a signal , the awning opens. I will check this delay. Maybe it is best to creata parameter for the delay in your app. One for extend and one for retract.

rstrouse, 2023-06-24:
Just to make sure we are referring to the same thing.  To me retracted is when all of the fabric is rolled up into the motor and extended is when the fabric is unrolled off the motor.  By default, in somfy, up = retracted and down = extended.  The up button is supposed to retract and the down button is supposed to extend.  

However, this can be inverted in the motor and when you throw HA into the mix it gets all twisted up but believe it or not the only cover type that is correct is awning.  The rest of them have either the buttons backwards or you have to show percentage of open.

So when we are talking about this, lets refer to extended and retracted (not up/down or open/closed) since these two terms are not arbitrarily mixed up depending on where you are standing around the awning.

Henk72, 2023-06-24:
Sorry, I already tried to use retracted and extended for a while, but I was inconsistent. Retracted/extended is indeed the only clear definition.

rstrouse, 2023-06-24:
Yeah I realized that I had been thinking the wrong direction in the past.  Let me know if you see any patterns here wrt to the delays.  The first delay is the one to extend the awning but I think that is probably as good as we are going to get at 2 minutes.  According to the docs that is fixed at 2 minutes of sun detection.

For the sensor to retract the awning (according to the docs) it takes a minimum of 15 minutes to a maximum of 30 minutes.  I suspect it is 30 minutes when the awning has been extended for a short period of time and 15 minutes if it has been out for a long period of time.  The way it is written in the docs they make it seem that the delay period is a sliding scale over the 30 minute period for no less than 15 minutes.

Unfortunately, there isn't any indication as to how long one needs to wait until the awing retracts at the 15 minute minimum.  I don't know how one could possibly test for that without making it a past time.  Maybe we pick 1 hour as the total dwell time for the heuristics and do a linear adjustment over that period.

Henk72, 2023-06-30:
The latter sound like a good suggestion. The solution is working quite well now so it is loosing my interest. I will check the behaviour in real life when using the awning automatic mode. If I find a better insight in the current working or improvement I will let you know and reopen an issue. 

Thank you for all your responsive help!

closed 2023-06-30 by Henk72