rstrouse/ESPSomfy-RTS #76: Cannot connect to Wifi

Lectere, 2023-06-22:
I've bought three ESP32 WROOM 32 devices, wanted to make the ESP Somfy and two stock. I've started soldering, and I had a hard time connecting to the WIFI. I've tried flashing one of the other EPS32 boards that don't have anything attached on it; same problem. If I scan in the web interface, it sums up the networks. And the SSID and password are saved correctly. I'm going to try another image to see if it's the hardware...

Lectere, 2023-06-22:
I've flashed with ESPHome, same problem... So apparently all three boards are fucked up... :(

Lectere, 2023-06-22:
I've setup a mobile access point with my phone, and it can connect to that. 

So it has something to do with my Wifi router. I've tried it in two locations, both Asus routers with Dual band smart connect.

Lectere, 2023-06-22:
I had a look at the code, and it's all written in the arduino ide v1.8. Could it be that your wifi lib is a bit old?

rstrouse, 2023-06-22:
I used the v1.8 ide but the code is using the v2.0.6 libraries and wifi is up to date.  This is because the 2.0.6 ide did not support littlefs images.  But alas, that is likely not the problem.

If you use the serial console on the ESPHome website you can see the connect sequence and see if there are any issues during connect.

rstrouse, 2023-06-22:
Hmmm... no errors almost like the router is just disconnecting directly after connect.  For shucks disable 802.11b if you have it enabled.  Also what authentication method is used in your router?  WEP, WPA, or WPA2?

rstrouse, 2023-07-13:
I am going to assume you got this sorted.  Closing this for now.

closed 2023-07-13 by rstrouse

Lectere, 2023-07-14:
Well, I don't think this is solved actually. But I cannot solve it, since I have a hard time loading this in visual studio. 

But facts are;
-I cannot connect to the Wifi with the ESPSomfy firmware
-I can connect with the ESPHome default image

I've used a work-around, disabling 802.11b made it 'work', but if you want ESPSomfy to be just as compatible as ESPHome work needs to be done on the ESPSomfy image. I've ran it on the same hardware...