rstrouse/ESPSomfy-RTS #77: RTS remote on 433.92 MHz

Calimerorulez, 2023-06-25:
Hi,

My two shades controlled by a Situo remote are working perfectly with your beautiful project, thank yo so much for that.

In my attic, I have two roller shutters, which I can't seem to pair. I've now used Airspy with an RTLSDR USB radio stick and RTL_433 to do some debugging, and the (brandless) remotes for the sutters do not use the 443.42 MHz frequency, but they use 433.92 MHz. That could explain why button presses on the brandless remote are not shown in the ESPSomfy radio log.

Strange thing is that the remotes do use the RTS protocol:

[image]

I've tried to tune the radio in ESPSomfy to the frequency 433.92 MHz, but frames aren't decoded or displayed.

Can you maybe point me in the right direction?

rstrouse, 2023-06-25:
If you install the firmware and application binaries from the main repo it will now allow you to set the base frequency all the way up to 434.399Mhz.

Calimerorulez, 2023-06-25:
Hi @rstrouse,

Thank you for the update, I've just installed 1.7.3. With a center frequency of 433.92 MHz, the log now shows Unknown commands for Up, Stop/My, Down:

```
[{"encKey": 150,
  "address": 746535,
  "rcode": 3513,
  "command": "Unknown(15)",
  "rssi": -50,
  "bits": 56,
  "valid": false,
  "sync": 4,
  "pulses": [10074,97476,2548,2574,2531,2586,4851,1286,1292,633,662,1279,1295,1291,646,624,1309,649,642,1282,659,618,1320,615,656,1292,1286,1296,1302,1277,646,634,1288,1291,1309,620,669,633,635,1278,669,635,1287,1293,1284,639,669,633,662,1254,670,634,635,641,1300,1283,657,618,1320,1266,1301,1286,1301,1281,634,646,643,632,664,639,649,644,1301,1276,1296,631,665,1271,661,614,1299,657,637,1279],
  "time": "2023-06-25T19:40:48.269+0200"
},
{"encKey": 149,
  "address": 746535,
  "rcode": 3514,
  "command": "Unknown(15)",
  "rssi": -48,
  "bits": 56,
  "valid": false,
  "sync": 4,
  "pulses": [10079,97478,2544,2580,2538,2573,4874,1260,1304,649,658,1267,1295,1289,1301,1279,1284,1271,660,640,1300,630,661,1290,645,628,1313,617,670,1261,654,643,1297,1266,1321,1266,669,627,661,631,641,646,1302,1277,1296,640,651,642,651,1267,648,645,654,638,1285,1270,686,616,1298,1282,1307,1273,1311,1268,660,615,652,644,649,635,660,625,1294,1289,1299,630,660,1282,643,632,1315,619,670,1286],
  "time": "2023-06-25T19:40:49.663+0200"
},
{"encKey": 152,
  "address": 746535,
  "rcode": 3515,
  "command": "Unknown(15)",
  "rssi": -49,
  "bits": 56,
  "valid": false,
  "sync": 4,
  "pulses": [10057,97492,2563,2562,2533,2576,4879,1271,1311,618,673,1268,644,646,1303,624,667,627,659,639,651,1269,659,640,1302,627,659,1266,669,627,660,615,1304,1280,657,623,1320,1260,1317,1259,1315,1279,657,620,1295,1284,1316,614,659,647,643,1283,656,619,669,635,1312,1269,659,614,1304,1277,1309,1270,1312,1270,654,638,658,621,668,609,25,663,640,1300,1280,1311,618,654,1292,645,623,1308,626,666,1281],
  "time": "2023-06-25T19:40:50.855+0200"
}]
```

rstrouse, 2023-06-25:
That is because the protocol looks like another RTW variant not RTS.  This protocol is also slightly different in that it is using a starting point of 149 for the encryption key.  Is there anything on the remote that indicates the protocol or in the docs?

rstrouse, 2023-06-25:
Update your device again from the repo it should include a protocol option for RTV.  This should send and receive your protocol.  I simply chose RTV because I have no information but I would like to know what the variant is actually called.  At this point I have seen 433.42Mhz protocols for RTS and RTW.  This one is probably not different and while your SDR outputs it with RTS it is definitely not RTS but a variant of the RTW protocol.

As an aside, the radio will not communicate on both 433.42 and 433.92 so you will probably need 2 devices.  You may be able to expand the RX bandwidth and find a center between the two but it really won't be in tune.

Calimerorulez, 2023-06-25:
@rstrouse 

This is great! :) I've paired both shutters, and linked both remotes, and is works 👍 

For this reason I already built two devices. On the remote there is no extra info, only the name "PL01 Transmitter, frequency 433.92 MHz".

I don't have the docs for the shutters, because they've been built in before I bought the house.

Thank you for your quick help!

rstrouse, 2023-06-27:
Awesome.  Glad it worked!

closed 2023-06-27 by rstrouse