rstrouse/ESPSomfy-RTS #79: Issue to pair

Icare2000, 2023-06-25:
Hello,

i just installed and crosschecked everthing i could, but i cannot pair the awning. I can have RX from the receiver, but nothing else. I do not understand your remark about: 

"ESPSomfy-RTS device with your motor, then you likely have the RX and TX selections backwards. The CC1101 can only transmit (TX) using the pin marked GDO0 but it can receive on either GDO0 or GDO2."

what should i do.

i have  this hardware

[image]

Thanks and best regards

rstrouse, 2023-06-25:
Go into the configure transceiver screen and swap the values you have recorded for RX and TX.  Then press Save Radio.

Icare2000, 2023-06-26:
Thanks, this is solved by inverting.... Many thanks for this GREAT job

closed 2023-06-26 by Icare2000