rstrouse/ESPSomfy-RTS #80: Controll Somfy device by time

do1kbl, 2023-06-27:
Hello,
Its possible in the future to control the Somfy device bei the Time?
for Example: 7:00 AM open, 1:30PM go to my 50%, 5PM open, 9PM Close. this will be nice if you can implement it.

its very nice Project, I used an old Remote control and connect it to my Arduino it work, but your project its much smarter. I love it

I used it only standalone

Thx for this Project!

Calimerorulez, 2023-06-27:
You can use something like Home Assistant to automate it 😄

do1kbl, 2023-06-27:
Yes but my apartment is not so smart, i used this esp32 StandAlone. :-)

rstrouse, 2023-06-27:
While in theory this is feasible, it would never reach the level that you would find in any home automation system out there.  I will keep this in the enhancement list just in case though.

do1kbl, 2023-06-27:
This is nice, If you will add it in the future :-) 

i used long time this:
http://somfy-domotica.blogspot.com/2014/05/somfy-controller-operational.html

it work for me, but your system ich much much better :-)

rstrouse, 2023-06-28:
Funny, I had been a long time Vera user but have since switched to Home Assistant.  If I still were using it I would have banged out a plugin.

011V32, 2023-06-28:
I've already considered giving away more of these ESPSomfy-RTS to other families in my area.
However, most families do not have a smart home. So, in addition to this ESP, I would also have to give away a RaspberryPi etc. with HomeAssistant, which then does not stand for it.
From my point of view it would be ideal if the ESPSomfy-RTS could support Matter. Then you can integrate it into any ecosystem.

prbtt, 2023-07-06:
> I've already considered giving away more of these ESPSomfy-RTS to other families in my area. However, most families do not have a smart home. So, in addition to this ESP, I would also have to give [...]

I've developing an app as well to interact with the firmware and a scheduler would be just perfect.

do1kbl, 2023-09-23:
any updates about this?

rstrouse, 2023-09-27:
@prbtt how is your app going.  Do you need any help?

devinfluencer, 2023-10-02:
Maybe one could borrow some code from Tasmota to define time plans.
Is there some memory left on the esp or does espsomfy use it up?

Nickduino, 2024-04-01:
I see two advantages over using Home Assistant:

- simplicity (HA can be tricky for some users. You'll tell me that installing ESPSomfy is trickier but the end user doesn't need to do it himself);
- reliability: it will keep working when HA doesn't.