rstrouse/ESPSomfy-RTS #81: Documentation / Default settings

Pulpyyyy, 2023-06-28:
Hello and thank you for this great tool! I now have my covers always synched.

Regarding documentation, there is an inversion between the default TX/RX values ​​of the documentation/image (GD00/GD02) and the initial default config in  application (GD02/GD00).
So by stupidly following the procedure, we manage to read the frames, but not to pair a new shutter.

011V32, 2023-06-28:
I think that's incorrect too, but the help has been revised once before.
Maybe this is a problem of different understanding?

Pulpyyyy, 2023-06-28:
Maybe it's easier with picture ....
From https://github.com/rstrouse/ESPSomfy-RTS/wiki/Simple-ESPSomfy-RTS-device we have :
[image]
TX is GD00 (pin 13) et RX is GD02 (pin 12)
But from default config (radio page and [doc](https://github.com/rstrouse/ESPSomfy-RTS/wiki/Configuring-the-Software)) 👍 
[image]
TX is GD02 (pin 12) et RX is GD00 (pin 13)
So obviously reading the doc let's user to correct values, but that' a pitfall

rstrouse, 2023-06-29:
The defaults have been changed in the next version.

rstrouse, 2023-07-08:
Version 2.0.0 has been released and the default TX/RX pin mappings have been flipped.

closed 2023-07-08 by rstrouse