rstrouse/ESPSomfy-RTS #83: Add restricted access

Pulpyyyy, 2023-06-28:
Hello,

May it be possible to set up an access restriction with user/password or token (for HA integration)? Currently anyone who knows the IP address is able to connect (and change all the configuration). It is indeed easier during the implementation but all the users of the network are not necessarily intended to do all these manipulations

Regards

rstrouse, 2023-06-28:
I am currently mopping up version 2.0 which has password restrictions for the gears menu among other things like groups, fixed ip addresses, tab interface, and additional mqtt interfaces.

Pulpyyyy, 2023-06-28:
Great  !

011V32, 2023-06-28:
I would prefere a password restriction for the whole device :)
(Normaly it is not operated directly via the ESPSomfy-RTS WebPage, but via HA or higher-level systems and only for the people who I have released it for.)
Also I think this would make it easier to implement :)

rstrouse, 2023-06-29:
Yeah I have built an api key implementation. You can choose between all or just the config.

rstrouse, 2023-07-01:
Getting closer I will need to secure the apis next but that will require some changes to the HASS plugin first.  When I am done you will be able to set either a pin or username/password.
[image]

There is a lot coming in this release including the ability to pair up groups.  This should be posted to the repo in the next week.

rstrouse, 2023-07-03:
The main repo contains the binaries for v2.0.0.  At this point you can flash your device with these binaries for the firmware and application. This will give you the security functions you are looking for as well as motor groups.

The interface has changed quite a bit and I have begin rewriting the wiki.  There is also a UI reference that can be found at
https://github.com/rstrouse/ESPSomfy-RTS/wiki/User-Interface-Reference

Pulpyyyy, 2023-07-04:
Are v2.0.0 binaries public ? I still can't see any corresponding tag or files

rstrouse, 2023-07-04:
I have not published so they do not yet have a tag.  The binaries are in the main repo.

Pulpyyyy, 2023-07-04:
Working great !!

rstrouse, 2023-07-08:
v2.0.0 has been released.  Don't forget to check out the new grouping functionality as well.

closed 2023-07-08 by rstrouse