rstrouse/ESPSomfy-RTS #85: Lost pairing ?

do1kbl, 2023-06-30:
Hello,
Today I disconnect the power from the ESP32 and I used the remote controls to control the shades. But now the ESP32 is not working. If I press the button from the shade nothing will move. I need to pair again or how I can synchronize the remote and ESPSomfy again ?

rstrouse, 2023-06-30:
Did you change the address of the shade to the same as the address of any other remote?  You must not do that although there are several solutions online that do this.  Eventually one or the other will stop working and the shade will need to be reset and re-paired.

do1kbl, 2023-07-01:
No I have this remote with 5 channel 
I set the shade into the prog mode via remote and then I add shade give a name and press there the prog button 

after this I press on link remote and press one button and it work Wirrheit problems few days 

Then I remove the power frei esp half day and I used the remote to control the shades in this time. Then I plug the power back and nothing work. All three shades are not working but on the Transmitter setting I see the dBm change when I press button. I try also to link remote again but not help

rstrouse, 2023-07-01:
Yeah it sound like you aren't trying to emulate an existing remote.  You did the correct thing.  The [Pair Shade](https://github.com/rstrouse/ESPSomfy-RTS/wiki/Configuring-the-Software#pairing-a-shade) function is what links ESPSomfy RTS with the motor.  And it should not be removed from the motor unless the motor is factory reset or it is unpaired.

do1kbl, 2023-07-01:
Ok I test today and now I have the same problem,
but I make a reboot of the ESP and then it work, maybe its some problem with the Transmitter module, I will check next time if I press the button on a SDR it will sent some signals or not.  The shades show the correct position on the webinterface but I dont check if I press open on the remote that ESPSomfy detect it when I have this problem

rstrouse, 2023-07-02:
I have never seen an instance where the radio just stops responding.  When you press the Save button ESPSomfy RTS reinitializes the radio registers.  This leads me to think that the transceiver at some point lost power from the ESP32.  

This is likely either the power supply, wiring, or the transceiver.  It sounds like the ESP never skipped a beat and as soon as the radio was re-initialized it came back on line.

do1kbl, 2023-07-07:
You are right, I change the TRX Module and now I work, i think the old modul has some problem with the power

rstrouse, 2023-07-07:
Great glad you got it working.  I am closing this for now open another issue if you have any more problems.

closed 2023-07-07 by rstrouse