rstrouse/ESPSomfy-RTS #89: Sonesse IO 4 motors

nerukoma, 2023-07-08:
Hi, it seems it does not work with these motors as they are 868-870 Mhz... Remote Situo5 Variation A/M io Pure II

https://service.somfy.com/downloads/bc_v5/sonesse-__40_io_installation_guide.pdf

Calimerorulez, 2023-07-08:
It is the IO protocol, not RTS 😄

rstrouse, 2023-07-08:
Yeah I don't have any IO protocol devices and from the looks of it your options are limited.  For me if the only option is some cloud device it does not and will not make it into my home.  Here are some folks that are looking at the protocol.
https://github.com/Velocet/iown-homecontrol

nerukoma, 2023-07-08:
> Yeah I don't have any IO protocol devices and from the looks of it your options are limited.  For me if the only option is some cloud device it does not and will not make it into my home.  Here are some folks that are looking at the protocol.
> 
> https://github.com/Velocet/iown-homecontrol

Thanks for clarification, I shall look

rstrouse, 2023-07-13:
Unfortunately, the IO protocol is out of scope for this project and I do not have any plans for its support.

closed 2023-07-13 by rstrouse