rstrouse/ESPSomfy-RTS #91: UI troncated / group list

Pulpyyyy, 2023-07-10:
Hello,

A minor problem, but as soon as a group contains too many covers, list is troncated
[image]

rstrouse, 2023-07-10:
For now this is to be expected.  I really don't want to grow the height of the element because you can have up to 32 of them in a group.  But it makes sense to show the grouped items to some extent.  Originally I simply showed the number of motors but the list of items in the group made more sense.  If you have any ideas I am all ears.

Pulpyyyy, 2023-07-10:
Just an idea : You may add different icon for groups and use the popup to show the list (instead of the slide for position)
No need to have a dynanic icon as for cover, it's not revelant for a group except IF ALL covers have the same position

rstrouse, 2023-07-10:
Amazingly I drew all those icons with css because I didn't want all that load on littlefs.  Perhaps that or a hover.

rstrouse, 2024-05-11:
After all this time and trying many different things I finally gave up and just listed the number of members for the group.  This turns out to be just fine because the room names give the ability to be more descriptive with the name.  Stacking the icons is a non starter given that you can have 32 of them and the icons are different depending on which types are members of the group.

closed 2024-05-11 by rstrouse