rstrouse/ESPSomfy-RTS #92: Configure Radio didn´t work on WT32-ETH01 in Version 2.0.0

wutz-ESP, 2023-07-10:
Version 2.0.0 configure radio did not work correct, save change always internal error

rstrouse, 2023-07-10:
I'll have a look.  I have that board.

rstrouse, 2023-07-10:
Ok so I think the lan module needs reset.  To do this go to the network section and press save adapter.  This will enforce the correct settings for the WT32-ETH01 module.  You will only need to do this once.  I think it has some weirdness with the PHY unless it is forced to reconfigure it sends out random stuff on the other pins.  v2.0 initialized this on boot but it needs the initial values to be set.

wutz-ESP, 2023-07-11:
Flashed this device again to Firmware 1.7.2 and configured LAN, shade and remotes.
Before backup didn´t work correct in 1.7.2 - guess this was the problem.
Other device WT32-ETH01 work after upgrade to 2.0.0 without any problem.
Now updated this one and it is working normal.

Maybe there was a problem in configuration before.

wutz-ESP, 2023-07-14:
Configuration lost after a few days.

closed 2023-07-14 by wutz-ESP