rstrouse/ESPSomfy-RTS #93: 30awg: truly wire wrapped?

bruno-monk, 2023-07-11:
In the wiki you mention using 30awg wrapping wire for the interconnection but the corresponding photo has the wires soldered. Would you mind comporting the text to the image for me? Thanks!

rstrouse, 2023-07-11:
I used the kynar (wrapping) wire because it works very well for soldering traces.  It is easy to strip and holds its shape when you bend it.  You could use any wire you like but the wrapping wire is very easy to work with and solder.  So in the pic I soldered this wire between the connections on the bottom of the board.

Wire wrapping is not a skill I possess but I suppose you could do the same thing without the board and a wire wrapping tool.  Even if you are new with a soldering iron this is an easy solder job.  There are lots of folks using their original setup with Dupont connectors though.

rstrouse, 2023-07-18:
I am going to close this for now.  If you have used a wire wrapping tool to make an ESPSomfy RTS device let us know.

closed 2023-07-18 by rstrouse

bruno-monk, 2023-07-18:
Thanks for the followup. Happy to report that the wire wrapping tool came in over the weekend and the wrapping with 30awg Kynar wire utilizing the 3D printed case in the wiki was a success.

rstrouse, 2023-07-18:
Awesome.  Can I get some pics?  Maybe we can include it as an option in the wiki.

bruno-monk, 2023-07-18:
Here are a few photos:
[image]
[image]
[image]
[image]

rstrouse, 2023-07-18:
That is what I call a lost artform.  I think I may open up the discussions section so folks can share their projects.  I really didn't realize how many folks wanted this repo.  The discussions section would allow folks to show their projects and help others.

bruno-monk, 2023-07-18:
Though I read contracts better than I sing, I sing praises for REM, nPC and ESPSomfy: thank you!

rstrouse, 2023-07-19:
It sounds like you have the entire gambit going on.   I have to say Nixie was one of those things I just did because I could.  Now it has more power than any of the Pentair panels.  Originally, I did it to fill the gaps that I felt were in my IntelliCenter but now there are an awful lot of Nixie implementations.  There were new schedule improvements just added as well.