rstrouse/ESPSomfy-RTS #94: I believe the CC1101 needs 5v instead of 3.3v

finder39, 2023-07-11:
So while connected to my computer's USB everything worked fine, but any time I tried to connect a usb powerline adapter (in USA) to the board it would boot up for a second then crash (though the red light would stay on).

It seems that connecting the VCC to 5v instead of 3v3 fixes this, though weird it also runs with VCC connected to nothing? I tried this after some googling and finding [this page](https://www.tindie.com/products/koral/external-cc1101-mini-gpio-board-for-flipper-zero/) which specified it needed 5v to run.

I'm using the [K0R41 1x CC1101](https://www.amazon.com/dp/B0C8RRG88V?psc=1&ref=ppx_yo2ov_dt_b_product_details), connected to [this ESP32](https://www.amazon.com/dp/B09Z7QRPXV?psc=1&ref=ppx_yo2ov_dt_b_product_details).

Anyways, I could be totally off on this. It's my first time working with esp32 or a CC1101 radio

rstrouse, 2023-07-11:
The max supply voltage of the cc1101 chip is 3.9 and I do not see a voltage regulator on the board.  I do not think that is correct.

finder39, 2023-07-11:
I think this was an issue with an bad esp32 board instead. I seem to be running into a different issue now that I changed boards, will do more testing and see if I need to create a new issue. Thanks!

closed 2023-07-11 by finder39