rstrouse/ESPSomfy-RTS #95: Rolling code gets reset to 1

finder39, 2023-07-12:
So I've set things up and everything works really great when connected to usb power on the computer. Ups, downs, and stops all work, and the rolling code changes with each action.

But as soon as I plug into a usb cube in the wall and try to work with things the first rolling code is correct (lets say 65), but then after that it resets to 1 and doesn't increment. If I try to set it manually to what it was, it still gets reset to 1 and doesn't increment.

It's a super strange issue and seems to happen regardless of the cube that I use.

I'm using the [K0R41 1x CC1101](https://www.amazon.com/dp/B0C8RRG88V?psc=1&ref=ppx_yo2ov_dt_b_product_details), connected to [this ESP32](https://www.amazon.com/dp/B09Z7QRPXV?psc=1&ref=ppx_yo2ov_dt_b_product_details).

While connected to usb on the computer i can watch the logs in real time and see the signals sent via esphome web, but I can't do that once its on a cube. So as a side it may also be good to have a way to read these logs in the web interface for it.

I'd love any thoughts on what could possible be happening. Thanks!

[image]

finder39, 2023-07-12:
Small update, once plugged back into the computer the rolling code still gets reset to 1 it seems. If there is a way to send logs or something I'm happy to

finder39, 2023-07-12:
Here is a screen recording of the issue happening

[image]

rstrouse, 2023-07-12:
At first blush it looks like a power issue.  Flash writes take the most power.  Change the cable as well.

finder39, 2023-07-12:
It seems that no matter what cube or cable I use the issue persists. I may just need to try a different brand of ESPHome devices. Maybe these ones have issues off of wall power somehow

finder39, 2023-07-12:
It seems that no matter what cube or cable I use the issue persists. I may just need to try a different brand of ESPHome devices. Maybe these ones have issues off of wall power somehow.

Whats strange is that when this issue happens, the only way to "fix" ESPSomfy is to connect the board to ESPHome web and be watching the logs. If I don't do that it will continue to have the Rolling Code be 1, even while connected to computer power.

rstrouse, 2023-07-12:
Wait a minute.  Did you load ESPHome firmware onto the device?  Unfortunately, the nvs reports no errors if it is full and if ESPHome is using up all the nvs then it won't persist the rolling code.  If you clicked the `Prepare for first use` button and installed that firmware then installed the two firmware files over top of it, then you need to re-install the firmware using these instructions.  The key is to ignore the 'Prepare for first use` button.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Installing-the-Firmware#using-esphome-web

finder39, 2023-07-12:
I don't believe that I did. I believe I only loaded ESPSomfy. I think if I did that it wouldn't work when plugged into the computer either, right?

rstrouse, 2023-07-12:
You would think that would be the case.  It is so odd as I haven't seen a situation like this.  Of course there are so many board variants out there that it is hard to say what is up all of them but in all the boards tried I haven't seen this weirdness.  

This definitely has a different pinout than the one in the wiki.  Are you sure you haven't connected the transceiver to the 5v pin.  It should be connected to 3v3.

finder39, 2023-07-12:
So since I created #94 and possibly harmed the other receiver I picked up a new one and used a different board from my 3 set to ensure that i was using all new materials and this issue persisted. Talking with a buddy who has more hardware knowledge than me (my specialty is software, I'm just starting to learn hardware), he believes that maybe the board I have doesn't like when there is no data connection (since that would be the main difference connecting to a wall outlet vs a computer.

I ordered [new different boards](https://www.amazon.com/gp/product/B09GK74F7N/ref=ppx_yo_dt_b_asin_image_o00_s00?ie=UTF8&psc=1)  that will arrive Friday and I will test with those and see if thats the case.

Are there specific boards that you've ordered on Amazon by chance?

rstrouse, 2023-07-12:
I purchased a lot of these but they look similar to the ones you referenced.  I did see were somebody posted that the usb resistor pictured is different than the one on the board (0).  This makes it goofy it the 3v3 pin is used.  I am a soft in the middle guy too and hardware is simply a necessary evil.

https://www.amazon.com/gp/product/B08246MCL5/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

finder39, 2023-07-12:
Interesting. They look pretty much identical. I'll order these and see.

I'm sure its either something weird with the boards I got or something I'm doing wrong haha. Its just a process of swapping out one part at a time and seeing what causes things to finally work.

Overall this project is amazing, so thank you! And thanks for being responsive. I'll keep this up to date as new info comes in

finder39, 2023-07-14:
I ported everything over to the new board (using the one you linked to) and it worked perfectly. I not longer have any issues with he rolling code resetting, and the commands work every time.

It looks like the issue was with those boards and them not liking only being plugged into power and not having a data connection.

Thanks for the fast response. Hopefully this helps someone else if they run into similar issues.

closed 2023-07-14 by finder39