rstrouse/ESPSomfy-RTS #98: settings iobroker mqtt

Chris86Xa, 2023-07-15:
hi,

i use the adapter in iobroker with mqtt. 
I see all roller shutter and the values change when they are used but when you change the direction manually in the iobroker, the roller shutters do not move.
Send values is activated.

What can I do, where is the mistake?

Thank you!

rstrouse, 2023-07-15:
The direction change is done through the set topic.  So the full path for the current direction value is
`<root topic>/shades/<shade id>/direction`  and the full path to set the direction is `<root topic>/shades/<shade id>/direction/set`

wutz-ESP, 2023-07-20:
command

> .../target/set 0
> .../target/set 100

work well

> .../directon/set 0
> .../directon/set 1
> .../directon/set -1

did not work

> auf SomfyHelga/shades/1/target/set 0
> zu SomfyHelga/shades/1/target/set 100
> 
> MY SomfyHelga/shades/1/directon/set 0
> AU SomfyHelga/shades/1/directon/set 1
> ZUF SomfyHelga/shades/1/directon/set -1`

<html>
<body>

10:05:58.507 | SENT | SomfyHelga/shade| 0
-- | -- | -- | --
10:06:02.963 | SENT | SomfyHelga/shades/1/directon/set | 0
10:06:11.460 | SENT | SomfyHelga/shades/1/target/set | 100
10:06:11.874 | client_00001066954ece8c | SomfyHelga/shades/1/shadeType | 3
10:06:11.917 | client_00001066954ece8c | SomfyHelga/shades/1/position | 0
10:06:11.934 | client_00001066954ece8c | SomfyHelga/shades/1/direction | 1
10:06:11.959 | client_00001066954ece8c | SomfyHelga/shades/1/target | 100
10:06:11.984 | client_00001066954ece8c | SomfyHelga/shades/1/remoteAddress | 970381
10:06:12.001 | client_00001066954ece8c | SomfyHelga/shades/1/lastRollingCode | 95
10:06:12.026 | client_00001066954ece8c | SomfyHelga/shades/1/mypos | -1
10:06:12.043 | client_00001066954ece8c | SomfyHelga/shades/1/tiltType | 0
10:06:12.060 | client_00001066954ece8c | SomfyHelga/shades/1/sunFlag | 0
10:06:12.077 | client_00001066954ece8c | SomfyHelga/shades/1/sunny | 0
10:06:12.094 | client_00001066954ece8c | SomfyHelga/shades/1/windy | 0
10:06:12.353 | client_00001066954ece8c | SomfyHelga/shades/1/shadeType | 3
10:06:12.372 | client_00001066954ece8c | SomfyHelga/shades/1/position | 1
10:06:12.405 | client_00001066954ece8c | SomfyHelga/shades/1/direction | 1
10:06:12.422 | client_00001066954ece8c | SomfyHelga/shades/1/target | 100
10:06:12.439 | client_00001066954ece8c | SomfyHelga/shades/1/remoteAddress | 970381
10:06:12.456 | client_00001066954ece8c | SomfyHelga/shades/1/lastRollingCode | 95
10:06:12.473 | client_00001066954ece8c | SomfyHelga/shades/1/mypos | -1
10:06:12.490 | client_00001066954ece8c | SomfyHelga/shades/1/tiltType | 0
10:06:12.507 | client_00001066954ece8c | SomfyHelga/shades/1/sunFlag | 0
10:06:12.523 | client_00001066954ece8c | SomfyHelga/shades/1/sunny | 0
10:06:12.540 | client_00001066954ece8c | SomfyHelga/shades/1/windy | 0
10:06:12.834 | client_00001066954ece8c | SomfyHelga/shades/1/shadeType | 3
10:06:12.853 | client_00001066954ece8c | SomfyHelga/shades/1/position | 2
10:06:12.885 | client_00001066954ece8c | SomfyHelga/shades/1/direction | 1
10:06:12.902 | client_00001066954ece8c | SomfyHelga/shades/1/target | 100
10:06:12.919 | client_00001066954ece8c | SomfyHelga/shades/1/remoteAddress | 970381
10:06:12.936 | client_00001066954ece8c | SomfyHelga/shades/1/lastRollingCode | 95
10:06:12.953 | client_00001066954ece8c | SomfyHelga/shades/1/mypos | -1
10:06:12.970 | client_00001066954ece8c | SomfyHelga/shades/1/tiltType | 0
10:06:12.987 | client_00001066954ece8c | SomfyHelga/shades/1/sunFlag | 0
10:06:13.004 | client_00001066954ece8c | SomfyHelga/shades/1/sunny | 0
10:06:13.021 | client_00001066954ece8c | SomfyHelga/shades/1/windy | 0
10:06:13.314 | client_00001066954ece8c | SomfyHelga/shades/1/shadeType | 3
10:06:13.333 | client_00001066954ece8c | SomfyHelga/shades/1/position | 3
10:06:13.366 | client_00001066954ece8c | SomfyHelga/shades/1/direction | 1
10:06:13.383 | client_00001066954ece8c | SomfyHelga/shades/1/target | 100
10:06:13.400 | client_00001066954ece8c | SomfyHelga/shades/1/remoteAddress | 970381
10:06:13.417 | client_00001066954ece8c | SomfyHelga/shades/1/lastRollingCode | 95
10:06:13.434 | client_00001066954ece8c | SomfyHelga/shades/1/mypos | -1
10:06:13.451 | client_00001066954ece8c | SomfyHelga/shades/1/tiltType | 0
10:06:13.468 | client_00001066954ece8c | SomfyHelga/shades/1/sunFlag | 0
10:06:13.485 | client_00001066954ece8c | SomfyHelga/shades/1/sunny | 0
10:06:13.502 | client_00001066954ece8c | SomfyHelga/shades/1/windy | 0
10:06:13.882 | client_00001066954ece8c | SomfyHelga/shades/1/shadeType | 3
10:06:13.901 | client_00001066954ece8c | SomfyHelga/shades/1/position | 3
10:06:13.918 | client_00001066954ece8c | SomfyHelga/shades/1/direction | 0
10:06:13.944 | client_00001066954ece8c | SomfyHelga/shades/1/target | 3
10:06:13.969 | client_00001066954ece8c | SomfyHelga/shades/1/remoteAddress | 970381
10:06:13.986 | client_00001066954ece8c | SomfyHelga/shades/1/lastRollingCode | 96
10:06:14.011 | client_00001066954ece8c | SomfyHelga/shades/1/mypos | -1
10:06:14.028 | client_00001066954ece8c | SomfyHelga/shades/1/tiltType | 0
10:06:14.045 | client_00001066954ece8c | SomfyHelga/shades/1/sunFlag | 0
10:06:14.061 | client_00001066954ece8c | SomfyHelga/shades/1/sunny | 0
10:06:14.078 | client_00001066954ece8c | SomfyHelga/shades/1/windy | 0
...

10:08:26.740 | SENT | SomfyHelga/shades/1/target/set | 0
-- | -- | -- | --
10:08:27.152 | client_00001066954ece8c | SomfyHelga/shades/1/shadeType | 3
10:08:27.171 | client_00001066954ece8c | SomfyHelga/shades/1/position | 20
10:08:27.189 | client_00001066954ece8c | SomfyHelga/shades/1/direction | -1
10:08:27.214 | client_00001066954ece8c | SomfyHelga/shades/1/target | 0
10:08:27.239 | client_00001066954ece8c | SomfyHelga/shades/1/remoteAddress | 970381
10:08:27.256 | client_00001066954ece8c | SomfyHelga/shades/1/lastRollingCode | 98
10:08:27.281 | client_00001066954ece8c | SomfyHelga/shades/1/mypos | -1
10:08:27.298 | client_00001066954ece8c | SomfyHelga/shades/1/tiltType | 0
10:08:27.315 | client_00001066954ece8c | SomfyHelga/shades/1/sunFlag | 0
10:08:27.332 | client_00001066954ece8c | SomfyHelga/shades/1/sunny | 0
10:08:27.349 | client_00001066954ece8c | SomfyHelga/shades/1/windy | 0
10:08:27.509 | client_00001066954ece8c | SomfyHelga/shades/1/shadeType | 3
10:08:27.528 | client_00001066954ece8c | SomfyHelga/shades/1/position | 19
10:08:27.561 | client_00001066954ece8c | SomfyHelga/shades/1/direction | -1
10:08:27.578 | client_00001066954ece8c | SomfyHelga/shades/1/target | 0

</body>
</html>

rstrouse, 2023-07-20:
I'm sorry I do not understand what works and what doesn't from your post.  Are you saying `.../target/set` works but `.../direction/set` does not?

wutz-ESP, 2023-07-21:
Yes hat is the problem.
SomfyHelga/shades/1/target/set 0 works
SomfyHelga/shades/1/direction -1 not

SomfyHelga/shades/1/direction/set -1 must bei right. 
I will give it a try today - sorry, i am already learning

Chris86Xa, 2023-07-21:
Do you use iobroker?
The negative value must be set separately in the settings!

wutz-ESP, 2023-07-21:
I am using MQTT2_Server of fhem.

If my broker send
SomfyHelga/shades/1/target/set 100 - awning ist driving down
SomfyHelga/shades/1/target/set 0 - awning ist driving up 

My problem is if i want to stop this device sending
SomfyHelga/shades/1/direction/set 0 - does not work
If i press MY on WEB-Interface ist stops immediately

SomfyHelga/shades/1/direction/set 1 - does not work
SomfyHelga/shades/1/direction/set -1 - does not work
That is the reason i am thinking there ist something wrong with command direction

Chris86Xa, 2023-07-21:
can you limit the values ​​(-1, 0 , 1) in the data point as possible choices?
I had to set this in iobroker mqtt, only then did the roller shutters move...
[image]

wutz-ESP, 2023-07-21:
in MQTT.cpp i find this 

line 107 single shade - **else if**

[image]

line 133 group shade - **if**

[image]

could this be the problem?

rstrouse, 2023-07-21:
I use MQTT explorer with a mosquitto instance and it all works perfectly.  Bear in mind ESPSomfy RTS never publishes to a `set` leaf.  It only subscribes to it.  And if you publish a value to its descendant you will get no response.  In fact that response will be overwritten on the next go around.

The above code is correct.  Is it possible to look at your broker in MQTT explorer to see what is actually being written to the leaf?

wutz-ESP, 2023-07-22:
I use MQTT2_Server - MQTT explorer show 
[image]

[image]

rstrouse, 2023-07-22:
That is very odd that in MQTT explorer I am not seeing values on your parent leaf.  It is like something is publishing null to them.  Notice my MQTT explorer below.  Here are the differences.
* The intermediate nodes have values but something has cleared these. for instance the topic `../shades` has index values that are set at startup to allow you to enumerate all the shade ids in the subsequent topic.  These may not always be 1,2,3,4 ... Since motors can be deleted out of the middle.
* You are publishing the set command to a misspelled topic.  You are publishing `SomfyHelga/shades/1/directon/set` when you should be publishing to `SomfyHelga/shades/1/direction/set` I suspect this is your problem.  You forgot the `i` in direction.

[image]

wutz-ESP, 2023-07-22:
Sorry, making trouble.
I will try on monday

rstrouse, 2023-07-22:
No worries sometimes it takes another eye to catch things like that.

wutz-ESP, 2023-07-24:
installed 2.1.1 and tried direct**i**on worked perfect.

rstrouse, 2023-07-24:
I am going to close this for now.  Let me know if you have other issues.