rstrouse/ESPSomfy-RTS #99: ESP Somfy does not seem to work with my newly installed shade as it uses a 868 Mhz.

InfriaAB, 2023-07-15:
Hi!
I got a vertical awning installed on the house a week ago and have been trying to get ESPSomfy to work with it since then to no avail. I have not even been able to get anything to show up in the transceiver log at all.

I have gone over the wiring several times and I'm quite certain that is correct.

When diving deeper into the techical specs I think I may have found the reason. They have changed the frequency, or at least the model I have seems to be using 868 Mhz. This datasheet shows that the remote I have does not use 433Mhz. 
[image]

https://service.somfy.com/downloads/eg_v5/doc149192002_tdst_-_situo_1_-_5_variation_io.pdf

So, am I screwed or is there a way to use this awesome project with my awning?

Here are some pictures that can possibly help with some more information.
[image]
[image]
[image]
[image]

InfriaAB, 2023-07-15:
Found an older issue with the same problem. So my guess is I have to spend the 300USD(!!) to buy their hub if I want to automate.
https://github.com/rstrouse/ESPSomfy-RTS/issues/9

rstrouse, 2023-07-15:
Yeah I would buy the clones.   Closed protocol and cloud implementation is not for me.  But then again if I had them I probably would have rooted it by now.

Johan-21, 2023-07-16:
Dear RStrouse,

What an excellent described project. I have no technical background but as retired person blessed with much free time and love to watch YT video’s.

We are going to order 2 shutters and 1 screen with Somfy IO remote protocol connectivity.
I will simultaneously buy a Raspberry Pi 4 to install Home Assistant.
( Experienced a bit with HA on my FriendlyElec R2R already ).

As said in the intro this GitHub project is excellent described and I read on the Somfy website that IO use 868-870 MHz protocol.
Amazon has also 868Mhz transceivers.

So my newby question is  : Is the process of the 868Mhz vs 433Mhz complete different why you would not even consider investigating to adapt this to the new IO protocol?

pS: The IO remote doesn’t obviously make a connection with the cloud so the only difference is the bandwidth of the signal.
( again speaking by a newby ).

Johan

rstrouse, 2023-07-16:
@Johan-21,

I have looked at the Somfy IO protocol but I do not have any of these motors.  While the frequencies between RTS and IO are different, the differences do not end there.  It isn't enough to simply tune the radio to 868MHz and send the same frames over a different carrier.  

The io homecontrol protocol uses a 2FSK modulation so there are actually 3 frequencies involved.  Even that would be a minor bump to overcome.  The bit decoding while different is in fact a bit simpler than the bit decoding for the ASK/OOK modulation used for the RTS protocol.  But alas, sending the same RTS frame over this different modulation will also not be successful.  The command structure has been reworked and the frames are encrypted.  I read somewhere that it is encrypted with AES-128 but event that is speculation.

I, am confident that I could get past the coding for that even without examples of the equipment.  But that isn't the primary issue with coming up with a decoding strategy.  The bigger issue is that it appears you have to be invited to the party to get access to the APIs and description of the encoded data.  The invitation includes the security keys needed to decrypt the packets.

Since my invitation was lost in the mail I guess my only option would be to crash the party, but at this time I have other plans.  If I had shades that communicated over IO homecontrol, then this project would have focused there since my fedup-O-meter was maxed out.  The RTS folks would be asking me to support RTS and the problems would focus around the weird frame preambles that ESPSomfy RTS accommodates today.

In fact I have a need for additional motors, but put that purchase on hold because I could not find an implementation that could hardwire these.  While I am anti-cloud device, I am equally as anti-battery when it come to things that just need to work in my home.  I would rather go without, than hit a button and be disappointed.  In the end I will probably run 24v to the motors and control them via RS485 to RTS.  

From what I have seen of IO homecontrol the compelling bits have too many warts for me at the moment.  At this point in time, IO homecontrol claims to be "So open" (see Somfy "so open" commitment) while doing everything it can to not be inclusive.   While half of this statement was true with RTS, there was never a claim to be open or closed,  but there was also no effort to actively work to keep end users out.  

Apparently, "So open" is a term that applies to the observer, none of which are the users of their products.  If you are a hub maker and willing to do a licensing deal, then perhaps you can join the party.  It will be BYOB but you don't need to bring your own glassware.  Alas, I am just a guy who wants control of his own damn shades.

What I have seen so far of the hub devices, but have no idea to their effectiveness or their cloudiness:
* TaHoma
* Velux KLF
* CozyTouch

Here is some work that involves an approach to digging into the radio protocol but as you can see it is very early and will take folks with actual hardware and a maxed out fedup-O-meter.
https://github.com/Velocet/iown-homecontrol

So hopefully my TLDR; gives you some insight into why this project does not include IO homecontrol.  Heck the information may not even be entirely accurate but it is not necessarily what I know, but what I see (again the observer thing).  If things change or somebody with the hardware has worked out some code for it, I would happily abstract the comms in the project to digest it and regurgitate it.  In fact the current CC1101 transceiver will happily speak and listen at the required frequencies.

Johan-21, 2023-07-18:
@rstrouse,

thanks.
A long but impressive and clear answer.

I found further information why cracking the IO protocol isn’t that simple.  ( see below ).

And to be honest , to compare RTS vs IO the ‘two way communication” option is the only difference. That is only interesting in those rare cases ( e.g. frost etc ) that I can be handy to know the exact position of a shutter .

The official Somfy connectivity Kit also works with RTS but it is great fun to build an own copy of yours ( rtrouse ) ESP32 solution.

soldering a few cables , and flashing the ESP32 is a piece of cake.
Unfortunately there is a backlog in delivery of shutters / screens ( at least that’s what our dealer told us ) .

So most likely per October 23 can I start testing.

= = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = =
Sorry the below text is in German. No problem for me but maybe not the case for everyone.
= = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = =
Ref.:  https://github.com/Velocet/iown-homecontrol

Bei der ersten Benutzung tauschen Fernbedienung (Sender) und Produkt (Empfänger) einen 128-Bit-Verschlüsselungscode aus und verbinden ihn bei jeder neuen Aktion mit einer zufällig gewählten Zahl. Daraus errechnet sich ein Code, den Sender und Empfänger miteinander abgleichen. Nur bei Übereinstimmung reagiert das Produkt auf die geforderte Aktion. Durch diesen Sicherheitsmechanismus wird die Reaktion auf einen fremden Sender ausgeschlossen. Die neue Anwendung sucht automatisch nach bereits bestehenden Produkten und berücksichtigt diese bei ihren Aktionen. Bevor Sender und Empfänger miteinander kommunizieren, wird die Verfügbarkeit des Kanals überprüft. Sollte eine Bandbreite überlagert oder besetzt sein, wartet die Anwendung vor der Kommunikation auf das Freiwerden der Frequenz (Listen before Talk) oder weicht auf einen der anderen Kanäle aus (Adaptive Frequency Agility).

rstrouse, 2023-07-18:
Here are some links to some 3d printed cases.  
https://github.com/rstrouse/ESPSomfy-RTS/wiki/Simple-ESPSomfy-RTS-device#3d-printed-cases

rstrouse, 2023-07-21:
I am closing this as ESPSomfy RTS does not currently support io-homecontrol.

closed 2023-07-21 by rstrouse