rstrouse/ESPSomfy-RTS #101: Can't Pair Shade!

KartmanCanada, 2023-07-18:
Firmware loaded and WiFi config was no issues... I specifically looked for the blue recommended transceivers. I configured as shown and my dBm numbers are very low (like -50dBm with remote directly beside the antenna) when comparing to your well documented examples. That said, I am able to see frames from my existing Somfy remote in the log...

When I follow the pairing instructions, my existing remote puts the blind into pairing mode (jog) but when I click the ESPSomfy 'PROG' button, my blind never jogs in acknowledgement. Eventually, the blind will jog on it's own but I think that is the just pairing mode timing out.

I've double checked the wiring. Is it possible I've just gotten garbage trancseivers? I bought 2x and both seem to be operating the same way. Any suggestions would be appreciated.

rstrouse, 2023-07-18:
Some motors will jog on their own to indicate that the pairing memory has closed.  Please swap the settings you have for RX and TX in the software, then save the radio.  In most cases this is because GDO0 on the transceiver needs to be connected to the pin mapped to TX on the ESP32.

KartmanCanada, 2023-07-18:
swapped... that seemed to work. Thanks.

Pin3 (GDO0) <=> GPIO13 on ESP32
Pin7 (GDO2) <=> GPIO12 on ESP32

Great project, BTW.

rstrouse, 2023-07-18:
Glad you got it working.  Have fun with your new gadget!

closed 2023-07-18 by rstrouse