rstrouse/ESPSomfy-RTS #102: No AP broadcast after flashing

steveomg, 2023-07-19:
I have a ESP32-WROOM-32 and after succesful flashing with no errors I never see the AP broadcast.  Anyone else have a similar issue?

steveomg, 2023-07-19:
Looks like its an issue with the esp board itself.  Will post back if/when I figure out whats going on.

whc-commits, 2023-07-19:
So VIN pin on this board will not supply enough power applying 3.3V.  VIN will need 5 volts.

steveomg, 2023-07-19:
Looks like its an issue with the esp board itself.  Will post back if/when I figure out whats going on.
******
Pin VIN needs 5V. 3.3V will show power indicated, but will not sufficiently power my board variant.

closed 2023-07-19 by steveomg

rstrouse, 2023-07-19:
That is correct if you are powering from the 5v (VIN) pin on any of the ESP32 variants.  Just don't apply 5v to the 3v3 pin.