rstrouse/ESPSomfy-RTS #106: I am having this same issue.  I have verified wiring dozens of times.  I have tried multiple hardware units.  I have tried sliding the Rx bandwidth to 250.  I am unable to get anything to print in the logs.

steveomg, 2023-07-20:
I am having this same issue.  I have verified wiring dozens of times.  I have tried multiple hardware units.  I have tried sliding the Rx bandwidth to 250.  I am unable to get anything to print in the logs.  

My remote is a somfy Telis 4 RTS US Patio.  White with blue edges.  Could the frequency be different?

_Originally posted by @steveomg in https://github.com/rstrouse/ESPSomfy-RTS/issues/51#issuecomment-1642923391_

rstrouse, 2023-07-20:
Look at your remote.  Is this an RTS remote or an IO remote.  This project is for RTS remotes.

steveomg, 2023-07-20:
Says RTS.  

Going to order new antennas.  I think these $2 for 2 are not working.

rstrouse, 2023-07-20:
> Says RTS
Awesome I was hoping it was not IO.

Lets also look at your RX and TX wiring.  Can you shoot a couple of pics.  If there is no center pin in the antenna (female) then it is an RP-SMA antenna and will not work.  Also if it is tuned for some frequency other than 433MHz then it is good as a tickle stick but nothing else.

steveomg, 2023-07-20:
RX and TX are valid.  I have verified 100 times and also swapped at least that many times. I am about to order more antennas and will give that a whirl.

rstrouse, 2023-07-20:
Make sure you order SMA (not RP-SMA) antennas tuned to 433MHz.

BTW: A tickle stick is used to coerce gulf lobster out of their holes.

steveomg, 2023-07-20:
Grabbed this one
NOYITO 433MHz Antenna 5dBi Gain Omnidirectional Antenna 430-470MHz SMA Connector - Internal Thread Inner Needle https://a.co/d/5ZiNJxX

More used to swamp lobsters over here.

rstrouse, 2023-07-20:
Mud bugs.  Delicious. 

That looks like a fine antenna according to the specs.   Normally the transceiver should pick up some signal with the remote next to it.   Try setting your RX and TX in the config to the same pin.   Don’t change your wiring just set them both to where the GDO0 is connected to.

steveomg, 2023-07-21:
Antenna was the issue.  This one is working fine, though not quite 'tuned'.

closed 2023-07-21 by steveomg