rstrouse/ESPSomfy-RTS #108: Can't login by password

blastrock, 2023-07-20:
I have set up the password authentication on espsomfyrts, but I can't get past the login form.

Looking at the code, it goes through some PIN specific code:

```
        let pin = '';
        for (let i = 0; i < 4; i++) {
            pin += sec.pin[`d${i}`];
        }
        if (pin.length !== 4) return;
```

`pin.length` is 0 there, so nothing happens when I click login.

011V32, 2023-07-20:
I have the same problem :(

rstrouse, 2023-07-20:
Uggh.  This plugged a ui error and painted into the corner.

Please run the following url from your browser.
`http://<ip address of espsomfyrts>/backup`

And save this file.  I am rebuilding an onboard file now.

blastrock, 2023-07-20:
As a workaround, commenting the `if (pin.length !== 4)` line in the dev console allows me to log in :)

rstrouse, 2023-07-20:
That works too!

rstrouse, 2023-07-20:
v2.0.3 no longer has this issue.

011V32, 2023-07-21:
I can confirm that it now works. 👍

I would have 2 suggestions for improvement:
1. It would be great if a key-enter would also work for login. (Currently you can only get in if you press the login button with the mouse).
2. When you reload the page you should stay logged in. Currently you are logged out after each refresh.

rstrouse, 2023-07-21:
v2.1.0 will process the enter key.  ESPSomfy RTS cannot maintain a session with the device.  The webserver on ESP32 is definitely not that robust.  A cookie could store the key hash locally but that creates even more headaches.

rstrouse, 2023-07-21:
v2.1.0 has been released.

closed 2023-07-21 by rstrouse

011V32, 2023-07-22:
The "normal" Enter works, but the enter from the Num-Block doesn't 🙈

rstrouse, 2023-07-22:
Hmmm different keyboard mapping.  Uggh.