rstrouse/ESPSomfy-RTS #110: Backup of all settings

011V32, 2023-07-21:
enhancement

Hello.

It would be great if all settings could be saved in the backup.
So not only those of somfy (shades/groups), but also those of the system, network, radio, ...
Thank you.

Regards

rstrouse, 2023-07-21:
This one is kind of dicey.  Indiscriminately changing the pin assignments especially with an ethernet board can prove to be bad news.  Incorrect settings on the PHY can destroy the board.  Also, importing network settings after having hooked up your network might prove to be a bit annoying.  

Timezones, mqtt, and even frequency/protocol settings are pretty innocuous so maybe there is a balance.

011V32, 2023-07-22:
How about a dialog where you can then select what can be restored?
(The "enable radio" needn't be set.)

rstrouse, 2023-09-15:
New functionality for backup and restore coming in the 2.1.6 release.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Configuring-the-Software#backup-and-restore

rstrouse, 2023-09-16:
v2.1.6 has been released and has the ability to backup and restore nearly all of the settings.  Any potential security items are not contained in the backup file.

closed 2023-09-16 by rstrouse