rstrouse/ESPSomfy-RTS #112: Timeouts - Dropout of service, board not reachable

jabums, 2023-07-21:
Thanks for sharing this project (and also for that very useful documentation!)

My installation seems to be working fine in general. I can remotely open and close my blinds via the web interface and also from Home Assistant. Pressing a button on the physical remote is also being registered.

Unfortunately my setup seems to not be very reliable. When I woke up the next day the service would not work any more. accessing the web interface would give a timeout, HA buttons disabled due to device not being available.
I hit refresh a few times in my browser and after a few minutes the web interface would show up again.

This is the hardware that I use:
AZDelivery ESP32 Dev Kit ( ESP32-WROOM-32 )
EBYTE E07-M1101D

Wifi Reception is not great in the space (kitchen), other wifi based appliances and devices seem to work okay though (HomePod, smartphone, thermostat, weather station)

As this is my first ESP project I am not sure how to access any log files.
When running ping from my computer to the ESP32 for the last couple of hours I would randomly get a timeout, but the web Interface seems a lot more reliable. (showing up any time I would try)
Wifi reception is only at -72DBm as displayed in the Web panel.

Further research led me to a few more possible reasonings
- power supply: I was first using an old AC to USB plug from an older smartphone. Then switched to a proper power supply from an old Nokia. The board does no longer seem to reboot randomly now.
- power save mode: Is there a power saving mode I can disable in this firmware? Here is where I found [https://esphome.io/guides/faq.html#my-node-keeps-reconnecting-randomly](information about power saving mode sometimes being an issue). If possible I would like to set:

`wifi:
  # ...
  power_save_mode: none`

- in case this is simply related to poor wifi reception maybe I should switch to board with ethernet? I really do not understand at this point what a serial port is a whether I might need to learn and/or buy further equipment in order to get a WT32-ETH01 based version up a running.

rstrouse, 2023-07-21:
A couple of things.  
* Did you reserve the IP address for the ESPSomfy RTS device in your router for DHCP?
* ESPSomfy RTS disables sleep mode in the firmware there is no need to do this.
* -72dBm is not great for wifi reception but I suspect that it just drops off occasionally.  If it remained at -72dBm then it would be fine.  If the ESPSomfy RTS device is near anything that can generate interference move it.  You might be surprised what a couple of feet can do.
* A WT32-ETH01 board requires an external USB to TTL serial adapter that connects to the pins on the board.  I have no idea why it is not simply soldered on the board.  You cannot load the firmware without it and it is the most likely power source.
* Go into the network tab and save the adapter after you have set up your DHCP IP reservation in your router.  This will force the ESPSomfy RTS to resend the mDNS packets.  Next go into the Options tab, check your timeszone, uncheck broadcast uPnP over SSDP if you have it checked and press Save.  This will force the NTP and SSDP server processes to update themselves.
* Sometimes disabling 802.11b in your router (you probably do not have any of these devices anyway) this protocol does all kinds of handshaking with everything connected since it needs to probe the connection whether it is connected that way or not.  If you have a lot of wifi devices this protocol is a menace.
* If the board was rebooting on that power supply you were definitely underpowering it.  While the symptom is a reboot, an underpowered ESP32 will often not properly write to the flash.  If you have not done this please update your firmware and application with the improved power supply connected.

jabums, 2023-07-22:
Thanks rstrouse!

I have checked all aspects you mentioned.
- IP reservation is set in DHCP (router)
- timezone is set correctly, unchecked broadcast uPnP over SSDP
- updated firmware and application to latest build
- tried different placements for the device. (only limited options in my kitchen)

While Wifi reception ist still poor I have gotten no dropouts of the service in the last several hours. fingers crossed. (I occasionally open and close the blinds via smartphone for testing purposes)
The setups needs to work reliably before I introduce it to the family :)

rstrouse, 2023-07-22:
Lets cross our fingers!

I have plenty of coffee at the moment.  If I could only figure out the a "Buy me some time" button.  I need a little bit of the 4th dimension.

rstrouse, 2023-07-29:
I am going to close this for now.  If you have further issues re-open this one or start another.

closed 2023-07-29 by rstrouse

jabums, 2023-07-29:
Using the settings you provided controlling my blinds has been working fine for several days now.
I can unfortunately not say what setting exactly did the trick. Wi-Fi Signal is still very low, but does not matter so much.
Thanks again!