rstrouse/ESPSomfy-RTS #113: Retransmit the command to ensure the motor has received it

SteenVestergaardLarsen, 2023-07-23:
I have noticed several times that the motors do not always accept the command send to them. I have two curtains that are running in parallel and some time only one of them reacts on the command. The curtains is located 50 cm apart.
It is random which one that reacts. I see the problem on all my curtains, but it is most noticeable on the two that runs in parallel.
From what I see the problem is biggest when the curtains has not been used for a while.
I have also see the problem when using the Somfy remote. 
It is the Somfy Cord lift 45WF I am using and with the latest FW/HW version that is available at current time.

When I used the TaHoma Switch I did not notice this problems (Only used for a month until I found this cool project that meets my reequipments better).
I currently have integrated the ESPSomfy with Loxone and a LoxBerry setup.
I have been using 1.7.0 for a while and just upgraded to 2.1.0 where I see the same problem.

rstrouse, 2023-07-23:
Currrently, it sends 1 command + a repeat.  I wonder if the switch sends more repeats.  Let me think about this for a bit as the software is set up to send repeats but the number of repeats are not currently settable.  This is because the number of repeats determines the long, mid, and short press functions of the protocol.

It is weird since it seems like the motors are potentially slow to wake up.  ESPSomfy RTS sends the proper wakeup pulse but perhaps they are lazy and have a bit of trouble getting out of bed.  Let's make sure this is the problem though before we try solving it this way.   Adding more repeats will get us dangerously close to mid press territory.  First, make sure your batteries are fully charged and sources of interference are mitigated.

rstrouse, 2023-07-23:
If you load the firmware located in the main repository SomfyController.ino.esp32.bin and SomfyController.littlefs.bin, you can select the number of repeats that will be sent for each motor.  Try this out and see if it alleviates the issue.

Out of curiosity when you say running in parallel are these grouped?   The best way to run these are to create and pair them as separate motors then use the grouping function to group them.

SteenVestergaardLarsen, 2023-07-24:
Just a quick update 1 repeat did not fix the problem. Increasing to 2....

rstrouse, 2023-07-24:
> Regarding your question of "running in parallel ". Perhaps I explained it a bit wrong. I have two motors that are parried to the same button on the Somfy remote. This way they should reach the same way when the button is pressed. Make sense?

This is a group where two motors operate at the same time from one remote address.   There are a limited number of channels on any given remote so installers sometimes double them up without first dedicating a channel to them.  With ESPSomfy RTS you have 32 of such channels so you are not so limited.  This can allow you to have the motors operate individually then group them together so they operate together.

The default has always been 1 repeat.

SteenVestergaardLarsen, 2023-07-25:
Result of test with 2 repeats. 
I have not had any problems controlling the up/down and stop command. So it seams more stable. The test will continue... 

To test the outer limits I can see that when setting the repeat to 4 the button press is seen as a long press and the motor only moves a bit.

When playing with the number of repeats I have seen that using the "position/set" command is not working if I set repeat to 2 (The motor only move a bit like described above).
This to me indicate that for some reason the "position/set" command trigger the repeat to be double up so it ends up with 4 repeats.

rstrouse, 2023-07-25:
> This to me indicate that for some reason the "position/set" command trigger the repeat to be double up so it ends up with 4 repeats.

I checked and the position/set is not sending multiple commands like that.  Are you doing this from Home Assistant or from the ESPSomfy RTS app and is this a group that you have set up?

SteenVestergaardLarsen, 2023-07-27:
I do it from my Loxone system. 
I can see that Loxone send the same command two times with no delay to the MQTT. 
It sesames that sending the same command two times trigger that the RF command is send two times. 
Do this make sense?
Just for fun I tried to send a invalid MQTT command "shades/3/target/set m" that trigger the software to move to position 0.

rstrouse, 2023-07-28:
If you are sending commands through MQTT you should send the command to that only once.  QoS takes care of that delivery.   The frames are repeated on the radio.

I you are moving to 0 then that means you have the favorite set to 0 and if the value is < 0 then up if > 0 then down.  Any other value is my or stop depending on the motion of the motor.

SteenVestergaardLarsen, 2023-07-29:
I have change the way Loxone is sending the command and it is all working as expected.
Also the setting with 2 repeat is a keeper for me as the motor has not failed  a single time since I change to 2. Thanks for fast respons.

rstrouse, 2023-07-29:
Awesome!  I will close this for now.  The changes to allow repeats on the motors have been integrated into the repo and released.

closed 2023-07-29 by rstrouse