rstrouse/ESPSomfy-RTS #115: Unable to set up groups

Ay1tsMe, 2023-07-24:
My hardware is: CC1101 and an esp32.

I am able to successfully connect all the blinds in my house with the integration. All shades operate as expected and I have set up home assistant as well so I can operate my blinds within my smart home.

However, on the web app, I am unable to set up groups. When I navigate to the group tab and click `open memory`, the blinds don't jolt as they should. My solution was to add helper groups in home assistant that way I can have groups for my upstairs and downstairs blinds but there is a problem to this. One of them being that when I close the downstairs group, sometimes it also starts closing other random blinds in my house. I suspect this is because the transmitter is getting clogged up with signals and is failing to close the correct blinds.

I also tried adding a group as a blind as my RTS remote has groups already assigned to certain blinds but I noticed that if I do this, then on the web gui, the group will say closed but the blinds associated with the group say open because the individual blinds are also setup in the web gui.

So I think, the solution to my problems would be to properly set up groups in ESPSomfy-RTS but I'm having issues. Any suggestions would be much appreciated.

Thank you.

rstrouse, 2023-07-24:
Try this.  Use your remote to get the memory to open.  Try to open the memory using ESPSomfy RTS.  When the confirmation screen comes up if the motor did not jog use your remote to make it do so at this time then answer yes that it jogged.  Continue on the process and you motor will be linked to the group.

rstrouse, 2023-07-24:
Yes that would be the process.  The long press means different things to different remotes.  Unfortunately if you hold too long it wont work and if you don't hold long enough it won't work.

Let ESPSomfy RTS try to open it.  And if it can't use the remote to make it do it.  The last part is the important bit where ESPSomfy RTS writes its group address into the motor.

Ay1tsMe, 2023-07-25:
Okay that worked. I am able to pair groups now. There is one problem though. On my Somfy remote, I have a blind group that i can navigate to and that operates all the downstairs blinds. If I operate this group on the remote, the individual blinds on ESPSomfy RTS don't detect whether they are going down or up.

I'm assuming this is because i havn't added the group that is on the remote to ESPSomfy. I did originally try adding this group as a single blind which works but then the other individual blinds that are apart of that group are not synced.

What's a work around this? Is there anyway to operate that blind group on my remote while also having the ability to operate it through ESPSomfy? Or will I just have to have one but not the other. 

Just to clarify, the problem is if I operate the blind group on the remote, it is not synced with ESPSomfy. And if I add this blind group as a blind. It doesn't sync with the other individual blinds apart of that group.

Ay1tsMe, 2023-07-25:
Just a follow. Once I added the group entities into home assistant, if i set a percentage of a blind group, all the other blinds in my house also go to that percentage. If i operate the groups with just up down and stop then it works as intended. Only when i set a percentage of the blind group it starts to do weird behaviour. 

It is strange because ESPSomfy does not recognise that it is closing other blinds in my house, so for example: If I set my upstairs group to 50%, the downstairs group will also go to 50% but ESPSomfy still thinks they are open. I think this might be a issue with the home assistant wrapper, I'm unsure.

rstrouse, 2023-07-25:
Each channel on each remote that can move the the motor in any direction should be linked to the defined shade in ESPSomfy RTS.  If you do that then ESPSomfy RTS will know the position.  They should always be in synch no matter how they have been moved.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Configuring-the-Software#linking-remotes-and-sensors

As for the weird group behavior from Home Assistant I will have a look.  You should not have to do anything special like create manual group entities or automations in Home Assistant.  ESPSomfy RTS will create these.

Ay1tsMe, 2023-07-26:
Attached is what the weird behaviour looks like. If I operate my upstairs group with up down or stop then it works correctly and only operates my upstairs. However, If i set a percentage position then it starts to operate the downstairs blinds that are not assigned to the upstairs group. The video is taken when I am downstairs. This only happens if I set a percentage which is weird. I have not created any helper groups, I am simply using the group entities that were created when the integration is linked to home assistant.

https://picoshare.tuxtower.net/-wEJwhCW9kq

My suspicion as to why this is happening might be due to the fact that if you set a percentage position on a group, it fires off every individual blind one by one rather than activating the group at the same time. I'm assuming this is done to get the correct position for each blind due to multiple blinds in groups being different up and down times. So because it's sending off multiple transmittions, it's also effecting the behaviour of the other blinds. Maybe a longer delay between each blind might help it but I'm not sure.

Also I fixed the sync issue by doing as you said and pairing the group option on the remote to each individual blind. :)

rstrouse, 2023-07-26:
Nope that’s not it.  Check the group state in developer tools.  It should list all the entities it thinks is part of the group.  Also make a backup and post it here.

Ay1tsMe, 2023-07-26:
I checked the group state and only the downstairs entities are apart of the group as intended. When operating the upstairs group, the state of all the blinds associated with that group aswell as the group state changes as intended. But the downstairs group still operates when a percentage position is set on a group.

Here is a screenshot that shows the blinds that are apart of the downstairs entity and the upstairs entity:
[image]

[image]

And here is a backup of my ESPSomfyRTS:
https://picoshare.tuxtower.net/-DSRZHK9c3r

Ay1tsMe, 2023-07-27:
Okay I found the problem. The lounge Blackout 1 individual blind is operating all of downstairs for some reason. I tried solving this by unpairing it from all my groups and now my groups don't operate downstairs which is good. Then I deleted it and repaired it to espsomfy but the issue still persists. The lounge Blackout 1 still operates all my downstairs blinds. Is there anything else I need to do? Do I need to repair all my downstairs blinds aswell?

rstrouse, 2023-07-27:
When you have an address on ESPSomfy RTS that controls a motor you do not want it to, you need to take a remote that controls just that motor.  Then hold in the prog button for that motor until it jogs only once.  Then you need to go into the shade definition on ESPSomfy RTS and click the unpair shade button. Click/tap the prog button on this screen until the motor jogs exactly once.  Exit the unpairing screen and verify that the ESPSomfy RTS no longer moves the motor with that address.

Ay1tsMe, 2023-07-27:
I'm sorry this is so back and fourth.

I'll just explain my setup so its a bit easier to understand. On my downstairs floor, I have a RTS remote that has a option for every individual blind and a group option for all the blinds. On the upstairs floor, I have another RTS remote that controls all the upstairs blinds individually aswell as some groups for each of my rooms. 

So you are saying for my case, I need to go to each individual downstairs blinds on espsomfy RTS and unpair the shade with the remote from the upstairs floor? I'm just not understanding what exactly I need to unpair.

I don't know where the conflict is and why it's operating that one upstairs blind with the downstairs group. I havnt programmed the same blind twice so I'm not sure.

Sorry if this is a tedious conversation

rstrouse, 2023-07-27:
Lets start with an explanation of how this works.  Any remote channel can be a group and this even includes ESPSomfy RTS channels.  In the ESPSomfy RTS device it is like having 48 channels.  In the Somfy it would be like a remote that you can select one of 48 channels.

Each channel on the remote is assigned an address so if it is a 5 channel remote then it will get 5 addresses.  Addresses are nothing more than a unique number within the area of all the motors within range.  I have read reports that Somfy has reused address numbers over time but have done this in different regions.  

Typically this number is assigned at the factory for each remote that is produced.  Motors do not have an address but they do have a small memory location where addresses from remotes are stored.  So while there is no intelligence in a Somfy remote channel there is also no address initially assigned to the motor.   

When the motors are initially shipped and after they have been factory reset.  The will start waiting for the first remote when they are powered on.  The first remote that sends it a prog command will get its address stored in the motor memory then the user can set the end stops.

There are up to 12 memory locations for remote addresses in each motor.  So when you long press on the prog button of a remote that has been paired to the motor it will open the memory to receive a prog command from a remote.  This is indicated by a short up and down movement of the motor.  The time the memory remains open varies between motors but eventually it will close the memory on its own.

If a remote sends a short press prog button while the memory is open, it will first look to see if it has this address stored and if it does it will remove it from its memory.  If it does not it will store it.  So if you long press the prog button on a remote that is paired to two motors both of them will open their memory.  

If the remote channel is currently stored in one of the motors and not the other it will pair one and unpair the other.  This is why Somfy recommends removing power from all other shades when doing this process.  But this is only because it is easier to tell them to do that than to explain why the consumer cannot get all their shades to pair properly.

So when a remote channel either from ESPSomfy RTS or a torpedo sends a command all it is doing is sending out a series of pulses that translate into a frame that contains its address.  Every motor within range will hear this frame, decode it, and decide whether it should respond.  So if the motor has the address in its memory then it will move.  If it doesn't then it ignores the frame.

In the end if you press a button in any remote and it moves a motor that you do not want it to move, then you need to remove the address for that channel (whether it is from ESPSomfy RTS or a Somfy remote) from the memory on the motor.  

To do this you perform the following steps.
1. Long press the prog button on a remote until the motor jogs exactly once.
2. Short press by pressing the prog button on the remote you want to remove.  In ESPSomfy RTS click on the unpair button and then press the prog button.  The shade should jog exactly once.
3. Once you have exactly one jog for opening the memory and exactly one jog for removing the address you can close the memory by short pressing the prog button on the original remote channel.

Now there are some motors that do a really stupid thing.  For some reason the require the transmitter to be close to the motor to perform this operation.   Frankly it is the most idiotic design as all it does is induce potential frustration.  So if you are having trouble getting the second jog to occur move the ESPSomfy RTS device closer to the motor temporarily.

Ok so now that you have a backgrounder on this lets get the group removed from Lounge Blackout 1.  We will need a bit of trickery here to make this happen.  Remember pairing and unpairing are actually a toggle.  So perform the following steps.

1.  Open the Lounge Blackout 1 memory with a remote.
2. Click on the edit pencil for the Downstairs group in ESPSomfy RTS.
3. Click on the trash can next to any of the members of the group.
4. Click the next button but DO NOT CLICK the OPEN MEMORY button.
5. Click next again.
6. Click the UNPAIR FROM GROUP button and the shade should jog.
7. When the Confirm Motor Response prompt appears click NO.
8. Click Cancel on the Remove Shade from Group dialog and you are done.  The group will no longer be in the Lounge Blackout 1 memory.

Ay1tsMe, 2023-07-28:
Great News! It Worked!

Thank you so much for the explanation. This is very helpful considering the guys who came out to install our Somfy blinds gave us no documentation or any sort of explanation as to how things worked other than how to use it.

Your method for unpairing the lounge blackout 1 blind from the downstairs group worked. I also had to unpair the downstairs group from the lounge blackout 1 blind as well but once you explained how they had the same address then that made much more sense.

Not sure as to what would be causing them to be on the same address in the first place. Even when I added the lounge blackout 1 as a new blind after deleting, I had to do your method to unpair the address. Does this have something to do with how they have set it up in my house? Could two blinds on the two different remotes be using the same address maybe?

Anyway thanks for the help! You're one of the better developers when it comes to response times so I really appreciate it!

rstrouse, 2023-07-28:
You can store many addresses in each motor.  So theoretically the addresses were different but one was a member of a group and Home Assistant runs through each one when setting position.

closed 2023-07-28 by Ay1tsMe