rstrouse/ESPSomfy-RTS #116: slats open/close and blind raise/lower ordering

BlackICE31, 2023-07-24:
In the documentation it is stated the following for blind's tilt options:
> Integrated - When the up button is pressed the slats fully open before the motor raises the blind. Conversely, the down button fully closes the slats before lowering the blind.

I have titlable blinds but they do not follow this order of things. Starting from a totally open blind posittion the following happens with my blinds:
- When pressing the down button: blind lowers until it "touches" the bottom, then the motor continues to function and slats start opening (from the bottom to top).
- When pressing the up button: blind start by closing the slats from top to bottom, then pulls up the blind.

It would be nice if there was an option to specify this behavior. There are options to invert commands and invert percentage, maybe there could be an option to "invert slats/blinds ordering"

rstrouse, 2023-07-29:
Let me understand this a bit better.  Can you close partially the blinds and still set the tilt position.  For instance can the blind be 50% closed and 50% tilt?  Or can you only change the tilt when when the blind is 100% closed?

BlackICE31, 2023-07-30:
No, I cannot set tilt position when blind is not closed completely. The my position is set to be with blind fully closed and slats closed. While the down arrow would close the blind completely and then continue to rotate the motor so that slats open.

Here is a link to a video I just did: [https://www.dropbox.com/s/wak9ef0isrw67n8/blinds.mov?dl=0]
In the video I press the up button. You'll see the slats closing first, then the blind raises. My button can always stop movement at any time. Down button will lower the blind and continue rolling to open the slats.

rstrouse, 2023-07-30:
That is not an inverse of how the integrated tilt mechanisms work,  Typically these pull the lift from either side of the slat to allow adjustment of the tilt at any position.  Do you know who makes these and are you sure it is installed correctly?

BlackICE31, 2023-07-31:
I am pretty sure they are installed correctly. There is no way to set the tilt angle of the slats if that is what you are referring when talking about adjusting the tilt at any position.
Here is the link to the product: [https://mvline.it/en/special-roller-shutters/orienta/]

All in all, it seems to me my blinds are not really blinds with tilt. They are normal blinds but when continuing the lowering after they touched the ground the slats start to open (the simple "pressure" done by the motor to continue lowring the blind makes them open).

I wonder if there could be a way of configuring this "slat opening" time frame then.
My blind is a normal blind but has XXX milliseconds to close the blinds with the "my" position and then additional YYY milliseconds to "continue" and open the slats.

Hope I explained my self.

rstrouse, 2023-07-31:
That is interesting.  It is a shutter but the slats disconnect and open when it reaches the bottom.  The really weird part of this is actually adjusting the tilt.   If a tilt position is targeted then the lift position has to be 100%.  If a lift position is targeted then the tilt position needs to be 100%.

I'll have to think about this one as these two things have been independent of each other.