rstrouse/ESPSomfy-RTS #118: get started - harware/firmware setup

tinarbraz, 2023-08-02:
Hi @rstrouse 

I followed your hardware guide to use your project in my house but unfortunately I'm stuck quite early in the process. 
I chose to mount an ESP-WROOM-32 with an E07-M1101D on a breadboard and followed your pin out.

Then I tried to follow your firmware guide but the ESP module was never recognized by my computer (even installing the drivers, in my case CP2102 from Silicon labs - [https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/establish-serial-connection.html]).

As I had a spare ESP32 module home, I decided to try out the firmware upload on the module alone. It worked perfectly. I then followed the Configuring the software guide until I had to connect to my home wifi network which did not work - supposedly because I was too far from the modem).

I decided to mount this somfy flashed FW ESP32 module to the breadboard and continue the setup while bringing my wifi modem closer.

Unfortunately I have the same issue than before, once mounted and routed to the CC1101 the ESP module does not start up anymore while powered by the USB.

I refurbished my previous ESP32 module to try it out and I could connect it to my desktop, observed the red diode it was powered up and I could even flash the firmware. I honestly thought I damaged the module manipulating it during soldering but apparently not.

I'm seeking out for an external pair of eyes and pieces of advice. Hereunder are some pictures:

[image]
[image]
[image]
[image]

wutz-ESP, 2023-08-02:
Have you tried flashing ESP32 without connecting CC1101?

tinarbraz, 2023-08-02:
Yes and it works well

rstrouse, 2023-08-02:
Please make sure you have pin 1 of the transceiver hooked to GND of the ESP32.  It looks like you have the 3v3 pin and GND pin reversed.

tinarbraz, 2023-08-02:
thanks for the second pair of eyes - I indeed inverted ground and 3v3.

closed 2023-08-02 by tinarbraz