rstrouse/ESPSomfy-RTS #120: Radio activated but no signal is being picked up, no RSSI

ionutm80, 2023-08-06:
Hi, as the title says I'm having a problem meaning that in spite of the fact the the radio of the E07-M1101D transceiver seems activated I do not pick any signals from the motor shades. The are 4 shades with Somfy Altus motors controlled through a Situo 5 RTS remote. I'm using an ESP32-WROOM-32U module with external WiFi. I have attached some screenshots of the Radio pages and of the remote. No shade is paired yet but from the Wiki I understand that the transceiver should be able to pick up the signal since it is in receive mode. Pls help me ti debug it. As a side comment I also have a Somfy Connectivity Kit which picked the shades motors immediately and I can controll them through Tahoma app.
[image]
[image]
[image]

ionutm80, 2023-08-06:
Solve it, the remote is sending 433.38 instead of 433.42.

closed 2023-08-06 by ionutm80

leonardpitzu, 2023-10-06:
Hi. I have the same remotes and I pick the signal from one of them intermittently. I already changes the battery and it's the same. It's got nothing to do with distance as I see the same even if the remote is 10cm away from the receiver's antenna. Any hints on troubleshooting this?

rstrouse, 2023-10-06:
Run the Scan Frequency function while holding down one of the buttons.