rstrouse/ESPSomfy-RTS #121: "Tube" control 433.92 MHz

sergiosafer, 2023-08-08:
Hi @rstrouse, I congratulate you again for the great work you have done as you have helped many of us. I have opened this case in the repository as maybe it can help someone else and not keep sending unnecessary code.

**Problem summary.....** 
I had already successfully configured some Somfy curtains and everything works great, before so much happiness (haha), I wanted to help a friend to do the same with his 6 curtains, but he has a 433.92Mhz "Tube RTL" control.

I have set the frequency on the transceiver and the codes from the remote control I receive them correctly, but when I need to link the curtain to the software with the "Prog" button I have no success.

**Reading data ........** 
After your advice that you gave me from the community, what I did was to set up 2 circuits to send and read the data from the "Prog" button of the remote control and from the "Prog" button of the software to see if with this information you can help me to solve the problem.

**Registered data ... 
PROG BUTTON FROM REMOTE CONTROL TUBE 433.92MHZ**

[{"encKey": 156,
  "address": 1368635,
  "rcode": 1740,
  "command": "Prog",
  "rssi": -63,
  "bits": 56,
  "proto": 1,
  "valid": true,
  "sync": 4,
  "pulses": [10066,97338,2545,2593,2524,2568,4842,1292,1294,635,664,1265,650,637,666,626,1296,636,666,627,643,1275,666,625,1295,662,641,625,669,628,659,1286,1291,1260,646,648,1304,625,666,1279,641,650,643,628,661,632,1293,1278,1291,1295,1286,1269,658,635,639,646,1300,1282,643,631,663,640,640,645,647,634,639,641,1291,1302,1283,1271,662,640,650,641,1279,1277,1296,1284,640,651,1288,647,644,1281,639,643,1301],
  "time": "2023-08-08T08:23:57.038-0300"
},
{"encKey": 156,
  "address": 1368635,
  "rcode": 1740,
  "command": "Prog",
  "rssi": -63,
  "bits": 56,
  "proto": 1,
  "valid": true,
  "sync": 14,
  "pulses": [32995,2530,2598,2522,2566,2539,2575,2518,2581,2529,2579,2514,2581,2528,2577,4850,1298,1277,654,641,1273,660,623,667,627,1290,642,645,658,633,1306,618,657,1286,643,648,636,665,626,644,1300,1292,1269,662,641,1290,646,634,1283,641,650,634,658,639,643,1280,1291,1294,1290,1295,1284,640,644,646,633,1288,1293,635,638,644,643,644,658,638,640,650,644,1279,1289,1295,1291,645,621,657,635,1292,1277,1314,1269,635,657,1292,653,635,1281,643,656,1287],
  "time": "2023-08-08T08:23:57.192-0300"
}]

**PROG BUTTON FROM SOFTWARE TO PAIR**

**WITH PROTOCOL RTW**

[{"encKey": 140,
  "address": 9502189,
  "rcode": 179,
  "command": "Prog",
  "rssi": -73,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [10417,7355,2579,2547,2562,2560,4874,1262,1292,644,629,636,661,1273,623,658,1287,646,619,655,663,1241,645,647,660,617,1281,1294,636,640,624,643,1300,1268,643,651,634,642,1280,1269,662,607,677,643,629,633,634,640,1298,627,637,639,665,1271,1281,616,671,645,609,1314,1270,1282,1284,1293,1286,620,642,1282,1282,1269,1310,1270,1277,640,623,1297,1284,1272,1274,659,639,642,652,1261,1290,654,613],
  "time": "2023-08-08T08:26:01.428-0300"
},
{"encKey": 140,
  "address": 9502189,
  "rcode": 179,
  "command": "Prog",
  "rssi": -72,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [28072,2589,2539,2596,2545,2566,2546,2559,2550,2598,2530,2562,2584,2546,2550,4866,1253,1295,636,679,593,669,1261,681,590,1296,634,664,618,673,1266,649,618,647,638,1310,1244,683,610,664,617,1296,1268,640,646,660,616,1283,1295,636,641,624,643,673,612,666,601,1323,609,660,608,656,1268,1283,655,643,620,658,1287,1266,1282,1307,1272,1288,619,645,1257,1334,1247,1287,1274,1301,613,677,1268,1286,1250,1295,639,655,618,669,1266,1276,646,629],
  "time": "2023-08-08T08:26:01.550-0300"
}]

**WITH PROTOCOL RTS**

[{"encKey": 164,
  "address": 9502189,
  "rcode": 180,
  "command": "Prog",
  "rssi": -70,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [10455,7332,2574,2545,2582,2551,4855,1273,1289,1292,1286,616,641,1269,1310,622,668,603,658,636,643,1263,1333,597,667,1259,1291,1270,1308,624,644,1275,1285,640,648,1286,1276,1275,662,600,1297,629,662,1267,1295,636,640,641,624,1295,1290,614,664,630,636,640,654,619,671,613,661,615,679,1243,683,610,658,645,621,656,637,642,650,619,647,611,1313,618,669,611,664,641,650,615,653,1286,1302,598,656,636,638,1269],
  "time": "2023-08-08T08:27:23.533-0300"
},
{"encKey": 164,
  "address": 9502189,
  "rcode": 180,
  "command": "Prog",
  "rssi": -71,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [658,27437,2569,2535,2596,2548,2561,2565,2553,2545,2590,2540,2567,2578,2546,2553,4870,1265,1295,1266,1294,636,640,1267,1320,613,662,600,672,645,633,1265,1301,606,684,1259,1291,1261,1318,595,677,1265,1272,650,635,1289,1295,1260,655,618,1296,637,666,1246,1304,632,640,643,628,1285,1292,629,658,642,615,646,644,628,662,615,678,617,645,1261,666,645,647,633,640,642,624,644,649,636,663,602,1300,633,660,615,659,644,622,657,636,1292,1266,613,663,653,667,1252],
  "time": "2023-08-08T08:27:23.706-0300"
}]

**WITH PROTOCOL RTV**

[{"encKey": 156,
  "address": 9502189,
  "rcode": 181,
  "command": "Prog",
  "rssi": -70,
  "bits": 56,
  "valid": true,
  "sync": 4,
  "pulses": [10453,7345,2568,2541,2569,2562,4850,1284,1309,595,676,1264,665,621,653,611,1311,618,648,660,613,1291,652,636,1291,618,647,637,634,671,614,644,655,614,661,1271,674,611,1289,644,645,610,662,615,679,617,652,1259,666,645,1273,1265,1296,1254,1311,1270,1288,1273,1274,640,677,618,647,1264,1280,1270,661,654,1293,1256,641,643,646,612,1312,621,666,610,660,642,648,1244,1310,1272,1301,1266,1269,1309],
  "time": "2023-08-08T08:27:50.603-0300"
},

{"encKey": 156,
  "address": 9502189,
  "rcode": 181,
  "command": "Prog",
  "rssi": -71,
  "bits": 56,
  "valid": true,
  "sync": 14,
  "pulses": [625,27431,2571,2542,2585,2553,2546,2570,2566,2550,2571,2551,2565,2559,2571,2543,4870,1271,1286,624,665,1256,660,622,668,602,1308,622,668,603,684,1242,664,626,1306,624,643,627,653,610,683,640,620,657,636,1271,640,648,1287,625,642,649,635,640,657,618,668,1260,667,597,1291,1288,1300,1252,1287,1270,1310,1274,1273,640,653,645,645,1262,1303,1270,639,627,1316,1260,641,640,645,624,1281,655,640,626,655,614,660,1274,1299,1292,1268,1263,1288,1274],
  "time": "2023-08-08T08:27:50.747-0300"
}]

Sorry for the length, but I just want to give you as much information as possible. 
Greetings and thanks in advance.

rstrouse, 2023-08-08:
I am traveling until next week. Thanks for the pulse dumps.  In the mean time you can try to set the address to 1502189 to see if maybe there is an address limit.  From the remote output it appears to be RTW but there may be some additional encoding.  When I get back I will take a look at the pulses.

sergiosafer, 2023-08-08:
Thank you very much @rstrouse I'll try and see you on your return. have a good trip.

sergiosafer, 2023-08-09:
Hi @rstrouse  
This is just an update for when you can see it. I did the following exercise and it worked but unstable.

1. Capture the address sent by the remote control 
2. Manually set that same address on the curtain 
3. Pair the curtain 
4. Set the protocol to RTL

And strangely it works ! !!! but ..... (there is always a but) When it works in the software, it stops working with the remote control.  To get back the control with the original remote control, I have to press several times the buttons randomly until it works and the same thing happens when I want to control it with the software.

Summarizing ..... With this configuration it does not work simultaneously with the remote control and the software.

Greetings

I leave some images

[image]
[image]

rstrouse, 2023-08-10:
Do not use the same address as an existing remote.   This looks very much like a motor that does not have any available remote slots.  This may require a factory reset of the motor.

sergiosafer, 2023-08-11:
I reset the motor, but this is still the only way the curtain is linked. Strange, isn't it?

rstrouse, 2023-08-11:
Did you try using RTV protocol for that motor.

sergiosafer, 2023-08-11:
Yes, Radio and motor in RTV
[image]
[image]

sergiosafer, 2023-08-11:
In order to link the curtain to the soft, the PROG button should be able to close the control programming by sending the code 3307, correct?

[image]

rstrouse, 2023-08-13:
So looking at your screenshots it appears that you have assigned the ESPSomfy RTS shade you defined to 1368635 and looking at the remote output this address is also 1378635.  Do not do that.  The rolling codes can and will get out of sync.  What will happen is what you are experiencing using ESPSomfy RTS will make the remote not work and using the remote will make ESPSomfy RTS not work for several button presses.  When rolling codes get out of sync the commands will be ignored until it catches up.  

Also the two prog commands above are what I would expect.  The first one opens the memory using a rolling code of 3306 and the the second sends a prog command that the motor ignores using a rolling code of 3307 which the motor promptly ignores because the address used to open the memory is the same one that is sending the command.  If it didn't do that then it would unpair the remote and you would have to start over.

What I really need is an output where you are using only the remote and the ESPSomfy RTS is not set to 1368635.  Then short press each button in the following order.  Up, Down, My, Prog.  This will give me the byte encoding for the second byte in the output.  I do not need the outputs from ESPSomfy RTS as I know what I am sending but the lower 4 bits of second byte on the RTV protocol appears to be different from the output above.  Normally on RT(x) those bits are 0 but in your case they contain some encoding.

sergiosafer, 2023-08-14:
HI @rstrouse here you are

[{"encKey": 150,
  "address": 1368635,
  "rcode": 3385,
  "command": "Up",
  "rssi": -75,
  "bits": 56,
  "proto": 1,
  "valid": true,
  "sync": 4,
  "pulses": [10055,97357,2530,2576,2534,2582,4852,1288,1276,650,660,1268,1294,1287,639,643,1305,614,670,1279,633,646,1296,635,665,617,676,614,658,648,643,630,661,1268,668,628,1290,1275,651,635,1310,1269,1313,1271,1307,1276,1297,1283,1268,662,632,647,644,1281,1289,652,660,640,628,642,649,639,664,625,669,1277,1283,1272,1312,644,647,634,661,1267,1296,1292,1295,634,635,1280,671,632,1291,644,646,1286],
  "time": "2023-08-14T18:32:02.508-0300"
},
{"encKey": 152,
  "address": 1368635,
  "rcode": 3386,
  "command": "Down",
  "rssi": -73,
  "bits": 56,
  "proto": 1,
  "valid": true,
  "sync": 4,
  "pulses": [10024,97387,2535,2552,2545,2580,4856,1258,1306,647,633,1292,645,648,1284,650,644,628,660,640,649,1271,660,641,1278,649,635,640,653,1294,633,644,1302,1272,661,628,1292,1287,650,618,652,637,1310,643,652,1265,1300,1279,1284,1302,1278,642,648,635,661,1274,1300,632,668,616,658,639,656,620,670,635,661,1267,1295,1268,1295,656,639,645,642,1271,1312,1269,1310,640,644,1273,665,640,1272,656,635,1279],
  "time": "2023-08-14T18:32:04.316-0300"
},
{"encKey": 149,
  "address": 1368635,
  "rcode": 3387,
  "command": "My",
  "rssi": -73,
  "bits": 56,
  "proto": 1,
  "valid": true,
  "sync": 4,
  "pulses": [10033,97367,2541,2570,2513,2594,4846,1275,1309,644,653,1264,1295,1284,1305,1270,1310,1271,658,617,1302,647,634,641,656,1291,1284,649,641,1280,664,617,1300,1266,667,628,661,633,650,647,1275,1290,1293,1269,1294,1283,1296,641,657,640,656,1272,1306,627,644,650,635,659,650,643,627,662,632,1275,1304,1292,1294,639,634,645,644,1281,1303,1274,1311,616,651,1285,665,621,1306,628,667,1276],
  "time": "2023-08-14T18:32:06.048-0300"
},
{"encKey": 156,
  "address": 1368635,
  "rcode": 3388,
  "command": "Prog",
  "rssi": -72,
  "bits": 56,
  "proto": 1,
  "valid": true,
  "sync": 4,
  "pulses": [10029,97370,2536,2570,2524,2589,4838,1288,1298,629,660,1282,642,630,661,637,1291,635,660,624,667,1280,636,643,1300,640,650,1268,1312,1274,1279,1279,670,634,1286,1271,1311,624,666,628,661,641,650,1269,1292,1296,1280,1294,1292,639,658,620,646,1282,1315,627,661,632,639,644,657,637,661,618,650,1291,1299,1279,1283,643,657,635,664,1269,1287,1272,1303,640,646,1294,635,641,1297,631,662,1291],
  "time": "2023-08-14T18:32:09.316-0300"
},
{"encKey": 156,
  "address": 1368635,
  "rcode": 3388,
  "command": "Prog",
  "rssi": -71,
  "bits": 56,
  "proto": 1,
  "valid": true,
  "sync": 14,
  "pulses": [619,32376,2521,2586,2534,2575,2525,2561,2532,2575,2532,2581,2525,2565,2543,2562,4855,1289,1300,627,661,1268,642,652,638,633,1304,648,634,640,653,1295,633,640,1304,646,635,1286,1296,1286,1278,1298,632,647,1295,1290,1296,633,636,640,647,644,658,1289,1294,1260,1317,1272,1291,645,648,634,661,1277,1298,641,650,618,652,661,634,645,643,657,635,1280,1295,1293,1297,633,635,653,644,1274,1310,1269,1311,622,668,1261,666,620,1295,640,654,1293],
  "time": "2023-08-14T18:32:09.462-0300"
},
{"encKey": 156,
  "address": 1368635,
  "rcode": 3388,
  "command": "Prog",
  "rssi": -72,
  "bits": 56,
  "proto": 1,
  "valid": true,
  "sync": 14,
  "pulses": [609,32386,2516,2595,2514,2595,2523,2564,2536,2575,2526,2589,2505,2603,2503,2584,4857,1285,1300,626,662,1267,643,651,634,641,1304,647,635,640,665,1269,660,636,1288,628,660,1282,1293,1287,1291,1286,637,646,1303,1252,1323,614,652,644,649,636,667,1275,1299,1266,1294,1289,1299,628,661,641,650,1270,1311,620,670,633,635,653,640,650,635,657,640,1294,1287,1269,1309,624,667,628,661,1268,1292,1289,1292,653,636,1280,644,658,1285,644,645,1288],
  "time": "2023-08-14T18:32:09.618-0300"
}]

rstrouse, 2023-08-14:
Ok I have verified that the RTV settings are correct.  Please update your device with the [`SomfyController.ino.esp32.bin`](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.ino.esp32.bin) and the [`SomfyController.littlefs.bin`](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.littlefs.bin) files from the main repository.  After you update your device the version should be v2.1.4 for both the firmware and application as seen on the System --> Firmware tab.

Next please make sure the address of your Shade in ESPSomfy RTS is **not** 1368635.  If it is then opening the memory and pairing the shade will not work.  Also ensure the protocol selected for the shade is RTV as this is the protocol that is being used by the remote.

sergiosafer, 2023-08-15:
Hola @rstrouse realice los siguientes cambios:
1. Actualice el firmware y la aplicacion a la version 2.1.4
2. Configure la radio con protocolo RTV
3. Configure la cortina con protocilo RTV y asigne un numero aleatorio  en remote address

Adjunto imagenes, pero no ha funcionado :(

[image]
[image]
[image]

sergiosafer, 2023-08-15:
I changed the channel and re-captured the codes.

[{"encKey": 152,
  "address": 1434171,
  "rcode": 250,
  "command": "Down",
  "rssi": -76,
  "bits": 56,
  "proto": 2,
  "valid": true,
  "sync": 4,
  "pulses": [10041,97282,2535,2569,2525,2565,4862,1266,1298,631,659,1291,643,631,1284,647,643,632,663,632,647,1272,661,629,1297,657,634,653,641,1274,658,639,1294,1265,649,648,1302,615,672,621,657,1265,671,621,654,639,1293,644,653,1265,649,644,1279,638,666,1281,633,647,1295,635,657,619,670,1261,654,644,1297,615,679,619,650,1289,651,620,1300,1291,644,628,659,634,1290,633,657,1281,1294,1266,1297,1282,1306,1271],
  "time": "2023-08-14T23:12:55.710-0300"
},
{"encKey": 150,
  "address": 1434171,
  "rcode": 251,
  "command": "Up",
  "rssi": -77,
  "bits": 56,
  "proto": 2,
  "valid": true,
  "sync": 4,
  "pulses": [9995,97284,2522,2596,2517,2571,4855,1271,1286,643,650,1288,1301,1256,657,648,1295,636,633,1298,626,662,1286,643,652,612,659,1274,1301,628,669,1271,639,638,1300,630,660,641,649,1269,1311,1270,1304,640,645,1272,660,624,1294,634,665,1276,646,634,1287,641,648,636,658,1265,650,636,1309,643,652,635,632,1296,627,637,1318,1268,659,634,652,618,1303,626,665,1274,1287,1293,1284,1272,1300,1291],
  "time": "2023-08-14T23:12:57.211-0300"
},
{"encKey": 149,
  "address": 1434171,
  "rcode": 252,
  "command": "My",
  "rssi": -77,
  "bits": 56,
  "proto": 2,
  "valid": true,
  "sync": 4,
  "pulses": [10012,97278,2545,2566,2532,2575,4832,1271,1309,625,670,1280,1285,1271,1310,1271,1305,1277,646,634,1285,644,647,1286,1291,1279,1282,1273,660,629,1316,634,634,1293,1294,1265,647,644,1307,626,644,1275,659,624,1294,635,659,1273,649,638,1311,619,650,636,659,1275,649,636,1286,644,650,637,659,1275,649,637,1287,1295,634,638,655,644,1297,634,638,1270,1311,1269,1305,1265,1301,1290],
  "time": "2023-08-14T23:12:58.471-0300"
},
{"encKey": 156,
  "address": 1434171,
  "rcode": 253,
  "command": "Prog",
  "rssi": -78,
  "bits": 56,
  "proto": 2,
  "valid": true,
  "sync": 4,
  "pulses": [9998,97314,2497,2604,2513,2578,4853,1278,1284,645,657,1260,648,644,655,635,1282,646,654,634,655,1269,655,638,1290,649,633,1291,1295,636,659,626,646,1300,634,648,1294,637,633,1299,1278,640,647,1296,1287,644,648,1286,626,643,1299,640,649,1269,660,640,1279,650,660,634,650,1270,659,633,1299,627,659,640,650,1268,659,640,1277,1300,634,647,646,625,1310,620,645,1283,1304,1273,1310,1268,1285,1299],
  "time": "2023-08-14T23:13:01.407-0300"
}]

rstrouse, 2023-08-15:
ESPSomfy RTS is actually sending a correct frame for the pair.  This means that the motor is either rejecting it or ESPSomfy RTS is not transmitting.  Previously, you mentioned that you have two ESPSomfy RTS modules and can see the frames from one to the other so this rules out the possibility of the transceiver not sending the command.

So this leaves only the possibility of the motor rejecting the request.  It may be rejecting the request because the address range on these motors are not as large.  So lets try to change the address by clicking on the Remote Address field and setting it to 1502189 then press Save Shade.  After you do this try the pair again.

If the above fails then the next thing to try is to clear any stored remotes from the motor.  Somfy branded motors will reject a pair if their motor does not have any available slots for the remote.  When this happens it requires a factory reset of the motor.  It is a fairly simply process using the following flowchart for your motor.

[image]

sergiosafer, 2023-08-15:
I have set the address to 1502189 but unfortunately it has not matched. 
Then I have started from scratch, resetting the motor and the remote control to re-assign a new channel but also without success! 
NOTE: The graphic you sent me is the same one that came with the remote control but it does not work as indicated. When I leave the down, stop and up buttons pressed, the motor does not work.
To enter configuration mode = 5seg off + 10seg on + 5seg off

rstrouse, 2023-08-16:
Did you press all 3 buttons at the same time.  It is not a sequence rather it is all 3 at the same time.

sergiosafer, 2023-08-16:
Yes, I have press all bottons at the same time

rstrouse, 2023-08-16:
Alright let’s try setting the bit length to 80-bit.  I am throwing noodles at the wall at this point to see what sticks.  I do not understand why this is not working.

sergiosafer, 2023-08-16:
haha ok,

sergiosafer, 2023-08-16:
Ok, I will try both alternatives and get back to you.

sergiosafer, 2023-08-16:
OK Here I attach the test with the last suggested configuration, but it doesn't work :(

https://github.com/rstrouse/ESPSomfy-RTS/assets/132413705/4c72339e-dfd9-4efe-a954-834ed67ce6e6

https://github.com/rstrouse/ESPSomfy-RTS/assets/132413705/074325c6-6c55-4496-8cab-134c274da8d6

sergiosafer, 2023-08-16:
here's something that might help: When I press the Prog button there are many repetitions until the motor goes CLACK CLACK CLACK

https://github.com/rstrouse/ESPSomfy-RTS/assets/132413705/5833ad58-0b75-4bdf-a86a-57a55d3f8bbf

the codes:
[{"encKey": 156,
  "address": 1368635,
  "rcode": 4293,
  "command": "Prog",
  "rssi": -74,
  "bits": 56,
  "proto": 2,
  "valid": true,
  "sync": 4,
  "pulses": [10015,97323,2519,2573,2520,2595,4842,1267,1318,614,666,1276,647,637,639,656,1294,637,633,653,645,1272,667,625,1295,1288,649,618,652,637,660,625,1296,1289,649,619,653,639,660,624,669,625,661,608,678,617,650,636,1314,1268,661,613,654,644,1298,1259,1319,1265,1300,1279,1283,1271,661,641,649,643,1279,640,646,1295,1287,644,647,1288,651,617,1302,641,648,643,653,1267,647,645,653,637,1283,1296,633,642,649,642],
  "time": "2023-08-16T18:45:41.792-0300"
},
{"encKey": 156,
  "address": 1368635,
  "rcode": 4293,
  "command": "Prog",
  "rssi": -74,
  "bits": 56,
  "proto": 2,
  "valid": true,
  "sync": 14,
  "pulses": [628,32361,2515,2568,2536,2573,2524,2589,2505,2577,2529,2588,2521,2570,2513,2574,4851,1277,1309,643,653,1267,650,642,654,638,1287,641,654,640,631,1274,675,611,1308,1269,660,632,647,645,654,614,1309,1271,659,631,641,645,655,640,630,646,648,628,660,634,645,645,1306,1254,670,633,636,654,1296,1259,1304,1275,1310,1269,1286,1274,675,616,656,624,1320,635,633,1297,1280,655,645,1274,639,654,1296,634,638,641,649,1294,636,629,672,622,1306,1278,646,635,635,641],
  "time": "2023-08-16T18:45:41.955-0300"
},
{"encKey": 156,
  "address": 1368635,
  "rcode": 4293,
  "command": "Prog",
  "rssi": -74,
  "bits": 56,
  "proto": 2,
  "valid": true,
  "sync": 14,
  "pulses": [621,32380,2515,2595,2494,2592,2516,2598,2525,2566,2509,2573,2529,2586,2529,2576,4838,1299,1279,642,646,1269,664,642,649,617,1303,653,640,648,636,1283,641,649,1288,1271,651,637,660,627,665,628,1287,1274,677,611,666,625,643,652,635,641,666,623,644,629,661,631,1298,1281,656,643,645,635,1286,1271,1308,1275,1299,1267,1294,1287,647,643,656,612,1310,643,654,1266,1299,628,660,1267,667,628,1312,620,645,657,635,1283,643,632,661,642,1300,1278,632,646,644,630],
  "time": "2023-08-16T18:45:42.085-0300"
},
{"encKey": 156,
  "address": 1368635,
  "rcode": 4293,
  "command": "Prog",
  "rssi": -74,
  "bits": 56,
  "proto": 2,
  "valid": true,
  "sync": 14,
  "pulses": [635,32352,2545,2568,2513,2600,2521,2569,2533,2576,2526,2563,2527,2576,2514,2587,4852,1288,1275,650,664,1262,644,650,633,635,1304,648,633,635,664,1268,660,642,1301,1276,641,624,665,629,661,615,1305,1273,661,622,666,629,659,634,652,615,651,637,659,624,668,627,1287,1296,653,613,658,648,1295,1267,1299,1280,1310,1269,1291,1270,657,640,642,624,1319,636,658,1272,1280,654,639,1276,659,625,1294,636,659,649,643,1282,658,625,640,631,1313,1271,655,639,658,620],
  "time": "2023-08-16T18:45:42.220-0300"
},
{"encKey": 156,
  "address": 1368635,
  "rcode": 4293,
  "command": "Prog",
  "rssi": -74,
  "bits": 56,
  "proto": 2,
  "valid": true,
  "sync": 14,
  "pulses": [619,32363,2529,2586,2525,2569,2513,2596,2520,2587,2513,2573,2521,2592,2507,2569,4871,1270,1286,644,649,1286,640,644,655,639,1283,645,655,637,633,1298,652,615,1308,1269,663,631,640,644,655,635,1291,1273,661,625,644,651,636,633,664,619,650,636,658,647,642,630,1312,1272,654,636,633,647,1293,1285,1291,1283,1291,1281,1282,1293,635,641,648,644,1280,655,644,1274,1291,643,649,1289,649,643,1279,641,649,644,655,1264,647,645,656,637,1288,1267,663,641,650,620],
  "time": "2023-08-16T18:45:42.457-0300"
},
{"encKey": 156,
  "address": 1368635,
  "rcode": 4293,
  "command": "Prog",
  "rssi": -74,
[... 270 lines cut ...]
},
{"encKey": 156,
  "address": 1368635,
  "rcode": 4293,
  "command": "Prog",
  "rssi": -74,
  "bits": 56,
  "proto": 2,
  "valid": true,
  "sync": 14,
  "pulses": [609,32361,2540,2570,2514,2572,2523,2591,2507,2575,2529,2585,2528,2554,2536,2574,4852,1281,1290,643,655,1285,641,647,633,639,1306,621,656,639,658,1270,658,636,1293,1269,660,641,626,644,650,636,1292,1295,633,639,641,647,645,656,635,632,648,646,653,637,633,647,1295,1265,647,646,655,637,1288,1276,1292,1295,1286,1276,1300,1266,668,628,660,614,1303,649,636,1291,1296,634,659,1265,650,635,1288,644,650,636,658,1291,627,635,657,641,1295,1286,644,648,632,639],
  "time": "2023-08-16T18:45:46.070-0300"
},
{"encKey": 156,
  "address": 1368635,
  "rcode": 4293,
  "command": "Prog",
  "rssi": -75,
  "bits": 56,
  "proto": 2,
  "valid": true,
  "sync": 14,
  "pulses": [631,32344,2513,2599,2503,2584,2527,2576,2538,2571,2523,2568,2513,2572,2547,2570,4838,1285,1291,636,658,1276,648,635,661,623,1292,635,658,647,642,1281,656,619,1321,1260,669,625,635,643,652,644,1297,1260,665,628,662,616,654,644,647,634,660,615,669,627,660,633,1291,1279,642,648,644,629,1312,1271,1304,1252,1297,1293,1294,1287,651,619,649,637,1310,618,650,1290,1300,627,662,1267,643,652,1287,651,640,655,635,1291,646,625,662,615,1304,1274,659,624,666,628],
  "time": "2023-08-16T18:45:46.070-0300"
}]

rstrouse, 2023-08-16:
Yes that is normal.  The long press on the prog button simply sends the prog command over and over again until the motor responds.  If you look closely you will notice the rolling code is repeated and the number of sync bytes starts at 4 for the first command then changes to 14 after the first.

sergiosafer, 2023-08-17:
Another interesting test that may serve as a clue: I reset the motor (on + 5sec off + on) and when the motor clack clack clack, instead of pressing Prog on the remote control, I did it from the software and the latching worked!!!! But then there is no way that the remote control works with the curtain. in short .... is working only with the software.

rstrouse, 2023-08-17:
What are the chances that only one remote can be paired to the motor.  Let me put together an open memory script. Then we can try to pair the remote.

rstrouse, 2023-08-17:
Try this command to open the memory.  Then press prog on the remote.

http://<ip address>/shadeCommand?shadeId=1&command=prog&repeat=40

sergiosafer, 2023-08-17:
here is the code

{"shadeId":1,"name":"CORTINA GABY1","remoteAddress":982189,"upTime":10000,"downTime":10000,"paired":true,"lastRollingCode":107,"position":0,"tiltPosition":0,"tiltDirection":0,"tiltTime":7000,"stepSize":100,"tiltTarget":0,"target":0,"myPos":-1,"myTiltPos":-1,"direction":0,"tiltType":0,"shadeType":0,"bitLength":56,"proto":2,"flags":0,"flipCommands":false,"flipPosition":false,"inGroup":false,"sunSensor":false,"repeats":0,"linkedRemotes":[]}

sergiosafer, 2023-08-17:
are working both, soft and remote

rstrouse, 2023-08-17:
Great!  I am not sure why this worked and the normal process did not.  Essentially, this is the normal process except in reverse and all of the same commands are issued just in a different order.  I wonder if there are any clues in the motor manual.  I am not sure which motor manual is applicable but they seem to indicate you need to set the end limits and perform some other steps before pairing other remotes.  One interesting thing I found in the motor documentation is that the prog button should be pressed for 2 seconds when pairing which conflicts with the instructions for the remote.  Look at the motor model number and lets get the correct manual for your specific motor.

There are likely clues in there as to why this did not work the normal way.  It does appear that it need some setup prior to adding another remote.

sergiosafer, 2023-08-17:
Unfortunately I don't have the engine manual, so I will try to find it on the web.

sergiosafer, 2023-08-17:
I found some information here. One is in Spanish

[CatalogTube2018_2019.pdf](https://github.com/rstrouse/ESPSomfy-RTS/files/12371628/CatalogTube2018_2019.pdf)
[957745150661153197288346343373684866770428884-Pdf.pdf](https://github.com/rstrouse/ESPSomfy-RTS/files/12371630/957745150661153197288346343373684866770428884-Pdf.pdf)

rstrouse, 2023-08-17:
Thanks for the English version.   If it were in any one of the following programming languages I could have read it but English is my only spoked word.  C, C++, C#, Typescript, ADA, Fortran, Assembler, JCL, Javascript, SQL, Avenue, Lua, SCSS, Cobol, VB, or a whole host of other esoteric crap that I have had the misfortune of coding in. 

Well that matches up with my suspicion.  The manual for the remote is incorrect.   It appears the prog command for pairing ESPSomfy RTS needs to be at least 2 seconds long for the motor to recognize it.  This is what is considered a medium press.  I think it works when the initial pairing is done because it only needs a short press for that.  We can test this theory.

Reset the motor again and pair up the remote.   Then while you have the pair shade screen open put the following command into a browser window.

http://<< ip address >>/shadeCommand?shadeId=1&command=prog&repeat=20

If this pairs then I need to create some press and hold logic for that prog button to continually resend the last frame.

sergiosafer, 2023-08-17:
Ok, right now I'm in the office, when I get back home I'll try to answer you.

sergiosafer, 2023-08-17:
My friend, it worked out well !!!! Puedo considerarte un gran amigo a esta altura no?

rstrouse, 2023-08-18:
I am going to close this as the next release will have the ability to hold down the prog button when pairing motors that do not respond to the prog command right away.

closed 2023-08-18 by rstrouse

sergiosafer, 2023-08-18:
Excellent @rstrouse thank you very much, you have really done an incredible job and I have nothing but thanks for the patience and time you invested to solve this problem. Thank you so much!!!

I would love to be able to help you or repay you in some way (a good Argentinian wine would not go amiss) for the time you have dedicated to it, I know how much I appreciate it and I appreciate it very much.

sergiosafer, 2023-08-25:
Hi @rstrouse I'm just asking when you would be releasing the next update? Thanks a lot.

rstrouse, 2023-08-31:
v2.1.4 has been released.  This will allow you to press and hold the prog button on the pairing screen until the motor responds.

sergiosafer, 2023-09-01:
Thanks for all @rstrouse. Incredible work!!!!