rstrouse/ESPSomfy-RTS #122: Not able to configure radio

LorisSchmit, 2023-08-08:
I'm having trouble with configuring my transceiver. I'm using an E07-M11010 just like the one in the hardware guide (except it's the v2.0), and I am using the same pinout. Maybe the pinout for the v2.0 is different? But as far as I could find out, it is the same (as in the hardware guide). However, if I boot my ESP32 with the transceiver connected, the ESP Somfy RTS homepage does not load (webserver not reachable) and it appears to be stuck in a boot loop, as this serial ouput repeats over and over again:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x32 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
inv⸮ets Jul 29 2019 12:21:46
```
When I connect the transceiver after the ESP32 has booted, configure the radio and hit the Save Radio Button, it get's stuck on a loading screen and ESP Somfy RTS is not responsive anymore. The serial output while trying to configure the radio is:
```
Deserialize Radio JSON SCK:23 MISO:13 MOSI:19 CSN:18 RX:5 TX:12
Save Radio Settings SCK:23 MISO:13 MOSI:19 CSN:18 RX:5 TX:12
Applying radio settings Setting Data Pins RX:5 TX:12
Setting SPI Pins SCK:23 MISO:13 MOSI:19 CSN:18
Radio Pins Configured!
``` 

I think it might be a hardware problem, maybe a faulty transceiver? Or I am using the wrong pinout? 
Did someone else experience this behavior?

Thanks!

rstrouse, 2023-08-10:
Recheck your wiring.  If the hardware is locking up it is most likely a wiring problem.

rstrouse, 2023-09-01:
@LorisSchmit did you get your issue resolved?

rstrouse, 2023-09-05:
I am going to close this for now.  Feel free to re-open it if your problem has not been resolved.

closed 2023-09-05 by rstrouse