rstrouse/ESPSomfy-RTS #125: Support request for ESP32-S2

Phil1988, 2023-08-14:
Hello,

I tried to test it on a WEMOS ESP32 S2 mini, but I first couldnt make it work.

I started with the long road and tried it with the Arduino IDE 1.8.13.
Boards, librarys etc. should be correctly installed.
Flashing the SomfyController.ino did work:
[image]
After I connected to the AP and opened the WebIF via a browser and 192.168.4.1, I only saw a white page with something like "config.cfg".

Uploading the Filesystem via LITTLEFS from the Arduino IDE was impossible and gave me that:
[image]

I tried that serveral times with no luck.
After a while, I tried the esptool and I could see it finishes but this time no AP was showing:
[image]
(and yes, I restarted the board and did several attempts - so this is confirmed)

After a while Itried the flash_download_tool_3.9.5.
I selected ESP32-S2...
[image]
...and it run threw using the SomfyController.onboard.esp32.bin without issues, but again no AP was showing:
[image]

When writing this post, I did everything again, to take screenshots and did flash the SomfyController.ino via the Arduino IDE.
This time, not only the AP was visible, but the blank page with something like "config.cfg" was not there, but the real WebIF!

I guess, that flashing the firmware with the SomfyController.onboard.esp32.bin set the files to the right location and flashing it with the Arduino IDE made the Wifi/Webserver work correctly and boom (!) it works.

It would be nice if someone could provide a pre-compiled SomfyController.onboard.esp32-s2.bin (for the ESP32-S2) that works out of the box for all the folks out there that are or will use this chip, because I see its getting more popular.

Thanks for this project by the way.
I will play tomorrow a bit with it, because it took me 5 hours today to make it work :D

rstrouse, 2023-08-14:
These binaries are created as an action on the repo. 

https://github.com/rstrouse/ESPSomfy-RTS/actions/runs/5849927452

Phil1988, 2023-08-14:
Very well hidden for me ^^

Might be good, if folks could find it in the releases.
Issue closed as it worked also with my "workaround", but I wanted to give a heads up for this ;)

closed 2023-08-14 by Phil1988

rstrouse, 2023-08-15:
Yeah I probably need to figure out how to attach it in the release.

lammetak, 2023-08-19:
Is there still a way to download the image?