rstrouse/ESPSomfy-RTS #127: Una le tot pari on motor/remote

leonardpitzu, 2023-08-18:
I have 3 Situo 5 RTS Pute II Remote and 5 window blinds. I managed to pair 2 remotes and 4 windows. The last windows and the last remote are a mystery though and have few ideas to start with.
Problem 1: i cannot see any button presses of the last remote in the log.
Problem 2: i cannot pair the last motor.
The last remote successfully operates the last window AND all the rest. One of the other remotes also works with this window but the second one does not (did so though until i messed with it?!?!?).
I tries to pair the last window using one of the working remotes but the shade jogs while entering pairing mode and never jogs when i press pair on the ESPSomfy…

rstrouse, 2023-08-18:
I have recently found that some motors are slow to respond to the Prog command.  Up until now I have seen motors respond immediately but it sounds like you are not seeing any response from one of your remotes in the logs.

If you install the firmware [SomfyController.ino.esp32.bin](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.ino.esp32.bin) and application [SomfyController.littlefs.bin](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.littlefs.bin) from the main repository you can hold the prog button down for these slow motors.

For the remote that is showing anything in the logs please take the remote closer to the ESPSomfy RTS device and see if you get some responses.  If you do then perhaps you need to widen the RX bandwidth.  I have found some remote to not be tuned to the center frequency very well and end up sacrificing range.

leonardpitzu, 2023-08-19:
OK, so I have made a very good progress with your help. The last motor was indeed "slow" AND I had to adjust the frequency to 433.38MHz.
As before: 5 windows, 3 remotes and the ESPSomfy.

- All windows work with ESPSomfy now.
- All windows work with remote 1-2.
- All but the last window work with remote 3.
The last window is the one with the "slow" motor but I think the issue is with the remote (Situo 5 RTS Pure II). When I try to pair the remote on channel 5 nothing happens. I use one of the working remotes and the shades jog. I select channel 5 on the non-working remote and simply nothing happens. In non-pairing mode any button I press when channel 5 is selected results in all LED's blinking (similar to the optical feedback you get when actually selecting channel 5). I have no idea if the remote is broken (although the first 4 channels work correctly with the first 4 windows) or if anything else is wrong and found no indication of a similar issue online.

rstrouse, 2023-08-21:
That sounds like a faulty remote or a motor that has had all its address slots filled.  Once the motor runs out of slots it will simply ignore the address.  You may need to factory reset that motor and start over with its configuration.

leonardpitzu, 2023-08-21:
Motor reset, entire setup made all over again. Last remote, on channel 5, is unable to be paired with any one of the motors. I thus consider the remote to be dead and have ordered a new one.

leonardpitzu, 2023-08-23:
So after a few days new issue. Then I use ESPSomfy the motor reacts instantly. AFTER using ESPSomfy once I have to press 3 times the button on the remote to get the shades moving. After that the motor reacts as expected to one button press on the remote, until I use ESPSomfy again.

rstrouse, 2023-08-23:
This happens if the remote address is the same as the ESPSomfy RTS address.  What is the address that is shown when you press the button on the remote from the logs.  It should not be the same as any of the addresses defined in ESPSomfy RTS.

leonardpitzu, 2023-08-28:
That was correct, the address of one remote was exactly the same to the one in ESPSomfy. Probably a copy-paste issue on my side. Issue solved and many thanks for your support!

rstrouse, 2023-09-01:
I am going to close this for now.  If you have any other issues submit a new oen.

closed 2023-09-01 by rstrouse