rstrouse/ESPSomfy-RTS #128: ESP32S3 keeps rebooting after install (m5stampS3)

senolsun, 2023-08-18:
Hello,

After flashing the esp32 s3 with SomfyController.onboard.esp32.bin the device stops responding and keeps rebooting. windows device manager also refreshes every second as the device restarts.

what could be the reason behind this? flashed the esp32 without connecting the radio module

rstrouse, 2023-08-19:
Try the firmware from here for the s3.

https://github.com/rstrouse/ESPSomfy-RTS/actions/runs/5906783834

senolsun, 2023-08-19:
thanks. i managed to flash that S3 file you mentioned and was able to reach the setup page. I entered the WLAN details and the device connected to my network and got assigned an ip. I can ping the IP but cannot reach the webpage when I try to browse to its IP address.

rstrouse, 2023-08-21:
If it is not repeating that PSRAM error that is likely the reboot process as ESP32 writes to the address to reboot.  If you are accessing the setup while in AP mode then the software and the associate webserver is running.  Double check the address assigned by your router.  You will see this address in the logs on a line that starts with Successfully Connected to WiFi!!!!! followed by the assigned IP address.

senolsun, 2023-08-23:
hello. the issue happens when you save the settings with default pin assignments and reboot. I usually get the wifi settings out of the way before focusing on the rest and this approach caused this trouble. When you set the pins to proper pins, it doesn't cause this issue. 

However, I think it'd be better if the S3 and C3 versions of the flash file were not hidden in the repo this much. Some explanation on the front page would help others to choose the right version in the first trial. 

Also, I could not get it to work using an ESP32 C3 mini, so I switched to an ESP Wroom 32 and using your pin layout, it worked perfectly. 

I would like to thank you for creating such an amazing software. Keep up the good work!

closed 2023-08-23 by senolsun