rstrouse/ESPSomfy-RTS #129: docu: ESP32 POE mounting holes

philipparndt, 2023-08-19:
Hi, 

thanks for this project. Some note about the [docu](https://github.com/rstrouse/ESPSomfy-RTS/wiki/Simple-ESPSomfy-RTS-device#olimex-esp32-poeevb):

the new Olimex ESP32 POE revision "L" comes with mounting holes.
see: 
https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware
https://github.com/OLIMEX/ESP32-POE/tree/master/HARDWARE/ESP32-PoE-hardware-revision-L

[image]

rstrouse, 2023-08-21:
Did the documented pins for the original Olimex work for this board?

rstrouse, 2023-09-01:
I updated the documentation to reflect revision L of this board.

rstrouse, 2023-09-02:
Closing this for now.  Let me know if there are any other differences that I am not aware of.  I do not have this hardware.

closed 2023-09-02 by rstrouse

philipparndt, 2023-09-03:
@rstrouse I've received the new board and (Hardware Revision L) and can confirm that everything is sill working (I've tested the radio and ethernet as documented)