rstrouse/ESPSomfy-RTS #130: [feature request] reverse proxy support or ssl certificate

leonardpitzu, 2023-08-21:
Currently the only way to access ESPSomfy is by being in the same network and accessing the IP address of the device.
There is no way to add ESPSomfy to an iframe (if HA is served over SSL).
I would kindly ask for SSL support (load own certificate maybe) or support for having ESPsomfy behind a reverse proxy..

leonardpitzu, 2023-08-21:
or even better - support the configuration of ESPSomfy directly from HA (if there is a way to do so).

rstrouse, 2023-08-21:
The last option is probably the most viable.  While the ESP32 is definitely a capable device there are limits to what I would throw at it.

That being said there is so much of the device configuration that require visual cues from the motors that make much of the configuration impossible to do remotely.  Are there specific configuration items that you were thinking about?

leonardpitzu, 2023-08-21:
By no means I am unable to type in the IP and have everything configured in ESPSomfy. The "nice to have" thing would be to have it all doable from one single place and that would be HA given that anyways everything ends up there. The "natural" choice would be to have ESPSomfy accessible through an iframe but due to (good) security reasons this will not work. So API level access to... everything would be nice. Not doing it only remote but also local - have the phone in your hand and walk in the rooms to set stuff up. And in my case the local instance is still https - I have a reverse proxy and a single web interface to access everything in my network (HA, Plex, NAS, Router, ADblocker etc.).

xvisor16, 2023-10-25:
I would like this too! I want to be able to use it through my reverse proxy. But for that the webserver at least needs to run https. It doesn't matter if there is a correct certificate or not because the reverse proxy will take care of it but https is needed since even through reverse proxy in most browser the elevation of websocket connection from insecure to secure won't work anymore when not served over https.

rstrouse, 2023-10-28:
I have been looking into this.  Is there a need for anything other than self-signed certificates or is the simple case workable.

leonardpitzu, 2023-10-29:
No, it would be enough for me.

xvisor16, 2023-11-06:
@rstrouse, that would be enough for me too! Offcourse it would also be nice to have the option to upload a pem certificate, although I don't think that it will be used by many. Unknowing people won't upload a cert because they don't have one and the others who need this will probably all use it through reverse proxy so self signed for 5 a 10 years or so would perfect. An option to regenerate one would be good, because if it becomes invalid some reserve proxies will also give issues. But you could just create a cert when turning SSL off and on so that if it expires turn SSL off and on fixes it.

xvisor16, 2024-01-22:
Hi @rstrouse, this still is an issue for me? Do you have any idea when you might have the time to work on this?

rstrouse, 2024-01-22:
I have been doing lots of research on this topic.  Unfortunately, so far the results using the existing libraries have not been good.  Simply adding the libraries to the project blows the code size so it no longer fits on the device.  I have begun refactoring some of the code to reduce the footprint but it is slow going.

xvisor16, 2024-01-23:
@rstrouse, Okey clear. Don't know if, but you probably already did check ESP32_HTTPS_Server out by Frank Hessel? It's a library I use a lot on my own inhome ESP projects. It's pretty simple and not that big, but maybe to big still? And I always store the certificates on the SPIFFS filesystem. I haven't checked your code maybe it won't suit at all. But maybe it may help you? Thanks again for all the hard work and I will patiently wait for it.

rstrouse, 2024-01-25:
I have tried out that server but I am already at 98% of program storage space but I don't think it is entirely his code that eats it up.  The app itself uses littlefs so cert storage is not really a problem.  I am a bit low on heap but I never got that far.  I may still try to put this on one of my 8Mb cards which would have enough program space.

Out of curiosity why do you need https for a reverse proxy.  I thought the whole idea was to encrypt on the other side of the proxy.

xvisor16, 2024-01-25:
Aah oke. Well the biggest problem is the Websocket. Modern versions of Chrome, Edge and Firefox don't support unsecure websockets anymore as endpoint. Since ESPSomfy uses websockets, these need to be available as secure websockets (wss). The actual webserver doesn't need to be. With old versions of the browsers it just works. Although proxies can upgrade the websocket it's no longer supported by browsers to have mixed content in the initial setup. So it won't work. See image below

[image]

The funny thing is that if you could serve the websocket on both ws and wss, for me there would be no need for a full blown https implementation, since I only want to proxy it. But for other users that would like https support this wouldn't be enough.
As to the best of my knowledge the websocket only has to be served on wss not even really being secure but I don't know for sure.

rstrouse, 2024-01-25:
> Modern versions of Chrome, Edge and Firefox don't support unsecure websockets anymore as endpoint. 

Actually that is entirely untrue.  The only time you cannot use a non TLS socket is when the browser model is already TLS.  All you need to do is set the proxy to route 8080 as well.  I will detect wss or ws in the client.

xvisor16, 2024-01-26:
True, but I meant they don't support it when the site is running over HTTPS/TLS, thought it was clear from the total message, sorry. I already route the websocket on 8080 through proxy with websocket upgrade. Let me know you added the check.

joneshf, 2024-05-06:
Looks like there's another reason _some_ solution will be needed in the future. Some browsers have started blocking downloads from HTTP sites. In particular, Brave is doing this now (I think from [this change](https://chromium-review.googlesource.com/c/chromium/src/+/4126471/7), but I'm not certain).

Blocking files from HTTP sites is relevant because the workflow of upgrading versions includes downloading a backup. And if ESPSomfy-RTS is served over HTTP, each backup will not be fully available until it's explicitly allowed. There is currently a "Keep" button when you download a file over HTTP:
[image]
That's the thing you click to explicitly allow the file.

I haven't set mine up behind an HTTPS proxy yet, so I don't know if that would actually be enough, or if the entire chain of requests need to be HTTPS–so the ESPSomfy-RTS would need to host some certificate. I assume it would be fine to be behind an HTTPS proxy (since a lot of real-world websites work like that) but the message in that Chromum change makes it seem like if even one link in the chain of requests is HTTP, it'll block the file and require explicit allowing.

---

In any case, it's a minor inconvenience/easy thing to forget at this point. And the workaround isn't hard: you click one button each time you upgrade. I have no idea about the timeline to disallow files entirely. But if other HTTP -> HTTPS rollouts are any indication, it'll probably be quite a while (years maybe?) before it's completely disallowed. So I doubt it's something that'll need to be addressed in the next few months. But one more reason to come up with _something_, even if that something is adding a sentence or two in the wiki about hosting behind an HTTPS reverse proxy.

alka79, 2024-05-31:
Let me +1 this request.
I am using Synology DDNS and reverse proxy. It works well to access my hosted web sites, including HA. Unfortunately, Espsomfy hangs while loading the page. 
Firefox console says  websocket connection error.  DOMException: The operation is insecure.
That led me tho this discussion and now I am lost. Is this something that can be solved in ESPSofmy?

rstrouse, 2024-06-01:
You need to include port 8080 on your reverse proxy for the socket.

alka79, 2024-06-02:
It was mentioned in the previous posts. I have opened port 8080, also on my router just in case.

hexa2k9, 2024-06-20:
It might be easiest to look at X-Forwarded Headers when requests are passed into ESPSomfy-RTS.

It seems to just use the `ws://` protocol instead of `wss://` in case the "Frontend Request" was served through https: https://github.com/rstrouse/ESPSomfy-RTS/blob/main/data/index.js#L472

Having this conditionally based on `X-Forwarded-Proto` (or another "communicated requirement") might eventually do the trick already. But it'll require to use port 8080 to be encrypted on the Frontend (I prefer HAProxy but that should not matter too much). Port 8080 however would "by thought" it's unencrypted.

The footprint should be relatively small and could look like this: 
```
let proto = window.location.protocol === 'http:' ? 'ws:' : 'wss:';
socket = new WebSocket(`${proto}//${host}:8080/`);
```

leonardpitzu, 2025-01-14:
Maybe adding it to the HA sidebar using [such](https://github.com/lovelylain/hass_ingress) an integration would be an option. But ws:// and some path needs to be fixed first