rstrouse/ESPSomfy-RTS #131: issue with remote after configuring shutters in espsomfy

groninge01, 2023-08-22:
Hi,

First off thanks for this amazing piece of software. Setting up the hardware & software took some tinkering but in the end i managed to do it :)

But now my original remote is acting weird, it's probably something i messed up but i can't figure out how to fix it.

I have a 5 channel telis remote which, before installing & configuring espsomfy) operated the 2 shutters on 2 different channels (back on channel 1 and front on channel 2).
After installing & configuring espsomfy, channel 2 still operates the front shutter but channel 1 now operates both (up/down/my). So when I use the remote to operate the shutters I need to remember to first operate the front one and then the back one. Otherwise the status (open or closed) might be wrong in espsomfy/ha (because channel 1 is linked to back and 2 to front in espsomfy)

Have you or anyone else also experienced this or knows how to resolve this?

rstrouse, 2023-08-22:
You have managed to pair the front shade to the remote channel 1.  This happens when the memory for the motor is open and you press the prog button on a channel.  So at some point you opened the front motor memory then pressed the prog for channel 1 on the remote.  You may need to factory reset the back shutter and pair it again to channel 2.  Somfy recommends unplugging any motors that you are not pairing so you don't make this mistake but it is not required.

If it is difficult to power the motors on and off we can trick the motor into pairing mode using ESPSomfy RTS.   To do this you need to know the id of the front shade that is configured in ESPSomfy RTS.  You will find this in the upper right corner when you edit the shade configuration.  It will have a number (x/32) where x is the id.

Then from a browser enter the following command in the address bar.  `http://<ip address of espsomfy device>/shadeCommand?shadeId=<front shade id>&command=prog&repeat=40`

The motor on the front shade should jog.  Then select channel 1 on your remote and short press the prog button on the back of the remote.  This should remove the front from the channel on the remote.

groninge01, 2023-08-22:
@rstrouse Thank you very much! That did the trick and it is working smoothly know.

closed 2023-08-22 by groninge01