rstrouse/ESPSomfy-RTS #132: mqtt position not correct position web interface ok

wutz-ESP, 2023-08-24:
i am using this great software in combination with fhem and mqtt to send position from ESPSomfy-RTS
board WT32-ETH01 connected wired ethernet

since 12:54 all shades are in position 100
opened web interface of ESPSomfy-RTS at 13:59 all shades submit 100 % to mqtt

the only thing changed by me is tilt-time - before i had 34.000 and now 3.000
everything worked perfectly in previous versions with wrong tilt time
the command was send by Somfy remote

[image]

[image]
[image]
[image]

> 2023-08-24_12:52:58 MQTT2_client_000000892815b80c shades_6_position: 100
2023-08-24_12:52:58 MQTT2_client_000000892815b80c shades_6_direction: 0
2023-08-24_12:52:58 MQTT2_client_000000892815b80c shades_4_position: 51
2023-08-24_12:52:59 MQTT2_client_000000892815b80c shades_5_position: 50
2023-08-24_12:52:59 MQTT2_client_000000892815b80c shades_4_position: 52
2023-08-24_12:53:00 MQTT2_client_000000892815b80c shades_6_position: 30
2023-08-24_12:53:00 MQTT2_client_000000892815b80c shades_6_direction: 1
2023-08-24_12:53:01 MQTT2_client_000000892815b80c shades_3_position: 49
2023-08-24_12:53:01 MQTT2_client_000000892815b80c shades_3_position: 100
2023-08-24_12:53:01 MQTT2_client_000000892815b80c shades_3_direction: 0
2023-08-24_12:53:01 MQTT2_client_000000892815b80c shades_4_position: 55
2023-08-24_12:53:02 MQTT2_client_000000892815b80c shades_5_position: 58
2023-08-24_12:53:02 MQTT2_client_000000892815b80c shades_4_position: 100
2023-08-24_12:53:02 MQTT2_client_000000892815b80c shades_4_direction: 0
2023-08-24_12:53:03 MQTT2_client_000000892815b80c shades_6_position: 37
2023-08-24_12:53:03 MQTT2_client_000000892815b80c shades_5_position: 100
2023-08-24_12:53:03 MQTT2_client_000000892815b80c shades_5_direction: 0
2023-08-24_12:53:03 MQTT2_client_000000892815b80c shades_3_position: 55
2023-08-24_12:53:03 MQTT2_client_000000892815b80c shades_3_direction: 1
2023-08-24_12:53:04 MQTT2_client_000000892815b80c shades_4_position: 61
2023-08-24_12:53:04 MQTT2_client_000000892815b80c shades_4_direction: 1
2023-08-24_12:53:04 MQTT2_client_000000892815b80c shades_6_position: 100
2023-08-24_12:53:04 MQTT2_client_000000892815b80c shades_6_direction: 0
2023-08-24_12:53:05 MQTT2_client_000000892815b80c shades_6_position: 38
2023-08-24_12:53:05 MQTT2_client_000000892815b80c shades_6_direction: 1
2023-08-24_12:53:06 MQTT2_client_000000892815b80c shades_5_position: 59
2023-08-24_12:53:06 MQTT2_client_000000892815b80c shades_5_direction: 1
2023-08-24_12:53:06 MQTT2_client_000000892815b80c shades_3_position: 56
2023-08-24_12:53:07 MQTT2_client_000000892815b80c shades_4_position: 62
2023-08-24_12:53:07 MQTT2_client_000000892815b80c shades_6_position: 39
2023-08-24_12:53:07 MQTT2_client_000000892815b80c shades_5_position: 60
2023-08-24_12:53:08 MQTT2_client_000000892815b80c shades_3_position: 57
2023-08-24_12:53:08 MQTT2_client_000000892815b80c shades_4_position: 63
2023-08-24_12:53:09 MQTT2_client_000000892815b80c shades_6_position: 40
2023-08-24_12:53:09 MQTT2_client_000000892815b80c shades_3_position: 59
2023-08-24_12:53:09 MQTT2_client_000000892815b80c shades_4_position: 65
2023-08-24_12:53:10 MQTT2_client_000000892815b80c shades_5_position: 62
2023-08-24_12:53:10 MQTT2_client_000000892815b80c shades_6_position: 41
2023-08-24_12:53:11 MQTT2_client_000000892815b80c shades_3_position: 60
2023-08-24_12:53:11 MQTT2_client_000000892815b80c shades_6_position: 42
2023-08-24_12:53:12 MQTT2_client_000000892815b80c shades_3_position: 62
2023-08-24_12:53:12 MQTT2_client_000000892815b80c shades_4_position: 68
2023-08-24_12:53:12 MQTT2_client_000000892815b80c shades_5_position: 65
2023-08-24_12:53:13 MQTT2_client_000000892815b80c shades_6_position: 47
2023-08-24_12:53:13 MQTT2_client_000000892815b80c shades_3_position: 66
2023-08-24_12:53:13 MQTT2_client_000000892815b80c shades_4_position: 71
2023-08-24_12:53:14 MQTT2_client_000000892815b80c shades_5_position: 68
2023-08-24_12:53:14 MQTT2_client_000000892815b80c shades_4_position: 72
2023-08-24_12:53:15 MQTT2_client_000000892815b80c shades_6_position: 48
2023-08-24_12:53:15 MQTT2_client_000000892815b80c shades_5_position: 69
2023-08-24_12:53:16 MQTT2_client_000000892815b80c shades_3_position: 67
2023-08-24_12:53:16 MQTT2_client_000000892815b80c shades_4_position: 75
2023-08-24_12:53:17 MQTT2_client_000000892815b80c shades_5_position: 71
2023-08-24_12:53:17 MQTT2_client_000000892815b80c shades_6_position: 51
2023-08-24_13:59:59 MQTT2_client_000000892815b80c shades_3_position: 100
2023-08-24_13:59:59 MQTT2_client_000000892815b80c shades_3_direction: 0
2023-08-24_13:59:59 MQTT2_client_000000892815b80c shades_4_position: 100
2023-08-24_13:59:59 MQTT2_client_000000892815b80c shades_4_direction: 0
2023-08-24_13:59:59 MQTT2_client_000000892815b80c shades_5_position: 100
2023-08-24_13:59:59 MQTT2_client_000000892815b80c shades_5_direction: 0
2023-08-24_14:00:00 MQTT2_client_000000892815b80c shades_6_position: 100
2023-08-24_14:00:00 MQTT2_client_000000892815b80c shades_6_direction: 0

rstrouse, 2023-08-24:
Upload a backup so I can replicate your setup.

wutz-ESP, 2023-08-24:
hope this is ok. this is not realy a problem because web interface is working well.

[ESPSomfyRTS 2023-08-24T19_08_16.zip](https://github.com/rstrouse/ESPSomfy-RTS/files/12431864/ESPSomfyRTS.2023-08-24T19_08_16.zip)

rstrouse, 2023-09-01:
I posted some work in the MQTT module to reduce traffic to the broker in v2.1.4.   I wasn't able to replicate this but reducing the number of times the state publishes cannot hurt.

rstrouse, 2023-09-05:
One other thing on this make sure you are not changing the /set topic(s) to ESPSomfy RTS with other than QoS 0 and retain = false.  This behavior will make the broker continue to send this command if it is not acked.

wutz-ESP, 2023-09-07:
created Filelog of this topic

2023-09-07_14:18:18 MQTT2_client_000000892815b80c shades_6_position: 0
2023-09-07_14:18:19 MQTT2_client_000000892815b80c **shades_3_position: 100**
2023-09-07_14:18:19 MQTT2_client_000000892815b80c shades_4_position: 99
2023-09-07_14:18:19 MQTT2_client_000000892815b80c shades_5_position: 94
2023-09-07_14:18:20 MQTT2_client_000000892815b80c shades_3_position: 95
2023-09-07_14:18:20 MQTT2_client_000000892815b80c **shades_4_position: 100**
2023-09-07_14:18:20 MQTT2_client_000000892815b80c shades_6_position: 79
2023-09-07_14:18:21 MQTT2_client_000000892815b80c **shades_5_position: 100**
2023-09-07_14:18:21 MQTT2_client_000000892815b80c shades_6_position: 0
2023-09-07_14:18:21 MQTT2_client_000000892815b80c shades_3_position: 96
2023-09-07_14:18:21 MQTT2_client_000000892815b80c shades_5_position: 95
2023-09-07_14:18:21 MQTT2_client_000000892815b80c shades_3_position: 5
2023-09-07_14:18:21 MQTT2_client_000000892815b80c shades_6_position: 80
2023-09-07_14:18:22 MQTT2_client_000000892815b80c **shades_6_position: 100**
2023-09-07_14:18:22 MQTT2_client_000000892815b80c shades_6_position: 0
.....
2023-09-07_14:18:59 MQTT2_client_000000892815b80c shades_3_position: 59
2023-09-07_14:19:00 MQTT2_client_000000892815b80c shades_4_position: 64
2023-09-07_14:19:00 MQTT2_client_000000892815b80c shades_6_position: 47
2023-09-07_14:19:00 MQTT2_client_000000892815b80c shades_5_position: 63
2023-09-07_14:19:00 MQTT2_client_000000892815b80c shades_3_position: 60

**opening WEB-Interface of ESPSomfy RTS - position went ok**

2023-09-07_14:59:07 MQTT2_client_000000892815b80c shades_3_position: 100
2023-09-07_14:59:07 MQTT2_client_000000892815b80c shades_4_position: 100
2023-09-07_14:59:07 MQTT2_client_000000892815b80c shades_5_position: 100
2023-09-07_14:59:08 MQTT2_client_000000892815b80c shades_6_position: 100

> 2023-09-07_14:17:28 MQTT2_client_000000892815b80c shades_5_position: 1
2023-09-07_14:17:32 MQTT2_client_000000892815b80c shades_5_position: 2
2023-09-07_14:17:33 MQTT2_client_000000892815b80c shades_5_position: 3
2023-09-07_14:17:37 MQTT2_client_000000892815b80c shades_5_position: 4
2023-09-07_14:17:40 MQTT2_client_000000892815b80c shades_5_position: 5
2023-09-07_14:17:40 MQTT2_client_000000892815b80c shades_3_position: 60
2023-09-07_14:17:41 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:41 MQTT2_client_000000892815b80c shades_4_position: 64
2023-09-07_14:17:41 MQTT2_client_000000892815b80c shades_4_position: 0
2023-09-07_14:17:42 MQTT2_client_000000892815b80c shades_5_position: 63
2023-09-07_14:17:42 MQTT2_client_000000892815b80c shades_6_position: 47
2023-09-07_14:17:43 MQTT2_client_000000892815b80c shades_3_position: 63
2023-09-07_14:17:43 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:43 MQTT2_client_000000892815b80c shades_4_position: 67
2023-09-07_14:17:44 MQTT2_client_000000892815b80c shades_5_position: 65
2023-09-07_14:17:44 MQTT2_client_000000892815b80c shades_6_position: 50
2023-09-07_14:17:44 MQTT2_client_000000892815b80c shades_4_position: 68
2023-09-07_14:17:44 MQTT2_client_000000892815b80c shades_5_position: 6
2023-09-07_14:17:44 MQTT2_client_000000892815b80c shades_5_position: 66
2023-09-07_14:17:45 MQTT2_client_000000892815b80c shades_4_position: 1
2023-09-07_14:17:45 MQTT2_client_000000892815b80c shades_6_position: 51
2023-09-07_14:17:45 MQTT2_client_000000892815b80c shades_3_position: 64
2023-09-07_14:17:45 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:45 MQTT2_client_000000892815b80c shades_4_position: 69
2023-09-07_14:17:45 MQTT2_client_000000892815b80c shades_5_position: 67
2023-09-07_14:17:46 MQTT2_client_000000892815b80c shades_3_position: 65
2023-09-07_14:17:46 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:46 MQTT2_client_000000892815b80c shades_6_position: 52
2023-09-07_14:17:46 MQTT2_client_000000892815b80c shades_4_position: 70
2023-09-07_14:17:46 MQTT2_client_000000892815b80c shades_5_position: 68
2023-09-07_14:17:47 MQTT2_client_000000892815b80c shades_3_position: 66
2023-09-07_14:17:47 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:47 MQTT2_client_000000892815b80c shades_6_position: 53
2023-09-07_14:17:47 MQTT2_client_000000892815b80c shades_4_position: 71
2023-09-07_14:17:47 MQTT2_client_000000892815b80c shades_5_position: 69
2023-09-07_14:17:48 MQTT2_client_000000892815b80c shades_4_position: 2
2023-09-07_14:17:48 MQTT2_client_000000892815b80c shades_3_position: 67
2023-09-07_14:17:48 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:48 MQTT2_client_000000892815b80c shades_4_position: 72
[... 398 lines cut ...]
2023-09-07_14:18:56 MQTT2_client_000000892815b80c shades_6_position: 43
2023-09-07_14:18:56 MQTT2_client_000000892815b80c shades_3_position: 55
2023-09-07_14:18:56 MQTT2_client_000000892815b80c shades_4_position: 60
2023-09-07_14:18:56 MQTT2_client_000000892815b80c shades_5_position: 59
2023-09-07_14:18:56 MQTT2_client_000000892815b80c shades_3_position: 56
2023-09-07_14:18:57 MQTT2_client_000000892815b80c shades_6_position: 44
2023-09-07_14:18:57 MQTT2_client_000000892815b80c shades_4_position: 61
2023-09-07_14:18:57 MQTT2_client_000000892815b80c shades_5_position: 60
2023-09-07_14:18:57 MQTT2_client_000000892815b80c shades_3_position: 57
2023-09-07_14:18:57 MQTT2_client_000000892815b80c shades_4_position: 62
2023-09-07_14:18:58 MQTT2_client_000000892815b80c shades_6_position: 45
[... cut ...]
2023-09-07_14:19:00 MQTT2_client_000000892815b80c shades_6_position: 47
2023-09-07_14:19:00 MQTT2_client_000000892815b80c shades_5_position: 63
2023-09-07_14:19:00 MQTT2_client_000000892815b80c shades_3_position: 60
2023-09-07_14:59:07 MQTT2_client_000000892815b80c shades_3_position: 100
2023-09-07_14:59:07 MQTT2_client_000000892815b80c shades_4_position: 100
2023-09-07_14:59:07 MQTT2_client_000000892815b80c shades_5_position: 100
2023-09-07_14:59:08 MQTT2_client_000000892815b80c shades_6_position: 100

rstrouse, 2023-09-09:
You know that doesn't even match the client id from ESPSomfy RTS.  I have run this through the ringer and the MQTT client only emits the correct position and it does this without retain.  Is it possible you have some other client that is setting this to some other position.  Can you clear the data from the MQTT broker?  Maybe this position is simply a retained message.

wutz-ESP, 2023-09-12:
MQTT2 broker cleared, same problem.
The messages up to 2023-09-07 14:19:00 came direct from ESPSomfy-RTS WT32-ETH01 over LAN - i can observe MQTT2-Traffic in fhem also.
At 2023-09-07 14:59 opened WEB-Interface from ESPSomfy-RTS device at get just at this time all shade position 100 what was at 14:17:30 set with Somfy-Remote only DOWN button pressed.

i dont know why ... shade_3_position get always 0 every second
it is not possible to move 60 % in 2 seconds, that is the reason thinking tilt-time is used getting position instead of moving-time

>
2023-09-07_14:17:40 MQTT2_client_000000892815b80c shades_3_position: 60
2023-09-07_14:17:41 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:43 MQTT2_client_000000892815b80c shades_3_position: 63
2023-09-07_14:17:43 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:45 MQTT2_client_000000892815b80c shades_3_position: 64
2023-09-07_14:17:45 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:46 MQTT2_client_000000892815b80c shades_3_position: 65
2023-09-07_14:17:46 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:47 MQTT2_client_000000892815b80c shades_3_position: 66
2023-09-07_14:17:47 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:48 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:49 MQTT2_client_000000892815b80c shades_3_position: 68
2023-09-07_14:17:49 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:50 MQTT2_client_000000892815b80c shades_3_position: 69
2023-09-07_14:17:50 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:51 MQTT2_client_000000892815b80c shades_3_position: 70
2023-09-07_14:17:51 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:51 MQTT2_client_000000892815b80c shades_4_position: 75
2023-09-07_14:17:52 MQTT2_client_000000892815b80c shades_6_position: 57
2023-09-07_14:17:52 MQTT2_client_000000892815b80c shades_3_position: 71
2023-09-07_14:17:53 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:54 MQTT2_client_000000892815b80c shades_3_position: 72
2023-09-07_14:17:54 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:55 MQTT2_client_000000892815b80c shades_3_position: 73
2023-09-07_14:17:55 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:56 MQTT2_client_000000892815b80c shades_3_position: 74
2023-09-07_14:17:57 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:58 MQTT2_client_000000892815b80c shades_3_position: 75
2023-09-07_14:17:59 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:17:59 MQTT2_client_000000892815b80c shades_3_position: 76
2023-09-07_14:18:00 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:18:00 MQTT2_client_000000892815b80c shades_3_position: 77
2023-09-07_14:18:01 MQTT2_client_000000892815b80c shades_3_position: 0
2023-09-07_14:18:02 MQTT2_client_000000892815b80c shades_3_position: 78
2023-09-07_14:18:03 MQTT2_client_000000892815b80c shades_3_position: 79
2023-09-07_14:18:03 MQTT2_client_000000892815b80c shades_3_position: 80
2023-09-07_14:18:05 MQTT2_client_000000892815b80c shades_3_position: 81
2023-09-07_14:18:05 MQTT2_client_000000892815b80c shades_3_position: 1
2023-09-07_14:18:06 MQTT2_client_000000892815b80c shades_3_position: 82
2023-09-07_14:18:07 MQTT2_client_000000892815b80c shades_3_position: 83
2023-09-07_14:18:08 MQTT2_client_000000892815b80c shades_3_position: 84
2023-09-07_14:18:09 MQTT2_client_000000892815b80c shades_3_position: 85
2023-09-07_14:18:09 MQTT2_client_000000892815b80c shades_3_position: 2
2023-09-07_14:18:10 MQTT2_client_000000892815b80c shades_3_position: 86
2023-09-07_14:18:11 MQTT2_client_000000892815b80c shades_3_position: 87
2023-09-07_14:18:12 MQTT2_client_000000892815b80c shades_3_position: 88
2023-09-07_14:18:13 MQTT2_client_000000892815b80c shades_3_position: 89
2023-09-07_14:18:13 MQTT2_client_000000892815b80c shades_3_position: 3
2023-09-07_14:18:14 MQTT2_client_000000892815b80c shades_3_position: 90
2023-09-07_14:18:15 MQTT2_client_000000892815b80c shades_3_position: 91
2023-09-07_14:18:16 MQTT2_client_000000892815b80c shades_3_position: 92
2023-09-07_14:18:16 MQTT2_client_000000892815b80c shades_3_position: 4
2023-09-07_14:18:17 MQTT2_client_000000892815b80c shades_3_position: 93
2023-09-07_14:18:18 MQTT2_client_000000892815b80c shades_3_position: 94
2023-09-07_14:18:19 MQTT2_client_000000892815b80c shades_3_position: 100
2023-09-07_14:18:20 MQTT2_client_000000892815b80c shades_3_position: 95
2023-09-07_14:18:21 MQTT2_client_000000892815b80c shades_3_position: 96
2023-09-07_14:18:21 MQTT2_client_000000892815b80c shades_3_position: 5
2023-09-07_14:18:22 MQTT2_client_000000892815b80c shades_3_position: 97
2023-09-07_14:18:23 MQTT2_client_000000892815b80c shades_3_position: 98
2023-09-07_14:18:24 MQTT2_client_000000892815b80c shades_3_position: 99

[... 33 lines cut ...]
2023-09-07_14:18:46 MQTT2_client_000000892815b80c shades_3_position: 38
2023-09-07_14:18:46 MQTT2_client_000000892815b80c shades_3_position: 39
2023-09-07_14:18:46 MQTT2_client_000000892815b80c shades_3_position: 40
2023-09-07_14:18:47 MQTT2_client_000000892815b80c shades_3_position: 41
2023-09-07_14:18:47 MQTT2_client_000000892815b80c shades_3_position: 42
2023-09-07_14:18:49 MQTT2_client_000000892815b80c shades_3_position: 43
2023-09-07_14:18:49 MQTT2_client_000000892815b80c shades_3_position: 44
[... cut ...]
2023-09-07_14:18:56 MQTT2_client_000000892815b80c shades_3_position: 55
2023-09-07_14:18:56 MQTT2_client_000000892815b80c shades_3_position: 56
2023-09-07_14:18:57 MQTT2_client_000000892815b80c shades_3_position: 57
2023-09-07_14:18:58 MQTT2_client_000000892815b80c shades_3_position: 58
2023-09-07_14:18:59 MQTT2_client_000000892815b80c shades_3_position: 59
**2023-09-07_14:19:00 MQTT2_client_000000892815b80c shades_3_position: 60**
2023-09-07_14:59:07 MQTT2_client_000000892815b80c shades_3_position: 100

rstrouse, 2023-09-12:
If you can open a web socket on ws://<ip address>:8080 it will send you a JSON update only when there is a change.  Essentially, the socket is a lighter weight broker and does not carry all the additional weight of MQTT.  If you open the dev tools in chrome you will see the socket messages coming through.  Polling http request on the ESP is ugly but you can get the current states
```
GET /shades  -- Returns all information about the configured shades
GET /groups -- Returns all information about the configured groups
GET /shade?shadeId=<shadeId> -- Returns the specified shade
GET /group?groupId=<groupId> -- Returns the specified group
```

Are you sure that you are never publishing to the position leaf of the topic.  You should only publish to the following topics.  Everything else should be read only.  The MQTT publish on the ESP is relatively expensive so it only publishes when it needs to for consistency and it subscribes to the topics below for its MQTT command interface.
```
SomfyBuec/shades/<shadeId>/direction/set
SomfyBuec/shades/<shadeId>/target/set
SomfyBuec/shades/<shadeId>/tiltTarget/set
SomfyBuec/shades/<shadeId>/mypos/set
SomfyBuec/shades/<shadeId>/myTiltPos/set
SomfyBuec/shades/<shadeId>/sunFlag/set
```

wutz-ESP, 2023-09-14:
If i want to get information with httpmod  into fhem
http://192.168.10.89:8081/shadeCommand?shadeId=1&command=shades
extractAllJSON = 1

[image]

next opening httpmod in fhem the shade did not move but in web interface it is shown like this

[image]

<html>
<body>

bitLength | 56 | 2023-09-14 14:38:35
-- | -- | --
direction | 0 | 2023-09-14 14:38:35
downTime | 49000 | 2023-09-14 14:38:35
flags | 1 | 2023-09-14 14:38:35
flipCommands | 0 | 2023-09-14 14:38:35
flipPosition | 0 | 2023-09-14 14:38:35
inGroup | 0 | 2023-09-14 14:38:35
lastRollingCode | 167 | 2023-09-14 14:38:35
light | 0 | 2023-09-14 14:38:35
myPos | -1 | 2023-09-14 14:38:35
myTiltPos | -1 | 2023-09-14 14:38:35
name | Dach WZ-rechts | 2023-09-14 14:38:35
paired | 1 | 2023-09-14 14:38:35
position | 100 | 2023-09-14 14:38:35
proto | 0 | 2023-09-14 14:38:35
remoteAddress | 374797 | 2023-09-14 14:38:35
repeats | 1 | 2023-09-14 14:38:35
shadeId | 1 | 2023-09-14 14:38:35
shadeType | 3 | 2023-09-14 14:38:35
sortOrder | 0 | 2023-09-14 14:38:35
stepSize | 100 | 2023-09-14 14:38:35
sunSensor | 0 | 2023-09-14 14:38:35
target | 0 | 2023-09-14 14:38:35
tiltDirection | 0 | 2023-09-14 14:38:35
tiltPosition | 0 | 2023-09-14 14:38:35
tiltTarget | 0 | 2023-09-14 14:38:35
tiltTime | 7000 | 2023-09-14 14:38:35
tiltType | 0 | 2023-09-14 14:38:35
upTime | 48000 | 2023-09-14 14:38:35

</body>
</html>

rstrouse, 2023-09-15:
> If there is a my-position shade drives to it every time using this command
http://192.168.10.78:8081/shadeCommand?shadeId=1&get=shades

Yes it should with that command.  Try this command `http://192.168.10.78/shade?shadeId=1`

rstrouse, 2023-09-15:
To get real time updates you need to use MQTT or websockets.  In fact MQTT uses websockets to communicate with the broker.  I have seen articles online on how to set up a websocket in FHEM.

I also took a look at the FHEM MQTT2 implementation and it does look like it has separate topics for set and get.  You should only read from read-only topics so these would be your available get topic lists
```
SomfyBuec/shades/<shadeId>/name
SomfyBuec/shades/<shadeId>/windy
SomfyBuec/shades/<shadeId>/sunny
SomfyBuec/shades/<shadeId>/remoteAddress
SomfyBuec/shades/<shadeId>/lastRollingCode
SomfyBuec/shades/<shadeId>/shadeType
SomfyBuec/shades/<shadeId>/tiltType
SomfyBuec/shades/<shadeId>/flipCommands
SomfyBuec/shades/<shadeId>/flipPosition
SomfyBuec/shades/<shadeId>/shadeId
SomfyBuec/shades/<shadeId>/position
SomfyBuec/shades/<shadeId>/direction
SomfyBuec/shades/<shadeId>/target
SomfyBuec/shades/<shadeId>/tiltTarget
SomfyBuec/shades/<shadeId>/mypos
SomfyBuec/shades/<shadeId>/myTiltPos
SomfyBuec/shades/<shadeId>/sunFlag
```

And these would be your set topics list (NOTE: You cannot set the position you can only set a target).  So if you included the position topic then that would explain the feedback loop in your device.
```
SomfyBuec/shades/<shadeId>/direction/set
SomfyBuec/shades/<shadeId>/target/set
SomfyBuec/shades/<shadeId>/tiltTarget/set
SomfyBuec/shades/<shadeId>/mypos/set
SomfyBuec/shades/<shadeId>/myTiltPos/set
SomfyBuec/shades/<shadeId>/sunFlag/set
```

rstrouse, 2023-09-24:
I added some more documentation to the MQTT integrations wiki.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Integrations

rstrouse, 2023-10-12:
Another user has used the MQTT functions to integrate with FHEM.  v2.1.9 reduces the number of publishes and adds some additional functions that are useful to include a last will and testament that indicates whether the device is online or offline.

closed 2023-10-12 by rstrouse