rstrouse/ESPSomfy-RTS #133: No information in the logs

dominik4545, 2023-08-24:
Dear @rstrouse,

sorry for annoying you with another 'no logs'-topic here but I am totally clueless why my radio does not do anything :/

## Components
These components are my hardware setup:
* ESP-WROOM-32 (from some dealer on AliExpress)
* [EBYTE E07-M1101D-SMA](https://www.amazon.de/dp/B07P8S9M4W)
* [Oiyagai 433 MHz 3dbi 400 MHz - 490 MHz Antenna](https://www.amazon.de/dp/B07S7HG6WF)
I also tried it using the following components but they did not work either:
* [2 x Mini Antennas 433 MHz WiFi SMA](https://www.amazon.de/dp/B0BXWTJ585)

## Wiring
I also checked my wiring dozens of times and it alway matches the [picture shown in your documentation](https://github.com/rstrouse/ESPSomfy-RTS/wiki/Simple-ESPSomfy-RTS-device#esp-wroom-32esp32s-models). Anyway, here are a couple of pictures, maybe someone can find any difference:
<details>
  <summary>Images</summary>

  [image]
  [image]
  [image]
  [image]
  [image]
  [image]

</details>

## Software
Furthermore, I installed the software twice with the complete onboarding process but even then there were no logs.

## Configuration
My configuration looks like the following:
[image]

I saw in a YouTube-video to your project, that the PINs for RX and TX may be swapped - but this did not help at all. Even setting both to GPIO-12 or GPIO-13 did not help.

Do you have any idea how to fix this issue? Are there any additional logs?
I also tried three different remotes but none of them worked.

Thanks a lot in advance!

Best regards,
Dominik

rstrouse, 2023-08-24:
First lets take a look at your remotes.  What model remote do you have are they Somfy or some other brand?

dominik4545, 2023-08-24:
We bought the official [Somfy renovation kit](https://shop.somfy.de/somfy-renovierungskit-funk.html). And what I just noticed: they are mentioning something about 866 MHz on their website:
```
868,000 MHz - 868,600 MHz e.r.p. <25 mW
868,700 MHz - 869,200 MHz e.r.p. <25 mW
869,700 MHz - 870,000 MHz e.r.p. <25 mW>
```

rstrouse, 2023-08-24:
That is correct.  Your motors are IO HomeControl and not RTS motors.

dominik4545, 2023-08-25:
So sad, but thanks for the clarification

closed 2023-08-25 by dominik4545

LorenzKahl, 2023-08-28:
Hi @dominik4545, 

I just made the exact same mistake if it makes you feel any better ;-) 

I sourced all the parts, successfully setup the wiring, flashed the firmware and had the nice web interface working. But then no logs or any sign of TX/RX...

What a bummer :-/
I was hoping to finally have a solid solution allowing me to dump that crappy Tahoma box and their "only-half-of-the-time-working" API.

@rstrouse do you know of any other project that tackled bringing the 866 MHz platform to a local level?

dominik4545, 2023-08-29:
@LorenzKahl: I also purchased a TaHoma Switch on eBay yesterday as there seems to be no project which successfully implemented the io-homecontrol protocol. The only thing I found was a [project](https://github.com/Velocet/iown-homecontrol) which described what they found out via reverse engineering but as far as I understood there is no implementation of the protocol - even after this project was released. 
I would be happy, if the protocol could also be covered by your setup @rstrouse but I guess this would be a lot of hard work (and time I don't have) to get this working... :/

rstrouse, 2023-08-30:
Unfortunately Somfy seems to have hitched their wagon to that mess.  Until someone with that equipment gets sick of the silliness or consumers stop paying for those cloud based systems,  manufacturers will continue to produce them.  Frankly Somfy can keep their io motors.  I built this solution because I was frustrated with the lack of connectivity but it would be more frustrating if my connectivity relied on some far off server.