rstrouse/ESPSomfy-RTS #135: Link for buying hardware

jboesh, 2023-08-25:
I am interested in getting started with your project and struggling to find the right hardware.  You made suggestions what I should buy but you still have many different options on for example ali express and amazon.  

To avoid buying (and having to return) the wrong hardware. I hardly dare to ask: do you perhaps have a direct link where I can buy it? Of course it can be a ref link where you might earn something. I'm grateful.

rstrouse, 2023-08-25:
The ESP32 module and transceiver you linked to will work great.  The connectors you linked to however are male to female you need female to female.  Each end of the wire will be pushed onto a pin on either the transceiver or the ESP32.

Yes you will need a micro usb data cable for initial firmware load and that can be used to power the ESP32 board from any USB port after the initial firmware is loaded.  Updates to the ESP32 are all done using wifi after the initial firmware is loaded.

jboesh, 2023-08-25:
Thanks for the fast reply and confirmation! I still have doubts about buying from ali express because you have to wait a long time for delivery, but on the other hand you won't get it cheaper anywhere

rstrouse, 2023-08-25:
If you want amazon here are some links but I am in the US.
https://www.amazon.com/Teyleten-Robot-ESP-WROOM-32-Development-Microcontroller/dp/B08246MCL5/ref=sr_1_1_sspa?crid=15T4CSRHQEK88&keywords=esp32&qid=1692999566&sprefix=esp32%2Caps%2C210&sr=8-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&th=1

https://www.amazon.com/CC1101-Wireless-Transceiver-915MHZ-Antenna/dp/B01DS1WUEQ/ref=sr_1_2?crid=2088B8H8ILKPL&keywords=cc1101&qid=1692999651&sprefix=cc1101%2Caps%2C191&sr=8-2

closed 2023-09-08 by jboesh