rstrouse/ESPSomfy-RTS #136: Percent of open

dgmltn, 2023-08-25:
> No matter if I am standing on my head or in any other position, I cannot square a percentage of open in my mind. It doesn't work, 100% is not open and 0% is not closed.

I always think of it as, "if I'm standing in front of that window naked and my neighbor walks by, how embarrassed will I be?"

Seriously, thanks for your awesome work and writeup!

rstrouse, 2023-08-25:
Thanks for the confidence!

I am never embarrassed, then again I might only be able to do the naked thing.  The ability to stand on my head long enough for someone to walk by was never really in the cards.  I guess if your feet smell and your nose runs you are upside down anyway.

dgmltn, 2023-08-25:
This is an acceptable solution 👍

closed 2023-08-25 by dgmltn