rstrouse/ESPSomfy-RTS #137: Web interface down on ESPSomfy-RTS

Henk72, 2023-08-26:
The web interface of the ESPSomfy-RTS went suddenly down and was unrecoverable. 
The ESP was still reachable via ping.  Multiple resets of the ESP and also Router did not solve the issue. I am quite sure this is not a local network issue.

Firmware version was the latest v2.1.3

I don't know how to reproduce this, but maybe there is a memory leak or a log file growing and preventing the ESP from web access. Just wanted to inform you about this.

rstrouse, 2023-08-26:
@Henk72

I do not write log files for this specific purpose.  The only persisted file is the shade configuration, its size is constant, and only gets written when there is a change.

Run browse to this address to see if this file is served by the webserver.
`http://<<ip address of ESPSomfy RTS>>/shades.cfg`

If you can hook the ESPSomfy RTS up to a usb and monitor the serial output perhaps we can see a failure in there.

Henk72, 2023-08-27:
Thank you. I had already re-installed your firmware. If this happens again, I will try to see if shades.cfg is served.

closed 2023-08-27 by Henk72