rstrouse/ESPSomfy-RTS #138: Unknown Command for Somfy Key Go

lammetak, 2023-08-26:
Hi,

Im using a Somfy Key Go for my garage door.  Unfortunately I recieve a "unknow command (15)". As it is not listed in the project it was worth the shot to see if the project was working for my garage door. Is there anyway to see if this can be integrated.

Regards,

Tim

Logs:
---
[{"encKey": 164,
  "address": 15958522,
  "rcode": 963,
  "command": "Unknown(15)",
  "rssi": -70,
  "bits": 80,
  "proto": 2,
  "valid": false,
  "sync": 24,
  "pulses": [2568,2563,2525,2553,2542,2574,2518,2568,2538,2547,2547,2569,2506,2575,2517,2581,2517,2568,2537,2548,2547,2565,2531,2450,4854,1313,1295,1311,1319,664,648,1294,1312,692,636,692,642,1306,1300,1307,671,651,1312,697,630,1318,1318,1284,1321,1315,672,661,1302,1304,1298,1310,1346,639,653,1296,687,647,1322,666,643,1328,1302,1332,645,649,1315,1320,1292,1317,1312,1293,685,644,676,666,1301,1336,1295,1311,670,654,1316,667,634,684,641,1328,1291,674,667,649,669,1312,669,654,664,646,1322,665,668,678,635,1299,1334,670,653,664,649,669,659,680,643,680,640,691,634,683,641,676,666,674,647,661,653,693,633,683,641,1329,650,669,1310,670,655,1291],
  "time": "2023-08-26T13:41:31.644+0200"
},
{"encKey": 164,
  "address": 15958522,
  "rcode": 964,
  "command": "Unknown(15)",
  "rssi": -70,
  "bits": 80,
  "proto": 2,
  "valid": false,
  "sync": 24,
  "pulses": [2560,2597,2493,2592,2517,2576,2518,2557,2525,2569,2540,2570,2521,2563,2535,2544,2540,2562,2527,2577,2536,2554,2525,2461,4849,1301,1306,1299,1315,683,640,1324,1309,670,656,663,647,1319,1308,1296,690,645,686,650,667,662,1308,696,616,1337,1285,1318,663,672,671,638,1307,1326,675,644,1314,669,656,1289,693,634,1334,671,636,1315,1299,1304,701,635,1313,1336,1293,1286,1317,1311,667,660,679,644,1308,1321,1311,1295,685,639,1320,666,668,653,658,1327,1307,671,653,663,653,1316,666,643,675,666,1300,681,657,669,654,1292,1320,665,651,666,661,666,669,653,660,669,676,665,649,668,662,656,668,681,631,670,678,663,646,671,660,1308,671,662,1300,680,641,1298],
  "time": "2023-08-26T13:41:34.442+0200"
},
{"encKey": 164,
  "address": 15958522,
  "rcode": 964,
  "command": "Unknown(15)",
  "rssi": -73,
  "bits": 80,
  "proto": 2,
  "valid": false,
  "sync": 13,
  "pulses": [658,1844,2514,2560,2549,2555,2535,2548,2543,2572,2513,2571,2523,2490,4831,1294,1313,1320,1288,669,659,1318,1292,690,644,672,661,1304,1304,1300,682,641,695,653,662,670,1296,691,645,1304,1299,1308,671,650,686,648,1320,1316,670,661,1291,683,655,1297,688,650,1295,686,643,1299,1316,1307,672,650,1312,1344,1292,1294,1316,1293,683,670,651,657,1316,1335,1293,1286,691,663,1307,671,637,690,646,1302,1299,681,669,640,681,1310,670,636,1303,676,658,672,651,1311,1319,665,651,669,663,666,643,677,660,674,649,687,639,669,657,689,646,671,634,690,648,687,626,693,634,1315,1321,665,643,675,661,1299,1307],
  "time": "2023-08-26T13:41:34.584+0200"
},
{"encKey": 164,
  "address": 15958522,
  "rcode": 964,
  "command": "Unknown(15)",
  "rssi": -73,
  "bits": 80,
  "proto": 2,
  "valid": false,
  "sync": 13,
  "pulses": [2475,2526,2573,2519,2584,2526,2541,2542,2575,2527,2567,2507,2473,4855,1287,1323,1286,1322,686,650,1295,1313,669,661,665,644,1301,1316,1306,690,656,660,648,670,663,1309,647,662,1315,1300,1315,676,663,676,645,1312,1296,688,651,1293,686,644,1299,677,670,1285,696,655,1286,1319,1315,668,634,1309,1345,1290,1303,1301,1307,673,648,687,650,1294,1340,1273,1317,668,662,1290,695,637,690,646,1306,1299,681,655,670,652,1316,668,659,1284,697,634,678,645,1309,1323,659,657,666,656,691,644,671,662,664,643,682,639,696,628,688,649,669,661,683,644,678,633,697,626,1311,1317,684,641,673,660,1300,1307],
  "time": "2023-08-26T13:41:34.728+0200"
},
{"encKey": 164,
  "address": 15958522,
  "rcode": 965,
  "command": "Unknown(15)",
  "rssi": -73,
  "bits": 80,
  "proto": 2,
  "valid": false,
  "sync": 24,
  "pulses": [2587,2556,2523,2568,2541,2545,2550,2561,2532,2547,2543,2558,2521,2573,2536,2566,2517,2567,2535,2545,2540,2566,2529,2474,4854,1286,1299,1307,1316,689,649,1296,1313,670,659,666,646,1297,1328,1307,672,646,686,639,690,634,685,668,1303,1315,1288,1321,661,666,669,656,1312,668,661,1291,1321,666,667,1305,674,646,1312,669,657,1312,1293,1317,668,661,1306,1323,1295,1306,1317,1307,672,647,688,651,1297,1335,1298,1285,695,639,1329,651,660,670,652,1313,1321,662,647,668,661,1317,670,636,689,646,1311,669,655,665,670,1296,1314,678,639,693,633,665,673,671,659,665,646,687,651,668,659,666,643,675,660,674,649,660,678,667,634,1317,670,661,1316,669,636,1304],
  "time": "2023-08-26T13:41:35.824+0200"
},
{"encKey": 164,
  "address": 15958522,
  "rcode": 965,
  "command": "Unknown(15)",
  "rssi": -75,
  "bits": 80,
  "proto": 2,
  "valid": false,
  "sync": 13,
  "pulses": [650,1843,2536,2546,2544,2565,2537,2546,2543,2565,2510,2569,2545,2462,4852,1293,1312,1294,1317,669,663,1290,1320,658,644,701,640,1295,1312,1319,661,671,672,638,679,669,660,666,1296,1313,1319,1289,693,634,684,668,1304,664,650,1313,1320,665,646,1319,658,668,1305,674,650,1310,1295,1315,668,657,1318,1320,1282,1319,1315,1302,664,676,662,649,1320,1317,1300,1308,671,664,1302,678,642,673,649,1314,1318,665,641,666,673,1286,691,660,1285,695,635,690,643,1306,1299,682,659,669,655,661,646,695,639,680,646,683,654,669,656,669,644,672,659,681,645,679,639,697,627,1318,1316,665,649,668,663,1291,1320],
  "time": "2023-08-26T13:41:35.941+0200"
},
{"encKey": 164,
  "address": 15958522,
  "rcode": 965,
  "command": "Unknown(15)",
  "rssi": -72,
  "bits": 80,
  "proto": 2,
  "valid": false,
  "sync": 13,
  "pulses": [2471,2522,2563,2535,2574,2503,2584,2524,2574,2514,2574,2523,2483,4824,1297,1312,1296,1317,670,663,1288,1322,690,645,677,640,1300,1312,1314,679,640,674,649,688,650,668,637,1335,1322,1274,1299,689,647,686,650,1292,694,643,1297,1329,651,658,1303,682,652,1319,673,640,1297,1311,1324,655,668,1305,1325,1287,1300,1309,1300,682,666,669,654,1289,1344,1283,1293,686,641,1326,665,644,682,665,1297,1289,695,629,689,664,1294,686,645,1299,690,643,680,634,1321,1310,670,652,663,671,671,635,680,644,683,655,670,655,665,672,670,639,666,671,650,660,670,675,662,649,1296,1310,694,636,679,643,1308,1322],
  "time": "2023-08-26T13:41:36.068+0200"
}]

rstrouse, 2023-08-26:
Is there only one button on this remote?  It looks like if you press the button while it is moving it will stop,  if you press it when it is down it will open, and if you press it while it is up it will close.  Does it toggle directions if it has stopped in the middle or can you not stop the motor.

lammetak, 2023-08-26:
Exactly, if it is down it goes up. If it’s up it goes down. If pressed while moving it stops. And if it is stoped in the middle it wil reverse, so if it is closing and stoped it opens.

rstrouse, 2023-08-26:
Just to be sure was this log from you pressing the button on the remote?

lammetak, 2023-08-26:
Yes this was the log of pressing:

Once for open. Once stopped half way. Once for closure.

rstrouse, 2023-08-26:
Thanks.  Do you know if this motor can be controlled from another type of remote?  Is there a prog button on it?

lammetak, 2023-08-26:
Yes If I open the motor there is a prog button on there.

rstrouse, 2023-08-26:
Is there a prog button on the remote?

lammetak, 2023-08-26:
No there is not. In had to reprogram it once by using the prog button on the motor and clicking on the remote.

rstrouse, 2023-08-26:
The instructions are impossible to read.  While there is an English section in them it appears they tried to use pictures to describe the process.  These pictures are about as effective as a Xerox copy of my butt.  I will see if I can get the toggle command decoded.  It will likely require some work to get the HA component to work as a toggle cover and still replicate position.

lammetak, 2023-08-26:
Thanks for all the work. I will open the remote to see if I can find a button hidden inside it, as I can see that some remotes have them hidden inside.

rstrouse, 2023-08-26:
Can I get you to verify that the encKey is always 164 from your remote?  The detection of the toggle is a bit weird as they used the 80-bit protocol but left the detection bits empty.

lammetak, 2023-08-26:
yes 164 again:

I did a up -> stop -> down again

[{"encKey": 164,
  "address": 15958522,
  "rcode": 966,
  "command": "Unknown(15)",
  "rssi": -76,
  "bits": 80,
  "proto": 2,
  "valid": false,
  "sync": 24,
  "pulses": [2561,2570,2520,2567,2539,2544,2544,2555,2522,2581,2518,2571,2520,2570,2539,2573,2519,2559,2525,2583,2514,2571,2518,2474,4846,1294,1314,1319,1314,667,634,1312,1319,657,666,677,638,1318,1291,1320,665,672,671,661,1291,675,661,650,669,1313,1318,1286,692,657,663,644,682,656,672,653,691,645,1300,679,643,1309,696,653,1290,668,659,1307,1320,1310,672,659,1315,1309,1295,1313,1320,1290,666,658,690,644,1302,1308,1327,1293,673,686,1269,682,640,696,628,1314,1320,662,646,695,639,1307,673,660,1315,654,662,674,650,660,678,1297,1315,668,661,657,666,652,659,672,652,687,640,668,658,687,645,676,662,652,672,657,667,664,650,686,652,1292,665,670,670,662,1290,1301],
  "time": "2023-08-26T20:46:44.175+0200"
},
{"encKey": 164,
  "address": 15958522,
  "rcode": 966,
  "command": "Unknown(15)",
  "rssi": -82,
  "bits": 80,
  "proto": 2,
  "valid": false,
  "sync": 13,
  "pulses": [640,1955,2540,2551,2546,2542,2536,2571,2523,2577,2527,2551,2545,2446,4873,1268,1319,1311,1317,666,643,1305,1330,651,660,664,670,1286,1324,1287,695,653,665,648,1321,683,642,653,667,1302,1310,1299,684,642,695,652,663,639,691,616,706,645,1327,675,646,1310,668,655,1317,644,683,1285,1320,1291,696,637,1317,1325,1290,1298,1307,1299,690,643,681,668,1295,1336,1272,1309,672,653,1313,670,635,685,644,1327,1289,700,641,670,652,1314,643,683,1291,693,635,679,646,680,656,1317,1287,698,626,685,648,670,657,680,644,681,629,697,628,711,622,695,634,690,621,707,618,696,631,707,618,1329,661,672,659,663,1297,1336],
  "time": "2023-08-26T20:46:44.318+0200"
},
{"encKey": 164,
  "address": 15958522,
  "rcode": 967,
  "command": "Unknown(15)",
  "rssi": -70,
  "bits": 80,
  "proto": 2,
  "valid": false,
  "sync": 24,
  "pulses": [2557,2574,2523,2567,2531,2576,2520,2573,2519,2557,2540,2568,2516,2573,2533,2573,2526,2562,2526,2553,2540,2556,2547,2455,4848,1306,1300,1308,1322,662,651,1294,1312,668,658,687,643,1298,1314,1307,694,626,687,647,1319,1308,1299,1304,1328,1301,680,641,696,651,663,647,1320,1307,1324,651,669,1283,689,654,1311,667,661,1293,1319,1290,700,636,1313,1312,1317,1287,1315,1318,667,662,686,638,1306,1323,1285,1315,688,651,1294,691,643,672,661,1302,1304,674,648,661,663,1319,657,667,678,641,1323,660,654,666,657,1311,1320,659,665,651,659,665,672,661,667,668,662,651,669,659,663,666,661,659,663,679,642,663,659,688,642,1298,688,646,1313,668,659,1310],
  "time": "2023-08-26T20:46:47.074+0200"
},
{"encKey": 164,
  "address": 15958522,
  "rcode": 967,
  "command": "Unknown(15)",
  "rssi": -72,
  "bits": 80,
  "proto": 2,
  "valid": false,
  "sync": 13,
  "pulses": [645,1833,2514,2574,2523,2568,2528,2575,2522,2578,2518,2557,2545,2472,4847,1287,1320,1289,1317,667,665,1296,1310,668,662,659,666,1303,1299,1308,673,649,689,649,1319,1292,1320,1292,1322,1308,672,660,664,649,689,649,1318,1291,1320,666,668,1300,665,648,1312,668,658,1310,1319,1290,693,639,1311,1333,1296,1285,1322,1312,669,655,686,645,1300,1340,1278,1305,675,660,1315,685,649,667,659,1291,1319,657,666,653,661,1323,660,665,1293,691,644,672,662,1302,1305,649,670,660,663,668,659,665,671,671,639,691,643,680,641,671,651,689,650,668,660,658,667,680,637,1295,1341,664,633,686,644,1299,1316],
  "time": "2023-08-26T20:46:47.214+0200"
},
{"encKey": 164,
  "address": 15958522,
  "rcode": 967,
  "command": "Unknown(15)",
  "rssi": -72,
  "bits": 80,
  "proto": 2,
  "valid": false,
  "sync": 13,
  "pulses": [2483,2513,2574,2521,2571,2535,2549,2545,2553,2523,2580,2515,2473,4844,1313,1294,1314,1295,692,644,1297,1314,680,639,693,635,1311,1318,1296,689,638,691,642,1306,1299,1309,1299,1308,1315,690,624,716,611,685,667,1306,1301,1303,672,650,1312,690,633,1317,668,663,1291,1321,1283,695,640,1316,1300,1317,1296,1316,1306,675,646,687,626,1318,1312,1319,1285,692,637,1318,672,661,652,670,1284,1314,662,666,669,658,1292,693,634,1319,669,662,666,668,1305,1302,654,669,672,652,660,663,668,662,680,642,682,636,668,660,683,643,675,666,675,645,660,664,669,661,1308,1296,679,643,683,642,1322,1310],
  "time": "2023-08-26T20:46:47.357+0200"
},
{"encKey": 164,
  "address": 15958522,
  "rcode": 968,
  "command": "Unknown(15)",
  "rssi": -71,
  "bits": 80,
[... 5 lines cut ...]
},
{"encKey": 164,
  "address": 15958522,
  "rcode": 968,
  "command": "Unknown(15)",
  "rssi": -83,
  "bits": 80,
  "proto": 2,
  "valid": false,
  "sync": 13,
  "pulses": [649,1840,2533,2549,2548,2546,2534,2573,2526,2562,2528,2554,2545,2451,4866,1283,1320,1292,1321,681,643,1307,1299,681,640,696,628,1313,1319,1316,1292,713,614,1323,1293,677,639,1318,1318,1294,1316,693,639,666,644,1301,728,602,1334,671,650,1287,691,659,1291,669,660,1314,1304,1325,660,665,1292,1310,1320,1286,1322,1311,673,661,674,646,1312,1316,1287,1323,659,655,1322,684,641,674,634,1327,1305,674,649,659,665,1322,662,675,1272,710,620,701,635,1324,1286,674,649,684,624,720,609,706,620,702,629,697,628,711,622,696,635,687,621,704,631,698,626,710,597,1345,1295,699,631,695,624,1335,1272],
  "time": "2023-08-26T20:46:49.057+0200"
},
{"encKey": 164,
  "address": 15958522,
  "rcode": 968,
  "command": "Unknown(15)",
  "rssi": -75,
  "bits": 80,
  "proto": 2,
  "valid": false,
  "sync": 13,
  "pulses": [2522,2476,2587,2503,2593,2518,2579,2522,2572,2514,2529,2571,2469,4850,1281,1306,1304,1304,700,640,1318,1286,694,635,685,644,1302,1304,1326,1301,656,684,1300,1308,666,655,1290,1322,1307,1319,634,691,652,659,1298,682,643,1315,686,644,1298,688,646,1312,668,658,1311,1293,1310,669,661,1314,1303,1291,1307,1314,1312,669,655,689,641,1304,1300,1308,1297,680,653,1318,662,670,670,662,1276,1327,675,648,683,650,1292,686,645,1322,652,670,658,667,1295,1314,670,657,663,647,669,660,679,645,681,657,668,655,662,672,670,636,678,644,683,667,663,648,686,650,1293,1318,668,662,666,644,1301,1330],
  "time": "2023-08-26T20:46:49.194+0200"
}]

rstrouse, 2023-08-27:
Thanks I’ll have a firmware soon for you to try out.

rstrouse, 2023-08-27:
I added two types of garage door motors and translated the toggle from your Keygo remote.  However, I believe you should be able to use the 3 button version of the garage door type when setting up the motor.  If it does not work try the 1-button but the instructions for the Dexxo indicate that there is such a thing as a remote with an up down and my button.

Install the firmware and application from the main repository.  These files are located here for the [firmware](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.ino.esp32.bin) and here for the [application](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.littlefs.bin)

rstrouse, 2023-08-28:
The command is correct.  Did you pair ESPSomfy RTS to the motor.  Also please make sure you have RTS selected as the protocol and 80 bits.

rstrouse, 2023-08-28:
Do you have more than 1 remote or are you using more than one channel on a single remote?  I see two sets of addresses in your outputs.

lammetak, 2023-08-28:
I have only 1 remote. Let me test with pairing it to the motor.

rstrouse, 2023-08-28:
If you ever wondered what one of those research chickens feel like that have a picture flashed in front of their beak and are branded intelligent because they can pick out a symbol, this is your chance to fulfill you dream.  The key to all of this is to peck on a picture at the proper location and a bit of food will drop into the tray below.  

That being said I originally misinterpreted the documentation and was sending the wrong command for the prog.  You need to update your firmware again with the [SomfyController.ino.esp32.bin](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.ino.esp32.bin) file located in the repository.

So once you channel your inner bird brain, it looks like is you need to press the two buttons on the bottom of an existing remote that is paired with the motor.  Then immediately press the button in the middle for 2 seconds at which time the light on the motor will flash.   Then you need to go into ESPSomfy RTS and press the prog button on the pairing screen.  If you have recited all the requisite voodoo chants the light should flash for 5 seconds.  This is the extent of the paring documentation for that Dexxo Pro motor.  After staring at this picture for several minutes I realized that you need to either put on a chicken head or comb your hair into a multi-tipped row down the center of your head.  I won't go into the lengths I took to understand this.

[image]

lammetak, 2023-08-28:
Dude I’m dying xD hahaha.I did pair the esp with the motor using the button on the motor, but is does not open(I can see the close state now) I need to wait until tomorrow to update the esp to the new firmware. I will update you as soon as I have tested on the new firmware.

lammetak, 2023-08-28:
Uploaded firmware using my phone. Unfortunately still not working. Let me redo all steps tomorrow and pair the remotes again.

lammetak, 2023-08-28:
/

lammetak, 2023-08-28:
Ooops misclick

so I did some tests trying to pair with the 2 button method. But according to the instructions it is also the way to unpair the device. So I think it unpairs itself when I try to pair it with the esp.

when I use the remote, I don’t see logging anymore. So I need to figure out what happend there.

lammetak, 2023-08-28:
Okay, got the logs back. But stil nothing changed. Pairing worked with the button on the motor. It flashed for 5 second which should mean it is working. Only thing I noticed that the rolling code in the log is higher then the rolling code I can read from the pairing section. Is this the last rcode in the ui or is this the one set while pairing?

rstrouse, 2023-08-28:
The rolling code is unique to the remote.  The ESPSomfy RTS definition should be different than that of the remote.  Have you tried switching between the 1-button and 3-button version?

lammetak, 2023-08-28:
Yes even tried changing bits and to rtv and rtw. But the garage door is not moving.

my last logging:

[{"encKey": 164,
  "address": 15892986,
  "rcode": 2770,
  "command": "Toggle",
  "rssi": -69,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 24,
  "pulses": [2568,2565,2531,2578,2513,2580,2524,2569,2516,2571,2525,2567,2528,2566,2541,2560,2529,2552,2539,2575,2519,2557,2542,2473,4843,1289,1318,1293,1318,689,645,1299,1304,672,660,676,647,1316,1292,1313,1318,691,617,694,636,1303,1328,1299,1308,1324,657,666,1294,1314,668,660,690,644,1301,677,645,682,666,1296,1310,665,657,1317,1293,1315,669,664,679,645,682,640,1322,1312,667,657,685,644,698,642,673,647,687,650,667,663,1306,1298,1304,674,659,675,647,688,650,666,661,668,640,675,661,674,646,686,650,1293,1310,668,663,665,668,651,658,1298,1337,667,652,660,667,669,635,692,641,681,658,668,653,666,645,695,637,678,643,684,653,669,656,664,670,1299,681,643,679,640,1322,1287],
  "time": "2023-08-28T22:42:15.194+0200"
},
{"encKey": 164,
  "address": 15892986,
  "rcode": 2770,
  "command": "Toggle",
  "rssi": -92,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 13,
  "pulses": [646,1875,2507,2574,2531,2549,2543,2571,2529,2555,2544,2542,2534,2477,4844,1303,1298,1318,1295,690,643,1308,1322,659,666,671,652,1291,1319,1293,1318,670,645,671,659,1316,1298,1307,1326,1266,698,658,1302,1303,694,622,687,648,1318,669,667,677,635,1297,1335,669,654,1287,1320,1290,693,633,684,644,701,641,1300,1309,670,650,687,649,668,640,706,643,679,641,696,626,1311,1293,1337,665,661,667,644,673,661,674,648,687,651,668,660,666,644,674,659,1306,1302,675,648,685,643,665,666,1288,1320,669,668,680,637,673,664,674,651,660,663,683,642,677,634,700,694,665,632,650,660,688,636,657,664,1292,699,629,674,665,1313,1307],
  "time": "2023-08-28T22:42:15.911+0200"
},
{"encKey": 164,
  "address": 15892986,
  "rcode": 2771,
  "command": "Toggle",
  "rssi": -68,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 24,
  "pulses": [2551,2572,2542,2562,2524,2577,2511,2580,2513,2569,2536,2573,2523,2552,2530,2576,2535,2555,2547,2556,2518,2572,2524,2467,4856,1293,1314,1318,1310,669,640,1315,1297,689,647,659,666,1297,1314,1293,1317,718,612,690,645,674,636,666,670,1305,1301,1308,672,676,1290,1318,1318,1295,1337,645,661,666,645,1299,1316,673,663,1300,1308,1325,661,656,693,636,687,646,1299,1317,672,640,702,644,686,650,667,663,665,643,702,634,1322,1310,1294,689,650,642,683,666,667,651,661,672,648,688,650,667,641,689,644,1299,677,643,1336,672,650,665,648,1318,1310,645,664,687,643,682,658,668,653,664,647,698,633,665,667,679,659,670,653,660,664,670,636,690,649,1311,670,653,1315,667,659,1291],
  "time": "2023-08-28T22:42:16.724+0200"
},
{"encKey": 164,
  "address": 15892986,
  "rcode": 2771,
  "command": "Toggle",
  "rssi": -69,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 13,
  "pulses": [644,1861,2537,2549,2522,2571,2540,2573,2527,2563,2527,2554,2536,2458,4852,1305,1299,1316,1298,691,645,1307,1299,681,656,667,655,1315,1296,1317,1296,713,619,672,665,664,646,680,641,1322,1311,1298,687,651,1294,1312,1317,1289,1352,638,650,688,637,1321,1312,653,665,1300,1309,1321,661,663,693,639,684,640,1312,1301,681,642,696,653,663,670,672,640,679,643,685,665,1296,1315,1320,663,647,668,660,681,641,680,664,667,635,690,641,681,656,669,655,1293,691,633,1311,673,658,1295,1321,687,645,661,657,669,658,663,672,671,660,652,670,658,653,670,658,665,672,670,662,666,642,681,642,1313,1311,664,657,1316,670,657,1310],
  "time": "2023-08-28T22:42:16.931+0200"
},
{"encKey": 164,
  "address": 15892986,
  "rcode": 2772,
  "command": "Toggle",
  "rssi": -69,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 24,
  "pulses": [2569,2566,2541,2548,2546,2566,2529,2549,2529,2561,2549,2551,2538,2574,2519,2566,2537,2547,2545,2568,2513,2572,2526,2467,4864,1289,1311,1292,1315,668,659,1316,1292,690,644,673,664,1303,1304,1304,1305,1322,660,664,668,654,1311,1320,1288,1327,676,644,685,640,1318,1293,692,662,1283,695,638,665,669,1303,1300,682,656,1318,1293,1319,665,648,695,639,678,669,1286,1324,683,639,669,660,688,648,686,640,668,657,687,645,1325,1302,1307,673,647,660,667,668,656,664,671,669,637,678,669,657,655,668,658,1292,691,632,1310,694,637,677,644,1308,1322,658,664,667,661,663,670,657,664,668,660,664,669,656,664,667,661,663,644,680,656,668,656,665,674,1295,682,642,1310,671,646,1312],
  "time": "2023-08-28T22:42:55.983+0200"
},
{"encKey": 164,
  "address": 15892986,
  "rcode": 2772,
  "command": "Toggle",
  "rssi": -70,
  "bits": 80,
[... 269 lines cut ...]
},
{"encKey": 164,
  "address": 15892986,
  "rcode": 2782,
  "command": "Toggle",
  "rssi": -69,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 13,
  "pulses": [644,1869,2509,2574,2546,2553,2532,2545,2543,2573,2521,2556,2539,2470,4843,1309,1295,1310,1317,659,665,1304,1314,659,664,667,661,1282,1323,1306,672,659,666,675,1289,1319,1314,1294,1309,1320,1310,1293,692,643,686,641,693,632,1312,668,662,666,668,1306,1301,681,639,1325,1311,1300,681,650,669,661,658,668,1306,1300,681,665,669,654,687,643,673,665,676,645,688,639,1319,1291,1321,686,644,671,635,692,641,679,642,691,633,694,640,672,664,662,643,1309,1322,658,654,673,654,687,640,1297,1316,679,640,697,626,687,650,667,661,664,669,658,655,693,633,665,670,671,662,674,645,659,653,669,660,1315,668,661,658,670,1303,1300],
  "time": "2023-08-28T22:45:07.654+0200"
},
{"encKey": 164,
  "address": 15892986,
  "rcode": 2782,
  "command": "Toggle",
  "rssi": -92,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 13,
[... cut ...] ,658,1309,1321,1308,1295,1306,1299,1312,1311,670,651,687,646,695,640,1304,676,665,665,668,1311,1296,686,626,1320,1313,1294,683,644,676,665,673,649,1313,1296,686,673,669,636,690,647,689,652,667,659,665,668,1309,1290,1312,693,634,688,645,671,635,689,645,680,656,670,654,663,653,695,632,1305,1299,692,642,679,657,671,652,1291,1361,621,671,678,629,691,645,686,721,598,635,689,644,678,663,674,644,660,674,661,664,659,683,674,637,1307,664,669,650,669,1308,1300],
  "time": "2023-08-28T22:45:08.555+0200"
}]

lammetak, 2023-08-28:
[image]

rstrouse, 2023-08-28:
When it was pairing did it immediately flash the light?  The commands sent are identical to the commands that are received from the remote so if the address is paired I would expect it to work.  Change the type to the 3-button and set the bit length to 56-bit.

lammetak, 2023-08-28:
That did the trick by setting it to 56 bit and 3-button. You’r a hero mate!

rstrouse, 2023-08-28:
The best part of that is you now have so much more control over that door.  You should be able to even set a favorite position.  Next I will be publishing a release for this.  I did some icon work so that it looks more like the motion of a garage door.

lammetak, 2023-08-28:
Yes that is great. Tomorrow I will test it a little bit more as I’m only can work from my phone and verify by checking my camera. If I find any issues I’ll you know.

rstrouse, 2023-08-28:
Is there a light control on your remote?  If there is, can you please capture the light button presses?

lammetak, 2023-08-29:
I got 1 walk thingy from key go or what is called (can’t find a name on it, only on google 1 result with key go wall search. This device has 1 button. I used this to integrate with the esp. 

The key go has 4 button (1 big and 3 small) which do all the same and that is to control the door. Yesterday I found a YouTube video on how to map 1 of the buttons to the light. Let me look in to this tonight.

lammetak, 2023-08-29:
I tried pairing the light but no succes, I’m wondering if the “external light” is the one on the dexxo. Found couple of manuals but all of them have a different process.

Is it possible that the home assistant integration is not working with the new firmware. If I press the button nothing moves. Within espsomfy it is working

rstrouse, 2023-08-31:
The new version as well as HA support for garage doors have been released.

rstrouse, 2023-09-01:
I am going to close this as I think I have gotten all the components in place and the icon display should now be correct.  If there are further issues open a new issue.

closed 2023-09-01 by rstrouse