rstrouse/ESPSomfy-RTS #139: Changing motor display orders

bbs2web, 2023-09-02:
This project is amazing! This was the first time tinkering with an ESP32 and your detailed guides made this an absolute breeze!

I first added the roller blinds in my office, then added the ones in the kitchen and then those in the TV lounge. The display order of the motors however unfortunately shows them in the order they were added, I had incorrectly presumed that this would be alphabetical, or that I could change the order manually.

Feature request:
Kindly asking for the ability to re-order the list of motors.

rstrouse, 2023-09-02:
Not a bad suggestion.  I will add this to the enhancements list.

rstrouse, 2023-09-03:
This will be in the next version 2.1.5 and I am doing testing on it right now.  You will be able to drag and drop the order in which you want them to appear.

rstrouse, 2023-09-09:
Alright v2.1.5 has been released.  To change the order of the display simply drag and drop the desired position in the configuration section for the shades list and/or the groups list.

closed 2023-09-16 by rstrouse