rstrouse/ESPSomfy-RTS #140: Tilt not working as expected

BasvanH, 2023-09-02:
Hi,

First i would like to congratulate you on a wonderful piece of software you put together, I'm really surprised by its features and looks.

First time user and have put v2.1.4 on my ESP32 with a E07-M1101D-SMA connected. All working fine, can add blinds and control them. The tilt however doesn't work as it should. Blinds are full down and current tilt is at 100%. When I change it to anything lower then 90% it triggers the blinds to go up. 99% also triggers blinds to go up. Between 98 and 90% controls the tilt.

I have set the tilt type to `Integrated`, when selecting `Tilt Motor` the tilting doesn't work.

Any idea what might be wrong here?

Regards,
Bastiaan

rstrouse, 2023-09-02:
The tilt motor setting is only for folks that have two motors per blind head.  These operate by long pressing on the up or down buttons to change the tilt.  It does not sound like the tilt mechanism you have.

The way the integrated tilt mechanism operates mirrors the operation found on many types of motors.  Because there is typically only one motor in these mechanisms, the motor can only roll up or down,  The way this works is that when going down the slats first rotate to the 100% position then the blind starts driving down.  When going up the slats first rotate to 0% then the blind starts moving up.  This means that you can open or close the tilt regardless of the blind position.

For this type of mechanism you set the tilt time to the amount of time that it takes for the slats to go from  configuration there are two sets of timings there is a tilt time which defines how long it takes for the motor to rotate the slats from 0% to 100%.  

Then there is the up/down time.   These are separate times because of that pesky gravity thing and the down operation is typically faster than the up.  These times are the time it takes for the blind lift movement and does not include the tilt time so if you have the blind in the full up position with the tilt at 0% then pressing the down button will first close the slats to 100% over the time set as the tilt time.  Then it will begin lowering to the 100% position.

If this doesn't sound like the way your blinds work you may have a different mechanism for the tilt.  Do you have a link to a user manual for your blinds.  Does yours may operate similar to what is described in this post.  This is where you can only change the tilt when the lift position is at 100%.  At some point I am going to tackle this tilt mechanism as well.
https://github.com/rstrouse/ESPSomfy-RTS/issues/116

BasvanH, 2023-09-04:
Ok I understand it better now. I do have a integrated motor where tilt is done by the up or down movement. I was confused as I have a remote wil a dial jog which controls the tilt. I never saw that its actually up - down movement. I had to decrease the tilt time to 1200ms for it to span the tilt from 0 to 100%. I now have full control over the tilt.

rstrouse, 2023-09-05:
Great I am going to close this for now.  Submit another issue if you have any other problems.

closed 2023-09-05 by rstrouse