rstrouse/ESPSomfy-RTS #141: Blinds going wild after reset of ESP

mn-gsc, 2023-09-02:
First of all cudos and thanks for the great SW. Really like it a lot. I am using the MQTT interface and do some nice stuff with iobroker to automate my home.
But from time to time my ESP does a reset. And then all blinds are going crazy. Down 2sec, stop, up 1sec, stop, down 3sec and so on. For about a minute, then everything is back to normal. Using the latest version 2.1.14, but also encountered this with 2.1.12.
Anything I have misconfigured on my side? MQTT is configured with user and password.
Thanks and keep on going!

rstrouse, 2023-09-02:
This sounds a lot like you have set the qos on your iobroker to something other than 0.  Also, there is no need to retain this message either for any set commands.  You want to use qos 0 and retain false.  

The MQTT broker will retransmit any request that it feels did not get delivered as soon as the MQTT client subscribes to the topic.  This will mean that every topic setting in the history will be sent.  The broker will be aggressive at above qos 0.  We do not desire, need, or want this type of guaranteed delivery.  There are use cases for it but this is certainly not one of them.

I do not know much about the settings in io broker but when you publish to the set topics these should be qos 0 and optionally retain false.

mn-gsc, 2023-09-03:
Awesome, thank you so much. I think I found the issue (it is sitting in front of the computer). Indeed I have used QOS 1 and retain flag. Setting it back to 0 and clearing the MQTT broker didn't help at first step. After some analysis I found out having used a wrong MQTT command at the beginning of setting up ESPSomfy RTS and instead to "/shades/3/direction/set" I was sending to "/shades/3/direction". And this in combination with the QOS 1/retain flag (according to my logs and my abilities to read it) caused the situation after a reset and explained that my first try of clearing things up did not work well.

I now have completely removed the MQTT broker plugin from iobroker and reinstalled it to clear all databases/retain queues and set it up with iobroker default settings (especially QOS 0, no retain flag) and sending only to "/direction/set". Now it looks like working as expected. I don't know whether there is something which could be done as a robustness meassure for people like me sending to "/direction" instead of "/direction/set". Apologies for that! 

Nevertheless thank you very very much for pointing me into the right direction in such a short time and getting things solved. This is really outstanding!

closed 2023-09-03 by mn-gsc