rstrouse/ESPSomfy-RTS #143: ESPSomfy not always responding

taugusti, 2023-09-03:
I have a garage door on the ESPSomfy - and when it works, it works great!
The motor is a "Dexxo Optimo RTS" pn 5147629A V93-00.
I have one real issue - and maybe the other not-so-important issue is related - I include both below to give you the full picture.

**#1: no response after idle**
When the door / espsomfy have been idle for some time - I think like a day or something, it takes many commands to the ESPSomfy before the door starts moving.

I have tuned the radio and on a 1m distance I get about -30dBm.

The radio log shows 80-V when I click a remote. That should indicate 80-bit RTV protocol. If I use that, the garage door does not move. Both 56-bit and 80-bit RTS works.
[image]

**#2: not understanding remote commands**
I have added the remotes to the garage door, but the operation of them are not indicated in the ESPSomfy position. 
It seems in the log that the commands from the remotes are in an unknown format: "Unknown(15)".

rstrouse, 2023-09-03:
You need to update to the latest version 2.1.4.  Support for the Dexxo was not added until this version.  It is 80-bit RTS.

rstrouse, 2023-09-03:
When you set the Dexxo motor up choose Garage (3-button) this will allow you to control many more aspects of that motor.  Also, if you use home assistant don't forget to update the integration.

taugusti, 2023-09-03:
Device went offline after online fw update
now reprogramming from scratch

rstrouse, 2023-09-03:
Hmmm... That is the second report of this yet I have updated at least 10 at this point with no issues.

taugusti, 2023-09-03:
I think it may have been my own fault
I think I might have mixed up the application / firmware / "onboard" versions
I have it running on 2.1.4 now - just need to get the radio and pairing to work again...
No more time today - will continue tomorrow.

taugusti, 2023-09-04:
Ok, now up and running again on 2.1.4 (2.1.3 HASS integration)
codes from the remotes are now converted to "toggle".
I selected garagedoor and 3-button and everything seems to work.
I now look forward to try it after 24 hr unused :-)

Thank you for your support!

rstrouse, 2023-09-05:
Great! Submit another issue if you have any more problems.

closed 2023-09-05 by rstrouse

taugusti, 2023-09-06:
Works like a charm now . thanks!

only issue seems to be that if I click toggle too fast - it ignores the message and I have to wait "some time" for the garage door to respond to toggle commands

rstrouse, 2023-09-06:
This may be the motor not allowing commands right away.  How long is "some time" and I assume you are using the 3-button setting where it has an explicit up, down, and my command.

taugusti, 2023-09-11:
Hi again
I have now tried to see if I can identify a pattern
but - it seems to work even with quite short intervals - down to about 1 sec.
From time to time it "blocks" for maybe 30 sec and then works again.
During this period the remote works - so it is apparently not the motor that fails
but I cannot find a pattern - and the problem is not present in normal use.
So - I think it is great!