rstrouse/ESPSomfy-RTS #144: ESP Auto reboot scheduler

kostas125, 2023-09-03:
Hello! Is there any way to schedule auto reboot everyday for esp?? I have to hard reset the board once a day, cause it seems unavailable every day at 2pm (probably a coincidence?)
[image] 

Thanks for the support and your effort!!!

rstrouse, 2023-09-03:
Double check to see if you reserved the DHCP address for the ESPSomfy RTS device in your router.  Your router may be reassigning addresses to the resolved name when you lose wifi.

kostas125, 2023-09-03:
> Double check to see if you reserved the DHCP address for the ESPSomfy RTS device in your router. Your router may be reassigning addresses to the resolved name when you lose wifi.

First thing  I do is the dhcp reservation. I ll check if the AP is activated when the problem occurs and I ll run a reboot script to see if will fixed the issue. An auto reboot option in the ui will be great by the way 😊

kostas125, 2023-09-04:
Yeap... Unfortunately it seems that every time I ve lost connection, the AP is activated on esp, but the home wifi is up n running...
I don't know what should I do...

rstrouse, 2023-09-05:
When you lose wifi and it cannot re-connect after re-trying 100 times or the SSID becomes unavailable it will open up the AP for 3 minutes and if nothing connects to that it will then retry the station mode looking for the SSID.  It will continue to do this until it gets a connection.  I have a test rig that has been connected non-stop for months.  When ESPSomfy RTS loses wifi connection it will no longer be available in HA.

What is the signal strength you are seeing?

kostas125, 2023-09-07:
> 

[image]

rstrouse, 2023-09-21:
@kostas125 did you get your issue resolved?

kostas125, 2023-09-21:
> @kostas125 did you get your issue resolved?

I still have disconnections but the issue seems to be on my side. I ll close this thread now.

I would like to thank you again for all the good work!!

closed 2023-09-21 by kostas125